ttlappalainen/NMEA2000 #10: Add support for PGN 127257

sarfata, 2016-06-28:
It would be great to have support for PGN 127257 so we can transmit and receive attitude messages (heel/pitch/roll).

[It is documented in Canboat](https://github.com/canboat/canboat/blob/master/analyzer/pgn.h#L1573):

````
{ "Attitude", 127257, true, 7, 0,
  { { "SID", BYTES(1), 1, false, 0, "" }
  , { "Yaw", BYTES(2), RES_RADIANS, true, "rad", "" }
  , { "Pitch", BYTES(2), RES_RADIANS, true, "rad", "" }
  , { "Roll", BYTES(2), RES_RADIANS, true, "rad", "" }
  , { 0 }
  }
}
````

It would be good to know what is the reference framework too. I am guessing Yaw is relative to true north. Not sure about pitch and roll.

ttlappalainen, 2016-06-28:
Hi,

No Idea. Do I add it and try with NMEA Reader, what it shows?

Timo

28.6.2016, 13:49, Thomas Sarlandie kirjoitti:

sarfata, 2016-06-28:
That would be great. I think if someone could capture data from a sensor that sends them (I think some of the Airmar wind sensors send that information) that would be great too.

ttlappalainen, 2016-06-29:
Hi,

Yaw could be also relative to  your course over water. For aircrafts it 
has been defined (at least on some referencies) angle between your 
course over air.
pitch = positive, when your bow rises
roll = positive, when tilted right

28.6.2016, 13:49, Thomas Sarlandie kirjoitti:

ttlappalainen, 2016-07-19:
I just added attitude PGN 127257. I could only test it with NMEA Reader. It also showed Yaw as negative value, so it may be defined as I mentioned in similar ways as in aircrafts as an angle between vessel course over water.

sarfata, 2016-07-22:
Thanks. I will try with my Raymarine i70 but I doubt it will be supported.

closed 2016-07-22 by sarfata