ttlappalainen/NMEA2000 #11: GNSS info decimal problem

TheAwesem, 2016-07-11:
GNSS info long and latitude only have 2 decimal so the outpust is rounded.
If I look up the coordinates it is a block away from where i am beacause the output is rouded with only 2 decimals This is the output i get:
latitude: 52.34
longitude: 5.53
is there any whay to get the full sentence?

ttlappalainen, 2016-07-12:
Sorry for delay.
Could you please explain more? What code and board/processor you are using? In N2kMessages.h/.cpp function ParseN2kGNSS(...) returns lat/long as double, so they will be returned with full resolution.

TheAwesem, 2016-07-14:
Thanks for the quick response
I mean that when we want to see our position from the nmea 2000 gps we look at the gnss data and that only shows us 2 decimal i get:
latitude: 52.34
longitude: 5.53
While I have to get the following:
Latidude: 52.336415
Longitude: 5.527889
I am using an arduino mega with the seeed can bus shield with the ActisenseListener script.

ttlappalainen, 2016-07-14:
Ok. So you have Mega with ActisenseListener  connected to N2k bus. What application you are using for showing the data? NMEA Reader?

TheAwesem, 2016-07-14:
picocom with 115200 baudrate

ttlappalainen, 2016-07-14:
So are you reading data from Mega USB in binary format? ActisenseListener just read data from bus and sends it to USB in Actisense binary format. I do not understand how you can read that with picocom. You can also download NEMA Reader from http://www.actisense.com/products/nmea-2000/ngt-1/downloads-ngt1.html?download=105:ngt-1-nmeareader-software and use that.

TheAwesem, 2016-08-30:
Hello,
Sorry I didn't reply.
So things have changed. The Arduino mega with the can bus shield thing broke so I began to use the Arduino Due with the "DataDisplay2" script. After some struggle I got it to send something from a NMEA2000 GPS. But the only thing I get are empty time stamps.

[image]

I didn't add or change anything to any of the files.
The strange thing is that if I add the Actisens NGW-1 to the system we receive the correct GPS data.
Is there any way I get to see the GPS without adding the Actisense NGW-1?

ttlappalainen, 2016-08-30:
Are you sure that you have right termination on the bus? 
What NGW-1 is doing? Jus staying connected or will it route some data from NMEA0183?
My test device is on the boat now, so I can not unfortunalety check that nothing would have broke DataDisplay2 since last update. At least my devices on boat works fine with updated code.

closed 2017-01-22 by ttlappalainen