ttlappalainen/NMEA2000 #14: Attribute imported code

lkarsten, 2016-07-25:
Hi.

Reading through your NMEA2000 routines for some inspiration, having just read the same code in CANBoat, I find the two following functions very similar:

* https://github.com/ttlappalainen/NMEA2000/blob/master/NMEA2000.cpp#L207
* https://github.com/canboat/canboat/blob/master/common/common.c#L366

The number of open source NMEA2000 implementations and developers is very small. If it is based on the other work and not attributed, I think that looks very bad and diminishes your other work. I'm just a potential user of your library, and for me this was a red flag.

Please consider adding a comment with attribution where (/if) your code use code from other projects.

ttlappalainen, 2016-07-25:
Hi,

I do not remember that I would have used canboat code, but can not be sure - I have not downloaded canboat code at all. You can find same canboat code from http://fresh-lucky.blogspot.fi/2010_09_01_archive.html or http://yachtelectronics.blogspot.fi/2010/09/code-for-converting-iso-11783-id-to.html, but I think I did not use either of them.

If some code has has some mention about e.g. GPL lisencing, I normally try to note the source. But yes when I am just quickly testing some code part found from internet, I may forget it and if it stays also in final solution, I may forget to comment the source. I'll try anyway be more strict in future.

About commenting what should I then comment, since the original source i unclear?

lkarsten, 2016-07-25:
Hi.

Thanks for replying.  

Isn't Kees the author of canboat? That explains one of the sources. The other makes my browser do something funny when I click the background, so I suspect it is some seo-spam with stolen content?

Anyway, thanks for looking into it and for writing the library.

-Lasse

ttlappalainen, 2016-07-25:
Hi,

Yes, Kees is the original author. He has done a great job for NMEA2000 database, which I learned via OpenSkipper. N2kMessages.h/.cpp are started by using OpenSkipper database and so leads back to Kees. Others who has added message handlers may have used pgn.h from SignalK also originated from canboat.

Hopefully you find library useful. I have been thinking to implement it also for rasberry.

Timo

ttlappalainen, 2016-08-18:
I added thanks to Kees and for OpenSkipper project to some places.

closed 2016-08-18 by ttlappalainen