ttlappalainen/NMEA2000 #15: DynamicTransmission flags not implemented as bools

sixstringsg, 2016-07-29:
DynamicTransmission has Transmission Oil Pressure as well as Oil Temperature, but does not have a way similar to DynamicEngine to trigger specific flags. The flags are as follows: 
Transmission Sail Drive - Bit 5
Transmission Low Oil Level - Bit 4
Transmission Low Oil Pressure - Bit 3
Transmission Over Temp - Bit 2
Transmission Check - Bit 1

ttlappalainen, 2016-07-31:
Hi,

Sorry for delay.

Copy this after current inline void SetN2kTransmissionParameters(...) 
function and report is it OK.
inline void SetN2kTransmissionParameters(tN2kMsg &N2kMsg, unsigned char 
EngineInstance, tN2kTransmissionGear TransmissionGear,
                      double OilPressure, double OilTemperature,
                      bool flagCheck=false,       bool 
flagOverTemp=false,         bool flagLowOilPressure=false, bool 
flagLowOilLevel=false,
                      bool flagSailDrive=false) {
   unsigned char DiscreteStatus1=0;

   if (flagCheck) DiscreteStatus1 |= B00000001;
   if (flagOverTemp) DiscreteStatus1 |= B00000010;
   if (flagLowOilPress) DiscreteStatus1 |= B00000100;
   if (flagLowOilLevel) DiscreteStatus1 |= B00001000;
   if (flagSailDrive) DiscreteStatus1 |= B00010000;
   SetN2kPGN127493(N2kMsg, EngineInstance, TransmissionGear, 
OilPressure, OilTemperature,DiscreteStatus1);
}

Timo

29.7.2016, 22:50, Jason Stewart kirjoitti:

sixstringsg, 2016-08-01:
When we tried this code in our project: https://gist.github.com/sixstringsg/958de3b85a3ef60b386b18bfee795d12
None of the alarms went off. We can manually make them go off by changing the hex code at the end of line 7, so we know the Lowrance is correctly seeing flags. Is there something I pasted wrong? Thanks for all the help!

ttlappalainen, 2016-08-01:
Hi,

Yes, you pasted it a bit wrong. I meant that paste it to the 
N2kMessages.h after:
inline void SetN2kTransmissionParameters(tN2kMsg &N2kMsg, unsigned char 
EngineInstance, tN2kTransmissionGear TransmissionGear,
                      double OilPressure, double OilTemperature, 
unsigned char DiscreteStatus1) {
   SetN2kPGN127493(N2kMsg, EngineInstance, TransmissionGear, 
OilPressure, OilTemperature, DiscreteStatus1);
}

Then you call it in your code like dynamic engine:
...
bool CheckTransmission;
bool TransmissionOverTemp;
bool TransmissionLowPressure;
bool TransmissionLowOilLevel;
...
// Set values on your test function. This is as example
void CheckTransmissionStatus() {
    CheckTransmission=digitalRead(1);
TransmissionOverTemp=digitalRead(2);
TransmissionLowPressure=digitalRead(3);
    TransmissionLowOilLevel=digitalRead(4);
    TransmissionSailDrive=digitalRead(5);
}
...
void SendN2KTransmissionParam() {
     static unsigned long TransmissionUpdated=millis();
     tN2kMsg N2kMsg;
     if (TransmissionUpdated+TransmissionUpdatePeriod<millis()) {
         TransmissionUpdated=millis();
         CheckTransmissionStatus();
SetN2kTransmissionParameters(N2kMsg,1,N2kTG_Forward,TransmissionOilPressure(),
TransmissionOilTemperature(),CheckTransmission, TransmissionOverTemp, 
TransmissionLowPressure,TransmissionLowOilLevel,TransmissionSailDrive);
         NMEA2000.SendMsg(N2kMsg);
     }
}

I have no description how DiscreteStatus parameter works. This as other 
N2k messages on library has been found by reverse engineering (most of 
them originally by CanBoat project). So you provided me the bits for 
DiscreteStatus, but we need more testing.

So now your saying "We can manually make them go off by changing the hex 
code at the end of line 7" confuses, since call
SetN2kTransmissionParameters(N2kMsg,1,N2kTG_Forward,TransmissionOilPressure(),TransmissionOilTemperature(),0x7);
should mean that you have set CheckTransmission, TransmissionOverTemp, 
TransmissionLowPressure all on instead of off.

So with above code you should be able to to control transmission status 
bits by turning on/off related io-pin. So please test the combinations, 
so we maybe find how they work.

Timo

1.8.2016, 6:59, Jason Stewart kirjoitti:

sixstringsg, 2016-08-01:
That worked perfectly, all alarms are correct! Thank you!

ttlappalainen, 2016-08-18:
This has been now implemented.

closed 2016-08-18 by ttlappalainen