ttlappalainen/NMEA2000 #17: Review behavior of PGN filtering

thomasonw, 2016-08-05:
_SetForwardOnlyKnownMessages()_  has the documented goal of deciding if a received CAN messages is to be forward to the attached serial stream for external processing.  If set=TRUE the function _CheckKnownMessage()_ will  compare incoming PGNs against the know message lists and forward those to the stream (if enabled).  However, when set FALSE (Default action), ALL incoming CAN messages will be forward to the stream.

This same filtering behavior is also applied to the optional program callback _SetMsgHandler()_.  The  question is:  Is this the desired behavior?  That _SetForwardOnlyKnownMessages()_ affects both streams and user  _SetMsgHandler()_ call backs?  

Or, should _SetForwardOnlyKnownMessages()_ be modified to ONLY affect streams (matching the user documentation), and a separate call be added _SetHandleOnlyKnownMessages()_ be created?

A related question:  [https://github.com/ttlappalainen/NMEA2000/pull/12](url)
Added the ability to replace the existing messages lists which are accepted by the NMEA2000 lib.  By replacing the default list one is able to extend the message PGNs supported (example, other standard J1939 messages, or another J1939 overlay such as RV-C).  It also allows one to 'pre-filter out' messages to the program callback which are not being used. example:  A battery monitor does not need messages for GPS or AIS..

Is there a desire to add an additional API call to _augment_ vs. _replace_ the default messages lists?  
  AugmentSingleFrameMessages() 
  AugmentFastPacketMessages()

ttlappalainen:  Did I capture these questions correctly?

-al-

ttlappalainen, 2016-08-06:
I check through the code logic and if there is no additional issues, I'll add SetHandleOnlyKnownMessages(). Note that this causes backward compatibility issue.

I think "Augment..." is not good. As a name it would tell that in principle user could call this several times, but this is not the case. Either we should decide to have only one extra set and use e.g. SetUserSingleFrame-/FastPacketMessages() or have vector size e.g. 5
     #define N2kMessageGroups 5
     const unsigned long SingleFrameMessages[N2kMessageGroups];
     const unsigned long FastPacketMessages[N2kMessageGroups];
adn define bool ExtendSingleFrameMessages(); , which would return false in case of vector is full. I just like word "extend" because I think that other than naturally English speakers would know it better than "augment", which I had to google. Or maybe I am wrong.

But anyway the first question is that is there any reason to have more than one extra set for user? In large code one may want to dynamically extent the set. But then there should be also possiblity to disable extra set. In case of need for extra set we could also extent functionality of 
void SetUserSingleFrameMessages(const unsigned long *_FastPacketMessages);
in future by changing definition to:
bool SetUserSingleFrameMessages(const unsigned long *_FastPacketMessages; int SetNumber=0);
Also we could change current definition to:
bool SetSingleFrameMessages(const unsigned long *_SingleFrameMessages; int SetNumber=0);
but that is not so clear.

thomasonw, 2016-08-08:
In my usage it is just as easy for me to receive ALL the messages and let me filter them.  The approach I am using is a table with PGNs and assigned handlers (An extension of the approach used in several of your examples).  It is just as quick for me to scan my table then call the handler - likely quicker overall then having two table scans (not to mention the extra Emery needed for the repeated PGN table).  

Of course, FastPacket messages NEED a table for you to identify them as fast-packet (makes no sense for me to replicate the fast-packet code).  But perhaps those are few in number (I am not using any)

I think just being able to pass in an 'extend' table is OK.  No real need to allow multiple levels of them.   If indeed the program is complex, perhaps it can best deal with any levels / needed changes its self - I would say let SetUserxxxMessages and ExtendUserxxxMessages simply take a pointer to a table to be used; replacing any prior prior pointer if there was one.   They should also be allowed to pass in NULL (or 0 if you wish), to 'clear' these tables and return to only the default one inside the NMEA2000 code.

Will look for the SetHandleOnlyKnownMessages() addition and change my code appropriately.  Not an issue at this time as I am waiting for these details to settle out before I release code.

ttlappalainen, 2016-08-08:
Hi,

So I'll add and test SetUserxxxMessages and leave possibility to extent by using SetNumber (or what ever it would be in future). I'll try to do these withing few days.

Timo

ttlappalainen, 2016-08-18:
So I added SetHandleOnlyKnownMessages, ExtendSingleFrameMessages and ExtendFastPacketMessages. If you have not set SetForwardOnlyKnownMessages(true), there is no compatibility effect.

closed 2016-11-13 by thomasonw