ttlappalainen/NMEA2000 #18: pressure issue with PGN 130310 and 130311

usauerbrey, 2016-08-14:
In N2kMessages.h at the PGNs 130310 and 130311 the comment says "AtmosphericPressure   Atmospheric pressure in Pascals. Use function mBarToPascal, if you like to use mBar"
But when I feed 1234.5 mBar with mBarToPascal the result on Raymarine i70 is 14964 and in Actisense NMEA Reader 5791400 pascal.
When I do not use the mBarToPascal-conversion, everythink looks fine. i70 show 1235 and in NMEA Reader say 123500 pascal.
Is the comment wrong?
please clearify.

Thanks Uwe

ttlappalainen, 2016-08-14:
Code seem to be wrong. Point was to use SI-units and related conversion routines. So you should have
    N2kMsg.Add2ByteUDouble(AtmosphericPressure,100);
in SetN2kPGN130310 in N2kMessages.cpp
I have other updates under testing, so I'll fix that on next update.

sarfata, 2016-08-14:
I am not sure I understand this.

In KBox, I store values in Pascal internally (in the BarometerMeasurement
class) and send them as is to PGN 130311 and it works well on my i70:

https://github.com/sarfata/kbox-firmware/blob/master/lib/KBox/src/converters/BarometerN2kConverter.cpp

And:
https://github.com/sarfata/kbox-firmware/blob/master/lib/KBox/src/tasks/BarometerTask.cpp

So for me, it seems to work as documented.

Thomas

ttlappalainen, 2016-08-14:
Thomas is right. I looked wrong for my last response. I tested code as it is and changed line on example MessageSender:
SetN2kOutsideEnvironmentalParameters(N2kMsg, 1, ReadWaterTemp(), CToKelvin(25.3), mBarToPascal(1013.5));
and NMEAReader shows me 101400 Pascal

ttlappalainen, 2016-08-14:
Sorry for messing, it it too late to test anything. I already made the correction for the library and tested it with that. So it is as my first response. Thomas: you read pressure in pascals, but then you create:
BarometerMeasurement m(temperature, pressure / 100); 
meaning you have converted it to mBar (or hPA). And as my document says, it expect values as PA (Pascal).

ttlappalainen, 2016-08-18:
I just updated the library to use Pascal unist, so now you have to use mBarToPascal!

sarfata, 2016-11-08:
@ttlappalainen I think you can close this issue. There is no issue :)

closed 2016-11-09 by ttlappalainen