ttlappalainen/NMEA2000 #20: Getting TemperatureMonitor up and running on Arduino Uno

pantze, 2016-08-25:
Hello Timo, 

I tried to get the library up and running on my Arduino Uno, in debug mode. (I don't have a CAN board connected (yet)).

It behaved strange and hung after just 13 message frames. I made two changes that fixed the problem in NMEA2000.cpp:

Added a check for dbMode in the Open method (line 232) when writing the state of the CAN bus and ForwardType is Text:

`if ( DeviceReady || dbMode!=dm_None)....
`
Otherwise you will get a lot of CAN bus failed to open

I also added a quick exit in ParseMessages (line 693):

`if (dbMode != dm_None) return; // No much to do here when in Debug mode
`
It would otherwise try to do an address claim 13 times and the get the Uno to get stuck.

With these changes, you get a nice address claim PNG and then the temp PNG every 2,5s.

Something to consider?
(I first tried to do a pull-request from my branch)

ttlappalainen, 2016-08-25:
Thanks for the fix. 
Line 693 is pointing to line 
if (!Open()) return; 
So shouldn't "if (dbMode != dm_None) return;..." stay after that so that it works as document that you do not need to call Open? Or is there something difference in sources we are looking?

If you do not make pull, I can add it to the next update.

pantze, 2016-08-26:
Hmmm, didn't read the documentation regarding no need to call Open. See the point of doing this implicitly. Adds more robustness when it is handled by the API.

The problem with the UNO (with no CAN controller) is that when i debug-mode, the CAN-API seems to corrupt the serial stream which results in a freeze.

Let's move the return-if-debug to line 694 in ParseMessages after the Open-statement, as you pointed out. This will keep the implicit Open behaviour and also safe guard from unnecessary calls to CANGetFrame.

The challenge is that you will get lot's of adress claim messages since DeviceReady will never be true when in debug mode. To solve this and to keep the original behaviour, could you adjust setting DeviceReady in line 230 to include a check for debug mode:

`DeviceReady=dbMode!=dm_None || CANOpen();
`

With this in place, the Temperature monitor works like it is suppose to when in debug mode on the UNO. One adress claim message followed by three temperature messages every 2,5 s.

BTW, would mind adding me as a contributor? Then I can create a branch and be able to create pull-requests. Right now I don't have permissions for either.

sarfata, 2016-08-26:
@pantze you can clone the project in your GitHub account and make pull-request based from branches in your clone.

pantze, 2016-08-28:
Thanks @sarfata. I think I'm doing this right but I'm new to GitHub. I have cloned the project to the "Desktop" (on a Mac). But I cannot "publish"/commit my new branch. Tried it both from Desktop and command line. I get: "remote: Permission to ttlappalainen/NMEA2000.git denied to pantze".

I understand that I need the branch committed before making a pull-request back to master.
(I've tried the same for a test repository I created and there it works so https connection to github should be working)

sarfata, 2016-10-18:
@pantze sorry for the delay here.

1. Fork the project on github (you might want to redo that because the project has evolved a lot since August)
2. Clone the project from your repo to your computer
3. Create a new branch, do your changes, commit them
4. Push your branch to your fork, that is in your GitHub account
5. Visit the project on GitHub, the button to create a pull-request should automatically appear.

Sorry I did not see your reply earlier...

pronaga2, 2017-02-24:
Hello friends! 
I am a beginner in the Arduino programming!
I tried to get TemperatureMonitor up and running on Arduino UNO . I am using Timo's libraries and documentation , but on actisense nmea reader I do not read anything! It reports a overtime problem .
I need help! Thanks

ttlappalainen, 2017-02-24:
Did you notice that library uses default SPI CS pin 53. So you have to use #define N2k_SPI_CS_PIN 10 (or what ever pin CS is connected) before any library include.

pronaga2, 2017-02-28:
i'm sorry for the delay! I have understood the problem!
I will inform you after the test.
Thank you for the reply

pronaga2, 2017-03-01:
Hello Timo,
I changed the pin in the nmea2000_CAN library, but in debug mode, on the Actisense reader, I don't read anything. Have I to do the changes suggested by Panze on the first post?
Thank's

Tayary, 2017-03-06:
Trying to get the actisense listener going on UNO with CANBUS shield from RobotShop. The receive led on the shield is blinking but no data received. If I use the listnensender app it fails with the messed CAN device failed to open. When I try the battery monitor script it claims to work but the transmist light not blinking and no data. It seems I am not talking to the shield but cannott see why. Also tied on a MEGA and same results
Dave Allin

ttlappalainen, 2017-03-07:
pronaga2: Sorry I missed your response. You should not mofidify NMEA2000_CAN.h code. It will be repalced, if you download new revision. Instead on your own .ino add definition before including NMEA2000_CAN.h. So do:
#define N2k_SPI_CS_PIN 10
#include <NMEA2000_CAN.h>

I also added changes discussed with Panze to the code.

ttlappalainen, 2017-03-07:
Tayary: Since you are using UNO, you will also need to have #define N2k_SPI_CS_PIN 10 (or 9) as I mentioned on previous post. Note that I have tested library with MEGA and schema https://github.com/ttlappalainen/NMEA2000/blob/master/Documents/ArduinoMega_CAN_with_MCP2515_MCP2551.pdf. So now when you are using ready CANBUS shield, you have to check that all SPI pins are right. As far as I understood from shield documentation, it has UNO pinout as default. Also now in shield version 1.2 the default CS pin is 9 (see doc on http://wiki.seeed.cc/CAN-BUS_Shield_V1.2/). So you should use #define N2k_SPI_CS_PIN 9. On MEGA board shield should use same hw pins as my schema but I can not see on the 1.2 shield layout picture that CS would be connected to MEGA default cs (pin 53), so you still may need to use #define N2k_SPI_CS_PIN 9 even for MEGA board.

Tayary, 2017-03-07:
Hi,

I am actually using the shield from robotshop which is http://www.elecfreaks.com/estore/can-bus-shield.html <http://www.elecfreaks.com/estore/can-bus-shield.html> and claims to use D10. However I have tried D9 as well and sadly both UNO and MEGA boards still giving the same error message. I have also tried the CAN.H examples and same problem so clearly I am getting something pretty basic wrong. 

I have installed NMEA2000_MASTER, CANBUS_shield master and NMEA2000_mcp_master with those names. Any ideas welcome

Tayary

ttlappalainen, 2017-03-08:
You say they both are giving same error message, but what is the message? I expect that CANBUS_shield has been downloaded from my GitHub, since that has the latest changes for N2k lib.

Tayary, 2017-03-08:
I think my problem is more basic. I decided to delete all the libraries, start from scratch and try the temperature monitor sample with no shield. I am running this on a UNO and downloaded your GitHub NMEA2000 master then MCP library and the CAN bus shield library for the MCP2515 controller.

I have followed the uncomment and comment instructions, the temperature monitor compiles but no data at all on the serial port. I added a test write of some text and that works but no PGN data.

The tried adding the ATMEL AVR libraries but this has no effect.
Have I loaded the right libraries?

Tayary, 2017-03-08:
I have now investigated further and am more confused than ever.  When I run temp monitor nothing happens so I added the code as shown below and when it runs mills appears not to be working properly.

void SendN2kTemperature() {
  static unsigned long TempUpdated=millis();
  Serial.print("millis reading ");Serial.print(millis());
  Serial.print("tempstore ");Serial.print(TempUpdated);
  
  tN2kMsg N2kMsg;

  if ( TempUpdated+TempUpdatePeriod<millis() ) {
    TempUpdated=millis();
    SetN2kTemperature(N2kMsg, 1, 1, N2kts_MainCabinTemperature, ReadCabinTemp());
    NMEA2000.SendMsg(N2kMsg);
    SetN2kEnvironmentalParameters(N2kMsg, 1, N2kts_MainCabinTemperature, ReadCabinTemp());
    NMEA2000.SendMsg(N2kMsg);
    SetN2kOutsideEnvironmentalParameters(N2kMsg, 1, ReadWaterTemp());
    NMEA2000.SendMsg(N2kMsg);
    Serial.print(millis()); Serial.println(", Temperature send ready");

ttlappalainen, 2017-03-08:
Why you think so? With your code you will fill serial buffer, since your print will be called with maximum speed. Note that TempUpdated is static, so it will be iniatialized once on first call. Then it will be updated within if clause.

Tayary, 2017-03-08:
Hi,

surely even at speed the loop will take some milliseconds  and always go up as I invoke mills within the print statement. If I remove the two print lines then nothing ever appears on the monitor. I would expect a print ‘temp send ready’ every 2.5 seconds.

ttlappalainen, 2017-03-08:
If you have those two lines, does it print 'temp send ready' in every 2.5 s?

Tayary, 2017-03-08:
No,

it just loops with mills showing 220 221 on alternate lines and never seems to get into the body of the function

ttlappalainen, 2017-03-09:
And does it same on Mega? Have you tried simple .ino with only printing millis()? Have you read document about buffers? Note that Uno has only 2kB SRAM while Mega has 8kB and library has some default buffers. Have you tried to run it on debug (NMEA2000.SetDebugMode(dm_ClearText); )mode and without CABBUS shield? Etc...

Tayary, 2017-03-09:
Hi,

thank you for the patience in helping me. I have tried on the MEGA without
shield and the loop does work which is encouraging. On my windows machine I
now get the screen below

[image: Inline images 1]

this is with the lines

  Serial.begin(115200);
  NMEA2000.SetForwardStream(&Serial);
  // If you want to use simple ascii monitor like Arduino Serial Monitor,
uncomment next line
 NMEA2000.SetForwardType(tNMEA2000::fwdt_Text); // Show in clear text.
Leave uncommented for default Actisense format.

  // If you also want to see all traffic on the bus use N2km_ListenAndNode
instead of N2km_NodeOnly below
 // NMEA2000.SetMode(tNMEA2000::N2km_NodeOnly,22);
  NMEA2000.SetDebugMode(tNMEA2000::dm_Actisense); // Uncomment this, so you
can test code without CAN bus chips on Arduino Mega
  NMEA2000.SetDebugMode(tNMEA2000::dm_ClearText);
 // NMEA2000.EnableForward(false); // Disable all msg forwarding to USB
(=Serial)

in the code. On my MAC system with the same code I get

[image: Inline images 3]

which is slightly readable on the Actisense reader in that I get one set as
shown below

[image: Inline images 4]
but it only reads once, maybe it does not like all the CANBUS error
messages. I installed the ide on Windows from the apps store and it is
version 1.8.1.0 the MAC version just took from the arduino site and run
from the downloaded file. I have installed the same libraries  on both
platforms so am a bit confused now.

ttlappalainen, 2017-03-09:
Images does not show. There is no sense to test Actisense reader on debug mode. Try with temperature monitor, which sends termperature PGNs.

Tayary, 2017-03-09:
I have converted the screen shots from PNG to a PDF file which I hope is readable. My explanation of actin sense reader was poor, what I meant to explain was that using the ACTISENCE PC reader program it can read the temperature data when the arduino mega is running the script compiled on the MAC. However it only seems to get one set of data before it stops recording. I assume this is caused by all the error messages about CANBUS not opening.

ttlappalainen, 2017-03-09:
It is NMEA Reader. In debug mode, it should not care of CANBUS. I just tested with WindMonitor example with Mega and without CANBUS by uncommenting Serial.begin(115200); and NMEA2000.SetForwardStream(&Serial); . Commenting NMEA2000.EnableForward(false); and adding NMEA2000.SetDebugMode(tNMEA2000::dm_Actisense); above it. Works fine - NMEA Reader continues showing data.

There is one other thing I have noticed that with some computer the serial (USB) just gets stuck after some messages. So maybe the problem is not in Mega. Instead it can be on your computer drivers.

pronaga2, 2017-03-22:
Hi Timo!
I bought an Arduino Mega and finally I can read the data on actisence reader!
thank you for the patience in helping me!

ttlappalainen, 2017-04-15:
Hi,

Did you try to reduce memory by adding:
  NMEA2000.SetN2kCANMsgBufSize(2);
  NMEA2000.SetN2kCANSendFrameBufSize(30);
before NMEA2000.Open();

r2d290 got Temperature monitor running on UNO, so it may be just memory problem. UNO has only 2K and as default there may be bigger dynamic reservations. Also disabling message forwarding to USB with
NMEA2000.EnableForward(false);
may reduse runtime memory.

So I think UNO should work for simple things.

Tayary, 2017-04-20:
Hi,

I have purchased a due and the battery monitor sample now works.  I will try some more things on the UNO and see if I can get it to work as well.

Can I use one arduino to write both battery data and temp data? Seems I would need to initiate the setup for each write, will this work?

Thanks

Dave

ttlappalainen, 2017-04-21:
You can send different messages from same device and you do not need to initiate setup for each write. See my example MessageSender. 

In principle on NMEA 2000 bus each device tells, what type it is. With my library this has been done on setup with call NMEA2000.SetDeviceInformation with class and function parameters. If you make device for your own use, it is not so critical, if it says to be Temperature function and still sends battery voltage. With call NMEA2000.SetProductInformation you can set "Manufacturer ID" to e.g. "Dave's bat & temp monitor". For this you could also set class to 25 (Inter/Intranetwork Device) and function 132 (Analog to NMEA 2000 Gateway), which is rather "wide" definition.

uzgit, 2017-05-10:
I am having a similar issue to what OP had been having with his Uno. I have the CAN-BUS Shield v1.2 on an Uno and I have connected the CAN bus to a Garmin Intelliducer for reading depth. Using the receive_check.ino file from the CAN_BUS Shield library I can see the binary data, but when I use the ActisenseListener file I cannot see any data. The device does write to the serial port "CAN device ready" after I set #define N2k_SPI_CS_PIN 9 and #define N2k_CAN_INT_PIN 10. The RX light blinks at one or two hertz, and the interrupt light is continually lit up.
Any advice?
Thanks

ttlappalainen, 2017-05-10:
Hi,

As I have wrote earlier, library requires 2-4k memory for full operation. Uno has only 2k and can not handle everything. Anyway if you squeeze momory on setup
NMEA2000.SetN2kCANMsgBufSize(2);
NMEA2000.SetN2kCANSendFrameBufSize(30);
before NMEA2000.Open();
it can handle some messages and works with simple node like battery or temperature monitor. It should then also run ActisenseListener, but will for sure miss messages. Did you read my post on 15.4 on same issue?

uzgit, 2017-05-10:
I did read your post on April 15, but even after adding the lines you had suggested, I got the same results. I have also tried this same .ino file with and without the edits on two different arduino mega 2560, getting exactly the same results each time. On the serial port I see "CAN device ready" but see no data.

Sketch below:
`// Demo: NMEA2000 library. Bus listener. Sends all bus data to serial in Actisense format.
// Note! If you use Arduino Mega, it can not handle lot of messages without using interrupt.
// enable interrupt by defining pin you have used for MCP2515 interrupt pin. See below
// definition

//#define N2k_CAN_INT_PIN 20
//#define N2k_SPI_CS_PIN 53  //tried for arduino mega

#define N2k_SPI_CS_PIN 9
#define N2k_CAN_INT_PIN 10

#include <Arduino.h>
#include <NMEA2000_CAN.h>

void setup()
{
  Serial.begin(115200);
  NMEA2000.SetForwardStream(&Serial);  // PC output on due native port
  NMEA2000.SetForwardType(tNMEA2000::fwdt_Text); // Show in clear text
//  NMEA2000.EnableForward(false);
//  NMEA2000.SetN2kCANMsgBufSize(2);
//  NMEA2000.SetN2kCANSendFrameBufSize(30);
//  NMEA2000.SetDebugMode(tNMEA2000::dm_Actisense);
  NMEA2000.Open();
}

void loop()
{
  NMEA2000.ParseMessages();
  //Serial.println("Testing.");
}
`

I am wondering if I should continue attempting to debug my setup on the Arduino Uno/Mega or if I should just get an Arduino Due.

ttlappalainen, 2017-05-10:
Hi,

With Uno you had those buffer definitions uncommented? Mega should work with your sample sketch, if the interrupt pin has been defined right. You could also try not to define int pin, so it will not use interrupt and looses messages. But at least you should see some messages.

Do you have terminal resistor on your bus? You need at least one on short testing environment.

uzgit, 2017-05-10:
Yes, with the Uno I had the lines relating to the memory issue uncommented.

The sketch is now working. The problem seems to have been the interrupt pin definition. I removed its definition and can now see data similar to the following:

'''
Pri:3 PGN:128267 Source:0 Dest:255 Len:8 Data:BD,FF,FF,FF,FF,FF,7F,FF
Pri:3 PGN:128267 Source:0 Dest:255 Len:8 Data:BE,FF,FF,FF,FF,FF,7F,FF
Pri:5 PGN:130312 Source:0 Dest:255 Len:8 Data:BE,0,0,64,75,FF,FF,FF
Pri:3 PGN:128267 Source:0 Dest:255 Len:8 Data:BF,FF,FF,FF,FF,FF,7F,FF
Pri:3 PGN:128267 Source:0 Dest:255 Len:8 Data:C0,FF,FF,FF,FF,FF,7F,FF
Pri:5 PGN:130312 Source:0 Dest:255 Len:8 Data:C0,0,0,64,75,FF,FF,FF
Pri:3 PGN:128267 Source:0 Dest:255 Len:8 Data:C1,FF,FF,FF,FF,FF,7F,FF
Pri:3 PGN:128267 Source:0 Dest:255 Len:8 Data:C2,FF,FF,FF,FF,FF,7F,FF
Pri:5 PGN:130312 Source:0 Dest:255 Len:8 Data:C2,0,0,64,75,FF,FF,FF
Pri:3 PGN:128267 Source:0 Dest:255 Len:8 Data:C3,FF,FF,FF,FF,FF,7F,FF
Pri:3 PGN:128267 Source:0 Dest:255 Len:8 Data:C4,FF,FF,FF,FF,FF,7F,FF
Pri:5 PGN:130312 Source:0 Dest:255 Len:8 Data:C4,0,0,64,75,FF,FF,FF
Pri:3 PGN:128267 Source:0 Dest:255 Len:8 Data:C5,FF,FF,FF,FF,FF,7F,FF
'''

ttlappalainen, 2017-05-10:
So check the right int pin definition then you should get reliable reading with Mega. Uno will anyway have problem for receiving on bus, which has some devices connected. For simple things like TemperatureMonitor, Uno should be ok.

uzgit, 2017-05-10:
I am using an Uno to read from only a single device and I can see both of the messages that it is supposed to output (temperature and depth), so I think it may be ok on the Uno. Haven't tried DataDisplay2.ino yet, and I will need to strip it down first, but I'm hoping I can get it small enough to fit on the Uno.

uzgit, 2017-05-11:
As a followup, my stripped down version of DataDisplay2.ino does work well on the Uno. I Just commented out all the handlers for messages I didn't need, in order to shrink the sketch enough that it actually fit on the Uno. It only handles temperature and water depth messages.

MoritzDanninger, 2017-06-01:
Hello together, I tried the same things with my Aaduino Uno and CAN-Bus shield as uzgit and managed to display the data on the serial monitor just like he did. But now I want to send the data to the N2k bus in order to display them on a Raymarine MFD. And here is where I got stuck because i can't seem to get any data from my Arduino to my MFD.
It would be nice if someone had an idea how I could fix my problem.

ttlappalainen, 2017-06-09:
Unos just has too little RAM, so it may cause troubles. Under this issue there are instructions how to define smaller RAM requirements. You can try them. I'll prefere some better board like Teensy 3.2, Mega or DUE. Check also https://hackaday.io/project/11055-kbox

ttlappalainen, 2017-06-11:
Please read "Memory requirements" on main page (Readme.adoc).

HoBingHong, 2018-12-25:
@uzgit 
I would like to ask: I bought a Garmin Intelliducer NMEA2000 sonar.
Try to execute arduino uno with the draft you wrote.
But got the following error message. Why?

C:\Users\user\Music\Documents\Arduino\libraries\NMEA2000-master/NMEA2000_CAN.h:140:39: note: in expansion of macro 'MCP_CAN_CLOCK_SET'
tNMEA2000_mcp NMEA2000(N2k_SPI_CS_PIN,MCP_CAN_CLOCK_SET,N2k_CAN_INT_PIN,MCP_CAN_RX_BUFFER_SIZE);

HoBingHong, 2018-12-25:
@ttlappalainen 
Excuse me, I am a beginner of arduino. Can I use my sonar and arduino uno to read depth and temperature?

ttlappalainen, 2018-12-25:
To read where? I do not think that it is possible. At least causes a some headace. There is some post under this issue on 2017-05-10. But after that library has got more features and you should also disable all feature under NMEA2000_CompilerDefns.h. Then you may run it so that you can print depth and temperature to Serial, since that library is anyway loaded. But if you e.g. think to add e.g. some LCD and library for that, I expect that you run UNOs 2k RAM out. If you need to stay on Arduino system, then I prefere Arduino Due as minimum. But also Teensy can be programmed under Arduino development environment by installing TeensyDuino for it. Teensy 3.2 is more powerfull than Due. And if you do not want to make traceiver connection by yourself, just order e.g. http://skpang.co.uk/catalog/teensy-canbus-breakout-board-include-teensy-32-p-1507.html

Note also that if you go to Teensy and install TeensyDuino, during installation unselect FlexCan library and use library under my git.

HoBingHong, 2019-01-14:
Hello, @ttlappalainen, I listened to your suggestion and bought Arduino due and MCP2562
I wire according to ArduinoDUE_CAN_with_MCP2562.pdf in your NMEA2000-master
When I run the sample program DataDisplay I get the following error message:
C:\Program Files\Arduino\libraries\NMEA2000_due-master\NMEA2000_due.cpp: In member function 'virtual bool tNMEA2000_due::CANOpen()':

C:\Program Files\Arduino\libraries\NMEA2000_due-master\NMEA2000_due.cpp:69:8: error: 'MaxCANReceiveFrames' was not declared in this scope

   If ( MaxCANReceiveFrames==0 ) MaxCANReceiveFrames=32; // Use default, if not set

        ^

C:\Program Files\Arduino\libraries\NMEA2000_due-master\NMEA2000_due.cpp:70:8: error: 'MaxCANReceiveFrames' was not declared in this scope

   If ( MaxCANReceiveFrames<10 ) MaxCANReceiveFrames=10; // Do not allow less that 10 - DUE should have enough memory.

        ^

C:\Program Files\Arduino\libraries\NMEA2000_due-master\NMEA2000_due.cpp:71:24: error: 'MaxCANReceiveFrames' was not declared in this scope

   Can0.setRxBufferSize(MaxCANReceiveFrames);

                        ^

C:\Program Files\Arduino\libraries\NMEA2000_due-master\NMEA2000_due.cpp: In member function 'void tNMEA2000_due::InitCANFrameBuffers()':

C:\Program Files\Arduino\libraries\NMEA2000_due-master\NMEA2000_due.cpp:132:3: error: 'InitCANFrameBuffers' is not a member of 'tNMEA2000'

   tNMEA2000::InitCANFrameBuffers(); // call main initialization

I am a beginner in this regard. Would you please give me your email for discussion?

ttlappalainen, 2019-01-14:
I have to update succestion. Currently I prefere Teensy 3.2 and up. DUE and Mega are also OK, but Teensy is more powerfull and draws less current - important for sailors.

There must be some  mix with the libraries. Arduino have different base for libraries. Libraries under C:\Program Files\Arduino\library has been provided with Arduino environment. Your own libraries are normally located under "C:\Users\<username>\Documents\Arduino\libraries". So I prefer to remove all NMEA2000* paths from C:\Program Files\Arduino\library. Remove also due_can* and CAN_BUS*. Then try again to load libraries as in NMEA2000_library_reference.pdf chapter 4.1 for mega and due. Check also that you not have duplicates of related libraries under "C:\Users\<username>\Documents\Arduino\libraries" with other name.

If you have libraries other than from my Git, they may be older versions and can not be mixed.

You can find my email by digging a bit, but I will not publish it here to avoid spamming.

HoBingHong, 2019-01-16:
Hello, Timo, I have successfully burned the DataDisplay into Arduino due but the sequence only shows Initialize buffers
But it does not show the depth and temperature of the sonar. I am in accordance with the ArduinoDUE_CAN_with_MCP2562.pdf in the NMEA2000-master. What is the reason? Also, where is the red (Vcc) and black (Gnd) of the NMEA2000 bus connected to the picture?

closed 2024-01-21 by ttlappalainen