ttlappalainen/NMEA2000 #21: Compilation warnings : might be important.

OzOns, 2016-08-31:
hi Timo,

First of all thank you a lot for the great job. I'am studdying it.

I embedded with succes the librairy in my project. For test purpose.

I detect somme warnings in compilation process. 

Like this one :

Arduino\libraries\NMEA2000-master\ActisenseReader.cpp:71:18: warning: comparison is always false due to limited range of data type [-Wtype-limits]

    if (MsgBuf[12]>tN2kMsg::MaxDataLen) {

Might be important. You can check this, if you set in IDE pref/compilation warning : "**show all**"

Regards and thumb up again :-).

Philippe

ttlappalainen, 2016-09-10:
Hi Philippe,

Sorry for delay - too muuch work on paid job.

Thanks for note. For some reason I got the warning only for Mega compilation. And you were right, it was rather critical. Now I fixed it and also fixed some other warnings. I just need to test the code before publishing, but I hope to get it tested withing few days.

Timo

ttlappalainen, 2016-09-12:
Hi,

Now there are some fixes.

OzOns, 2016-09-12:
Hi Timo,

Thks a lot for your kind answer

I did a quick check. Maybe too quick..

\Arduino\libraries\NMEA2000_mcp-master/NMEA2000_mcp.h: In member function 'void tNMEA2000_mcp::SetSPI(SPIClass*)':

s\Arduino\libraries\NMEA2000_mcp-master/NMEA2000_mcp.h:60:43: error: 'class MCP_CAN' has no member named 'setSPI'

     void SetSPI(SPIClass *_pSPI) { N2kCAN.setSPI(_pSPI); }

...

Maybe I missed something.

Cheers

Philippe

ttlappalainen, 2016-09-12:
Hi,

No I missed. I had added possibility to choose SPI. Someone should have tested this, but seem that it has never gone further. So now I forked CAN_BUS_Shield, which has those changes. Please download that.

OzOns, 2016-09-12:
Sorry Timo
I m not used to github. I can't find
Can you tell me the right link plz  to the new fork of CAN_BUS_Shield ?

ttlappalainen, 2016-09-12:
https://github.com/ttlappalainen/CAN_BUS_Shield

OzOns, 2016-09-12:
Thks  alot. all is ok now :-)

ttlappalainen, 2016-09-12:
Thanks for you.  I forgot the important rule that projects should compile with all warnings on without any warnings, since there may be critical warning hiding between non important warnings as in this case.

closed 2016-09-12 by ttlappalainen