ttlappalainen/NMEA2000 #23: Understanding N2K device selection

usauerbrey, 2016-09-19:
Hi Timo

hopefully you can help me in unstanding the behaivior of N2K devices. How selects a receiving device the "correct" source, if more than one device on the bus are sendind the same type of information. Is there a selection mechanism or so?

Best Regards Uwe

ttlappalainen, 2016-09-19:
Hi,

If you have my library devices, ensure that they have different Unique number (set by SetDeviceInformation) and different Model serial code (set by SetProductInformation). Then MFD should show these as separate device so that you can define, what source to use. My Garmin GMI 20 can do that. But unfortunately this depends also MFD.

You can also try to use different instance number on different devices.

Timo

ttlappalainen, 2016-09-19:
Now there is a new version, which hopefully is much better.

usauerbrey, 2016-09-20:
Hi Timo

I have the following roblem. My log (wheel) gets dirty often and then it gives a STW of 0 kn. This is not a problem as such, I have my SOG an this is ok. But the True Wind is calculated by the wind instrument using the STW and I cannot set it to use SOG instead. So I would like to wirte a program for the libray device, which reads STW and SOG and if STW is 0 but SOG not, I would send a STW-information with the SOG value. But how can I make the wind instrument take my STW value and not the value of 0 from the log. The MFD (Raymarine does not give me the seection of sources) at all.

So my question is, how does the N2K standard defines which souce is used. Or is it just up to the manufacturer of a device which source it takes?

Best Regards Uwe

ttlappalainen, 2016-09-21:
Sorry for delay,

I do not know is there any stndard for that. I think some devices just use PGN information. This is very problematic, if you have e.g. two heading sensor having 1 degree difference, so value will be jumping all the time. I had this case with my autopilot so I had to leave my spare sensor disconnected.

How are you devices constructed? Is your log N2k or NMEA0183 device? The point is that could you connect the log directly to your own device and make that as gateway? If the log is N2k log, then you would need to use Arduino due or add second cab bus to the system.

Timo

usauerbrey, 2016-09-21:
Hi Timo

What delay, you are really fast  ;-)
The log device is n2k, but your proposal making a gateway sounds interesting, even so I had expected, that there is some kind of selection mechanism on the bus. But maybe there is one, but it is not really known.

Best Regards Uwe

ttlappalainen, 2016-09-21:
Hi,

The selection mechnism exist. Each device has unique ID and they will get some source number on bus. So the device, which should select data should record unique ID, which you select on configuration. Then they can send ISO address claim query and see that device source number. Then they will use that information. Garmin GMI20 seem to work like that. I think I tested system by feeding cabin temperature from two sources and I could select, which one to show. If Raymarine does not offer that, then you can not do much.

So in principle you could have gateway, where all sensors are on one N2k bus and others on other bus. Then in your code you just pass all traffic directly from one bus to other except STW, which you check and replace with SOG in case of problem with log information.

I think it would be cheaper to buy a new log than spend several hours writing reliable code, but of coarse it is not so fun. Like if I convert the time I have spend with NMEA2000 library to money, I could have bought several converters and MFD:s etc.

usauerbrey, 2016-09-22:
Hi Timo

concerning the fun, I absolutely agree with you. 
But in this case, it is not a question of buying a new log. The log is relatively new and the situation of blocking the wheel by seagrass, small mussels, or so come regularly.

Best Regards Uwe

ttlappalainen, 2016-09-22:
OK. I think I need to do some development for library to check that there is no restriction to have two tNMEA2000 objects. There should not, but I have not tested. Also I need to check that all messages, which will not be handled, will be passed to gateway handler.

usauerbrey, 2016-09-22:
Hi Timo

It is not urgent. First I have to get the HW. In summer I live on my boat, so it it difficut to get the HW. I think this will be a project for the winter, when I live in germany. So take your time  ;-)

Best Regards Uwe

ttlappalainen, 2016-09-22:
Hi,

Within 3 weeks I have to take my boat up for winter, so I can take my Arduino Due for test bed. It has two can internally, so I can make ready gw code.

Timo

usauerbrey, 2016-09-24:
Hi, I am in the mediteranian, I will be sailing till mid of Nov. ;-)

fred2222, 2016-11-29:
Hi,
I have Arduino due with nmea2000 working however I am not able to have two Address Claims on one Arduino program for two devices. Anybody can give me a hint how to send two Address Claims to the NMEA2000 network ?

Thanks

Fred

ttlappalainen, 2016-11-29:
Hi,
I am sorry I di not understood your question? Normally address claiming will be done automatically by library. As in examples in setup you only need to define your node type and optional default address:
NMEA2000.SetMode(tNMEA2000::N2km_NodeOnly,22);
Then in your loop you call periodically:
 NMEA2000.ParseMessages();
With those library will automatically respond to ISO Address requests and do address claiming if necessary.

Or are you thinking about having two devices - e.g. temperature monitor and wind monitor with different device information - on one same program? If yes, you currently can not. I have option for that, it is still under construction, since it is very rarely needed.

fred2222, 2016-11-29:
Hi Timo,

Thank you for the very fast reply !
Yes I want to have two devices : speed sensor and engine tilt on one same program.
I want to have two devices in one Arduino due program that is connected to the n2k network.
Currently those two devices are on the network and work OK but I can only claim one address.
Sorry for the confusion. Newbie here with little programming knowledge.

Fred

ttlappalainen, 2016-11-30:
Hi,

Still question is that do you actually need two different device address? On the bus sight it is only important that each device providing any type of data has unic ID, created from device information. Then they would work right on bus. Other devices as far as I have experienced are not much interested about device class and device function codes, which in principle defines the type of device. So even you define your device as class 60 (Navigation) and function 135 (Depth/Speed), you can still send tilt information and MFD:s should show both. So currently having two devices on same code is more just informative.

I'll do the mutidevice code readey on some day, but you have to wait some weeks for that.

fred2222, 2016-11-30:
Hi Timo,

The Garmin GPSMap 740 does not see the device if the Address Claim is not sent once. Then it cannot be configured. The data is on the bus. The Arduino program always does the Address Claim for one device automatic so the Garmin only sees this one and the other cannot be configured. Both devices are sending the device class and function codes when I look with NMEA reader on the bus,
There are 2 Tilt devices for 2 engines and they need to be configured to the engines.
No problem if it takes several weeks. The holiday season is here..... Thanks for looking at it.

Fred

ttlappalainen, 2016-11-30:
I think you only mean that address claiming should be resent. Normally devices should request ISO address claim from other devices. We have noticed that some devices expect that device will send it periodically. So try this:
void SendIsoAddressClaim()
{
  static unsigned long LastSent=millis();

   if ( LastSent+2000<millis() ) {
     LastSent=millist();
     NMEA2000.SendIsoAddressClaim();
   }
}
...
void loop() 
{
 ...
  NMEA2000.ParseMessages();
  SendIsoAddressClaim();
}

fred2222, 2016-11-30:
Hi Timo,

Thanks for the tip to resend the Address Claiming on a regular basis. This is not what my problem is.
However it does solve the common problem if some devices need a regular claiming.
( for those that want to test it, just remove the "t" in millist !)
My problem only needs Address Claiming once for each device at the start of each device.
I found however a quick and dirty solution if I send the Address Claiming for the second device later.
I need to clean up my code and then it should be OK.
Your solution would also help to delay the Address Claiming at the start of each device. This often is needed if devices do switch on before the bus is active.
So  thank you for the tip. I hope others can use it too. 
I intend to look at the bus in short time to look at the traffic as I still need to understand what devices are sending. 
Wish you a good holiday season !

Fred

ttlappalainen, 2016-11-30:
In principle it should not matter when bus becomes active. Lets say you turn on one device, so it will do address claiming and start to send data to the bus. Then you turn on other and it does same. If the addresses are same, then one of those devices may reclaim its address. Next you turn on MFD. MFD should send ISO Address request for PGN 60928 (=ISO Address claim) with broadcast to find out what devices are on the bus. Other devices would again respond to this with their information. Normally MFD would then send ISO request for 126992 (Product information) for each device one by one. So now it knows alldevices on the bus. OK, then you turn on new device. It will again send address claiming. Now again MFD should then see this and query product information for new device. Also at least my GMI 20 MFD sends address claiming priodically (e.g. 20 sec) to check that which devices are still available on the bus. If some device does not respond, it drops it out from the list. So some devices does not do not send periodical request. Instead they expect others to send address claim in every 2-5 sec.

fred2222, 2016-11-30:
You are right. But I noticed the Garmin GPSMAP740 lost two engine devices on the bus and did not reclaim them  even after a long period. Those devices have fuel consumption information and so the Garmin did not update. They are removed from the NMEA2000 list of devices. There must be a protocol problem.
This is one of the problems I would like to see when watching the bus traffic.

Thanks Timo your expertise is of great help.

Fred

ttlappalainen, 2016-11-30:
Sorry but I am still curious about your construction. So do you have those two engine devices on same Arduino? And speed also? So are we actually talking about 2 engine instances + speed, which can be send from one N2k bus device?

fred2222, 2016-11-30:
Yes exactly. In fact there are 2 engine devices not on the Arduino that provide fuel and RPM and on the Arduino are just two engine Tilt devices + paddle speed sensor ( waterspeed ). I would include some temperature sensors maybe later.  If it is not time critical you could include a lot more devices. As the engine fuel and RPM devices are in the back of the boat it makes no sense to include them in the Arduino as that box is near the helm
The engine Tilt use the ADC's on the Arduino.

ttlappalainen, 2016-11-30:
OK. But still I do not see the real requirement for address claiming for two devices. You just do periodically e.g. 100 ms:
NMEA2000.SetN2kEngineParamRapid(N2kMsg,0,N2kDoubleNA,N2kDoubleNA,Engine0Tilt);
NMEA2000.SendMsg(N2kMsg);
NMEA2000.SetN2kEngineParamRapid(N2kMsg,1,N2kDoubleNA,N2kDoubleNA,Engine1Tilt);
NMEA2000.SendMsg(N2kMsg);

and also speed with 1000 ms period. Also you may need to send ISO Address claim 2000 ms period as discusssed earlier. 

MFD should be happy to see your system as one device, which just send data from different sources. As far as I know there is no limitation on NMEA2000 how much different type of data one device can send.

fred2222, 2016-11-30:
Yes, I would like to try this. But there is no need to send Address Claim periodically. 
Once the Garmin has seen the device it can be configured and I do not think it looses the Address like the external engine interfaces do. However full test and looking at the bus with NMEAreader I need to wait until my friend is back after Christmas.
With two devices I mean not the two engine Tilt devices but two different devices like speed and engine Tilt.  The two enigine Tilt devices are working OK when I can configure them after they claimed address. They do not need to be configured as separate devices. So once they have been seen by the Garmin there is no problem.

dhopf, 2022-03-18:
> OK. I think I need to do some development for library to check that there is no restriction to have two tNMEA2000 objects. There should not, but I have not tested. Also I need to check that all messages, which will not be handled, will be passed to gateway handler.

I thought I'd raise this ancient discussion from the depths of the Github servers instead of creating a new issue since it exactly fits my use-case:

### The scenario
I'm having the following simplified starting scenario:
`[KUS NS-5 fluid sensor (Diesel)] <-------> [NMEA2k bus] <------> [SIMRAD Plotter]`

My issue is that due to a special construction where and how the fluid sensor is attached to the tank, it always reports with some offset (e.g. tank in reality has 15% remaining while reporting 24%).

So my approach is to create a Gateway which simply looks for PGN127505 messages from the fluid sensor side on a first CAN interface, corrects the reported fluid level and sends the message on to the actual bus (out on a second CAN interface). I slightly patched your NMEA- and teensyx-libs for that to support two devices.

The setup now looks like:
`[KUS NS-5 fluid sensor (Diesel)] <-------> [(CAN1) MY GATEWAY DEVICE (CAN2)] <-----> [NMEA2k bus] <------> [SIMRAD Plotter]`
BTW: I took care of the "second bus" I now opened towards the fluid sensor (termination, power supply) - I created a custom PCB/circuitry.

### The issue
Due to lack in understanding NMEA2000, I first **did NOT** forward any other communication between the fluid sensor and the bus, I thought I just need to manipulate the actual PGN127505 messages.
The SIMRAD then reported some unknown network device with fuel level information, but couldn't reliably list it nor detect it when the SIMRAD plotters were powered off and back on.
From the discussion here I now get the understanding that the Fluid sensor likely sends ISO address claims onto the bus and the SIMRAD plotters somehow either acknowledge those (or they actually don't send any reply to this, but use the information from the ISO address claim to list the fluid sensor correctly?)

### The solution?
So, I'm now figuring that I need to set my Gateway device into a mode where (in addition to manipulating the PGN127505 messages) I simply forward all other messages from the fluid sensor onto the bus and - if not even more important - I have to forward messages from the bus back to the fluid sensor (maybe I could find out which ones specifically, but I can simply gateway everything I guess). For this, I now set both of my tNMEA2000_Teensyx objects to tNMEA2000::N2km_ListenAndSend and added code to forward all messages besides the one I'm manipulating.

@ttlappalainen: I would really appreciate if you could confirm or correct my understanding/approach. Also: Did you ever continue investigation in supporting instantiating multiple tNMEA2000 objects and maybe adding example code for a simple "Gateway" functionality?

Thanks & BR
Daniel

ttlappalainen, 2022-03-19:
First: There is no limitation to have two instances of tNMEA2000.

Second: In principle in N2km_ListenAndSend mode it may work. You should relay all messages from Simrad side to sensor side and opposite and then modify only tank level message. Instead of using #include <NMEA2000_CAN.h> you do:
```
#include <N2kMsg.h>
#include <NMEA2000.h>
#include <NMEA2000_teexnsyx.h>
#include <N2kMessages.h>
...
tNMEA2000 &NMEA2000_sensor=*(new tNMEA2000_Teensyx(tNMEA2000_Teensyx::CAN1));
tNMEA2000 &NMEA2000_bus=*(new tNMEA2000_Teensyx(tNMEA2000_Teensyx::CAN2));
...

void SensorMsgHandler(const tN2kMsg &N2kMsg);
void BusMsgHandler(const tN2kMsg &N2kMsg);

// *****************************************************************************
void setup() {
  ...
  NMEA2000_sensor.SetN2kCANMsgBufSize(10); // Should be enough
  NMEA2000_bus.SetN2kCANMsgBufSize(10);
  NMEA2000_sensor.SetN2kCANReceiveFrameBufSize(100); // probably not even necessary. I think teensyx driver had big buffers as default
  NMEA2000_bus.SetN2kCANReceiveFrameBufSize(100);
  NMEA2000_sensor.SetMode(N2km_ListenAndSend,255);
  NMEA2000_bus.SetMode(N2km_ListenAndSend,255);
  NMEA2000_sensor.SetMsgHandler(SensorMsgHandler);
  NMEA2000_bus.SetMsgHandler(BusMsgHandler);
...
}

// *****************************************************************************
void loop() {
  NMEA2000_sensor.ParseMessages();
  NMEA2000_bus.ParseMessages();
}

// *****************************************************************************
double FixFluidLevel(double FluidLevel) {
  // do something with it
  return FluidLevel;
}

// *****************************************************************************
void SensorMsgHandler(const tN2kMsg &N2kMsg) {
  double FluidLevel,Capacity;
  unsigned char Instance;
  tN2kFluidType FluidType;

  if ( ParseN2kFluidLevel(N2kMsg,Instance,FluidType,FluidLevel, Capacity) ) {
    tN2kMsg  N2kMsgFwd(N2kMsg.Source);
    N2kMsgFwd.Destination=N2kMsg.Destination;
    SetN2kFluidLevel(N2kMsgFwd,Instance,FluidType,FixFluidLevel(FluidLevel), Capacity);
    NMEA2000_bus.SendMsg(N2kMsgFwd,-1); 
  } else {
    NMEA2000_bus.SendMsg(N2kMsg,-1);
  }
}

// *****************************************************************************
void BusMsgHandler(const tN2kMsg &N2kMsg) {
  NMEA2000_sensor.SendMsg(N2kMsg,-1);
}
```
Have not tested above code.

It should also work with single bus.

Solution 1. Just do normal device (like on example BatteryMonitor) with your own product and device information. This should show up on Simrad with your device information. Manufacturer will be unknown, if you do not use some predefined MfgCode. Then catch fluid level from your sensor, modify it and send it as your own device fluid message.

Solution 2. Spy your sensor product and device information. Increment unique ID with one and use those information for your device. Then do the same as above. Now your own device should be visible on SIMRAD as second fluid sensor and same manufacture as other.

Actually above solutions are bit more complex, since you have to follow your fluid sensor source address so that you handle right sensor. For that you need to know your sensor NAME, which is 8 byte data provided by ISO address claim message. So you need to catch also ISO address claim message (60928) and if its data matches to your sensor NAME, you take source address and listen 127505 messages from that source. Library N2kDeviceList module does that for all devices and you can request source by name.

ttlappalainen, 2022-03-23:
OpenSkipper does not care of devices and so shows device data without knowing anything about devices..

If library device is in node mode, it does address claiming automatically and you do not need to care of that. If you do it with my gateway way, address claiming will be done by devices automatically since all messages will be transfered to both sides.

My NMEA Simulator simulates devices and you can analyze bus messages. Unfortunately free version parses only first 100 messages and message console has been limited to 5000 messages. You can find it from http://www.kave.fi/Apps/

ttlappalainen, 2025-01-03:
It is reserved bit and you can not set it to 0.

In your setting device class and device function would be 0. They do not have any definition and so not allowed.

Oscarwaterski, 2025-01-03:
So, the only way is to do a custom pgn ? 
thank you for your answer. 
the objective is to have the same as my old gps. (now broken :( ) 
[image]
can you help me in this way ?   I will use gps module be 880 this compass. 
here a custom pgn : 
[image]

ttlappalainen, 2025-01-04:
Why you should have PGN 65256. Do you have on bus some device, which reads it?

Oscarwaterski, 2025-01-04:
Hello,
I have a speed regulator using courses with gps waypoint 0.2km/h précision
how work with.
I have these information from the factory.
Thank you ;)

ttlappalainen, 2025-01-04:
Doesn't it read PGN 129026?

Oscarwaterski, 2025-01-04:
Yes, it read too.
The screen shot that you see is just the reading of the original gps
antenna.
So I need to send the same pgn as original.

ttlappalainen, 2025-01-04:
I understood that screenshot was what your old gps sends? But does your speed regulator work with PGN 129026? Then you do not need to send PGN 65256. And can you configure your speed regulator, which device it listhens. Then you do not need to worry about device network NAME.

Oscarwaterski, 2025-01-04:
Yes it is the old gps. according to the speed regulator I am not sure what
it’s working with, and I can’t check. Nevertheless I know it exists 2
different kinds of antenna wich are distinguished by the PGN Adress Claim.
The system only works if the address claim is verified.
The system is linked to the dashboard and to the nav information
I guess the PGN 65256 is used for exact velocity measurement?
Would you be okay to help me finalizing the code in exchange of money ?
Thank you so much.
I can’t afford to give you much but steel it seems normal to pay for your
help.
The original gps antenna cost 1000€ and it is really interested to learn
how to do it.
Thank you so much.

Oscarwaterski, 2025-01-10:
Hi Timo,
have you seen my last email?
Can you help me with this code ? :
#include <Wire.h>
#include <NMEA2000_CAN.h>   // Bibliothèque NMEA2000 pour ESP32
#include <NMEA2000.h>       // Bibliothèque principale NMEA2000
#include <TinyGPS++.h>      // Pour module GPS BE-880
#include <Adafruit_Sensor.h>
#include <QMC5883LCompass.h> // Pour boussole QMC5883L
#include <Arduino.h>
#include <N2kMessages.h>
#include <N2kMaretron.h>

using tN2kSendFunction=void (*)();

// Structure for holding message sending information
struct tN2kSendMessage {
  tN2kSendFunction SendFunction;
  const char *const Description;
  tN2kSyncScheduler Scheduler;

  tN2kSendMessage(tN2kSendFunction _SendFunction, const char *const
_Description, uint32_t /* _NextTime */
                  ,uint32_t _Period, uint32_t _Offset, bool _Enabled) :
                    SendFunction(_SendFunction),
                    Description(_Description),
                    Scheduler(_Enabled,_Period,_Offset) {}
  void Enable(bool state);
};

extern tN2kSendMessage N2kSendMessages[];
extern size_t nN2kSendMessages;

static unsigned long N2kMsgSentCount=0;
static unsigned long N2kMsgFailCount=0;
static bool ShowSentMessages=false;
static bool ShowStatistics=false;
static bool Sending=false;
static bool EnableForward=false;
static tN2kScheduler NextStatsTime;
static unsigned long StatisticsPeriod=2000;

// Forward declarations for functions
void CheckCommand();
void CheckLoopTime();
void OnN2kOpen();

// Déclaration des capteurs
TinyGPSPlus gps;             // GPS BE-880
QMC5883LCompass compass;     // QMC5883L

// Pins CAN Bus
#define ESP32_CAN_TX_PIN GPIO_NUM_16 // If you use ESP32 and do not have TX
on default IO 16, uncomment this and and modify definition to match your
CAN TX pin.
#define ESP32_CAN_RX_PIN GPIO_NUM_17 // If you use ESP32 and do not have RX
on default IO 4, uncomment this and and modify definition to match your CAN
TX pin.

void setup() {
  Serial.begin(115200);      // Debug via Serial
  delay(2000); // Give a bit time for console to be ready

  // Uncomment 3 rows below to see, what device will send to bus

   NMEA2000.SetForwardStream(&Serial);  // PC output to default serial port
   NMEA2000.SetForwardType(tNMEA2000::fwdt_Text); // Show in clear text.
Leave uncommented for default Actisense format.
   NMEA2000.SetForwardOwnMessages(true); // Do not print own messages.

     // We act as real node on bus. Some devices does not show messages, if
they can not request information.
  NMEA2000.SetMode(tNMEA2000::N2km_ListenAndNode,28);
  //NMEA2000.SetDebugMode(tNMEA2000::dm_ClearText); // Uncomment this, so
you can test code without CAN bus chips on Arduino Mega
  NMEA2000.EnableForward(EnableForward); // Disable all msg forwarding to
USB (=Serial)
  NMEA2000.SetOnOpen(OnN2kOpen);
  NMEA2000.Open();

  Serial.println("Starting message sender!");
  Serial.println("  Message sending to bus starts in 10 seconds.");
  Serial.println("  Type ?<lf> for available commands.");

  Serial2.begin(9600, SERIAL_8N1, 5, 4); // GPS via UART

  Wire.begin();              // Initialisation I2C

  // Initialisation de la boussole QMC5883L
  compass.init(); // Initialisation du capteur

  // Envoi initial du PGN 60928 (ISO Address Claim)
  tN2kMsg N2kMsg;
  N2kMsg.SetPGN(60928);
  N2kMsg.Priority = 6;
  N2kMsg.AddUInt64(0xC8C3440E000042); // Données spécifiques
  NMEA2000.SendMsg(N2kMsg);
}

void loop() {

    // Lecture des données GPS
    while (Serial2.available() > 0) gps.encode(Serial2.read());
    double lat = gps.location.lat();
    double lon = gps.location.lng();
    double speed = gps.speed.kmph();
    double altitude = gps.altitude.meters();

    // Lecture des données de la boussole
    int x, y, z;
    compass.read(); // Lecture brute
    x = compass.getX();
    y = compass.getY();
    z = compass.getZ();
    double bearing = atan2(y, x) * (180.0 / PI);
    if (bearing < 0) bearing += 360;

    // Envoi des PGNs
    sendPGN129025(lat, lon);
    sendPGN129026(bearing, speed);
    sendPGN129539();
    sendPGN126992();
    sendPGN65256(bearing, speed, altitude);
  }
}

// PGN 129025 : Position
void sendPGN129025(double lat, double lon) {
  tN2kMsg N2kMsg;
  N2kMsg.SetPGN(129025);
  N2kMsg.Priority = 2;
  N2kMsg.AddByte(0xFF); // Reserved
  N2kMsg.Add4ByteDouble(lat * 1e7, 1e7); // Latitude
  N2kMsg.Add4ByteDouble(lon * 1e7, 1e7); // Longitude
  NMEA2000.SendMsg(N2kMsg);
}

// PGN 129026 : COG & SOG
void sendPGN129026(double cog, double sog) {
  tN2kMsg N2kMsg;
  N2kMsg.SetPGN(129026);
  N2kMsg.Priority = 2;
  N2kMsg.AddByte(0xFF); // Reserved
  N2kMsg.Add2ByteDouble(cog, 0.0001); // Course over ground (degrees)
  N2kMsg.Add2ByteDouble(sog, 0.01);  // Speed over ground (knots)
  NMEA2000.SendMsg(N2kMsg);
}

// PGN 129539 : GNSS DOPs
void sendPGN129539() {
  tN2kMsg N2kMsg;
  N2kMsg.SetPGN(129539);
  N2kMsg.Priority = 3;
  N2kMsg.Add2ByteDouble(1.0, 0.01); // HDOP
  N2kMsg.Add2ByteDouble(1.0, 0.01); // VDOP
  N2kMsg.Add2ByteDouble(1.0, 0.01); // PDOP
  NMEA2000.SendMsg(N2kMsg);
}

// PGN 126992 : System Time
void sendPGN126992() {
  tN2kMsg N2kMsg;
  N2kMsg.SetPGN(126992);
  N2kMsg.Priority = 3;
  N2kMsg.Add4ByteUInt(millis()); // Time since start (ms)
  N2kMsg.Add2ByteUInt(0xFFFF);   // Date not available
  N2kMsg.Add2ByteUInt(0xFFFF);   // Reserved
  NMEA2000.SendMsg(N2kMsg);
}

// Fonction pour configurer et envoyer le PGN 65256
void SendPGN65256(float pitch, float altitude, float compassBearing, float
speed) {
  tN2kMsg N2kMsg;
  N2kMsg.SetPGN(65256);
  N2kMsg.Priority = 6;
  N2kMsg.Add2ByteInt(pitch * 100);               // Pitch (en centi-degrés)
  N2kMsg.Add3ByteInt(altitude * 100);            // Altitude (en
centimètres)
  N2kMsg.Add2ByteUInt(compassBearing * 100);     // Compass Bearing (en
centi-degrés)
  N2kMsg.Add2ByteUInt(speed * 100);              // Vitesse (en
centi-mètres/seconde)
  NMEA2000.SendMsg(N2kMsg);
}

thank you so much.

ttlappalainen, 2025-01-11:
I do not have much time to play with and my charge is 105 €/h.

You should not build existing messages from cratch. e.g. for 129026 use function from N2kMessages as in many examples.
```
void sendPGN129026(double cog, double sog) {
  tN2kMsg N2kMsg;
  SetN2kCOGSOGRapid(N2kMsg,N2kUInt8NA,cog,sog);
  NMEA2000.SendMsg(N2kMsg);
}
```
If you wnat to override 60928 sending from library, you need to comment `inline void SetN2kISOAddressClaim...` on NMEA2000.h and declare it instead as `extern void SetN2kISOAddressClaim...`. Then you can define in your own module:
void SetN2kISOAddressClaim(tN2kMsg &N2kMsg, unsigned long UniqueNumber, int ManufacturerCode,
                   unsigned char DeviceFunction, unsigned char DeviceClass,
                   unsigned char DeviceInstance=0, unsigned char SystemInstance=0, unsigned char IndustryGroup=4
                   ) {
   SetN2kPGN60928(N2kMsg,0xC8C3440E000042);
}

Oscarwaterski, 2025-01-11:
Thank you for your help,
105€/h sounds good. How many Time do you need to do it ?
Thanks a lot.

ttlappalainen, 2025-01-12:
What hw you tought to use? I can do tested code only for ESP32 of Teensy 4.x.

Please emal me directly.

Oscarwaterski, 2025-01-12:
Capture d’écran 2024-12-04 132938.png
<https://mail.google.com/mail/u/0?ui=2&ik=aec9c04f75&attid=0.1&permmsgid=msg-a:r-2122604954480245089&view=att&disp=safe&realattid=f_m5l64mip1&zw>
(33 Ko)
<https://mail.google.com/mail/u/0?ui=2&ik=aec9c04f75&attid=0.1&permmsgid=msg-a:r-2122604954480245089&view=att&disp=safe&realattid=f_m5l64mip1&zw>
Capture d’écran 2024-12-04 132938.png
<https://mail.google.com/mail/u/0?ui=2&ik=aec9c04f75&attid=0.1&permmsgid=msg-a:r-2122604954480245089&view=att&disp=safe&realattid=f_m5l64mip1&zw>
(33 Ko)
<https://mail.google.com/mail/u/0?ui=2&ik=aec9c04f75&attid=0.1&permmsgid=msg-a:r-2122604954480245089&view=att&disp=safe&realattid=f_m5l64mip1&zw>

Hello Timo,
First of all, thank you for your help, as I’m really going to need it.
Know that if you ever need help with 3D printing, CNC machining, laser
cutting, or part fabrication, I’d be happy to assist you too. :)

So!
As you’ve understood very well, the goal is to create an antenna using a
GPS module and an ESP32.

I’ve already configured the GPS module to NMEA0183 format with a baud rate
of 9600 bps.
I can display and interpret the GPS data in the IDE.

my code :

#include <Wire.h>
#include <NMEA2000_CAN.h>   // Bibliothèque NMEA2000 pour ESP32
#include <NMEA2000.h>       // Bibliothèque principale NMEA2000
#include <TinyGPS++.h>      // Pour module GPS BE-880
#include <Adafruit_Sensor.h>
#include <QMC5883LCompass.h> // Pour boussole QMC5883L
#include <Arduino.h>
#include <N2kMessages.h>
#include <N2kMaretron.h>

using tN2kSendFunction=void (*)();

// Structure for holding message sending information
struct tN2kSendMessage {
  tN2kSendFunction SendFunction;
  const char *const Description;
  tN2kSyncScheduler Scheduler;

  tN2kSendMessage(tN2kSendFunction _SendFunction, const char *const
_Description, uint32_t /* _NextTime */
                  ,uint32_t _Period, uint32_t _Offset, bool _Enabled) :
                    SendFunction(_SendFunction),
                    Description(_Description),
                    Scheduler(_Enabled,_Period,_Offset) {}
  void Enable(bool state);
};

extern tN2kSendMessage N2kSendMessages[];
extern size_t nN2kSendMessages;

static unsigned long N2kMsgSentCount=0;
static unsigned long N2kMsgFailCount=0;
static bool ShowSentMessages=false;
static bool ShowStatistics=false;
static bool Sending=false;
static bool EnableForward=false;
static tN2kScheduler NextStatsTime;
static unsigned long StatisticsPeriod=2000;

// Forward declarations for functions
void CheckCommand();
void CheckLoopTime();
void OnN2kOpen();

// Déclaration des capteurs
TinyGPSPlus gps;             // GPS BE-880
[... 172 lines cut ...]
  I’ve made an NMEA2000 reader to read and test my setup.
A friend lent me his antenna, and I was able to read the messages and
frequency.

[image: image.png]

Complications:

   1. *Two specific PGNs:* 65256 and 60928
      - *60928*: This is the machine address, but the library blocks the
      last two digits, so I need to create a custom PGN.
      - *65256*: A specific PGN with data sent in the following format:

[image: image.png]

I have a BE 880 GPS module with an integrated compass and an electronic IC
compass (QMC5883L).

Attached are the beginnings of the code...

I’ve made significant progress in my research. :)

Thanks again for your help.

Have a great day!

Oscarwaterski, 2025-02-17:
Hello Timo,
I managed to implement the complete code with all the PGNs. All I have to
do is resolve the problem of the address claim. you advised me I added
extern in front of the “inline void” line 3063 and line 3076.
[image: image.png]

When I redefine the function in my code it doesn't compile because it
explains that I am redefining the function.
Do I have to delete the product and set product info lines in the setup?
Do I have to define the function before setup to send the custom
ardessCLAIM all at once?
Thank you for your help.

ttlappalainen, 2025-02-17:
Actually does not work like I first mentioned. Just reset definition to original. Instead you can just bypass Deviceinformation setting on lines 1140-1146. At the end of that function force NAME you want to use:
void tNMEA2000::SetDeviceInformation(...) {
...
   if (_IndustryGroup!=0xff) Devices[iDev].DeviceInformation.SetIndustryGroup(_IndustryGroup);
  Devices[iDev].DeviceInformation.SetName(0x420000000E44C3C8);
}

Then library should use exactly that NAME.
If I read it right that NAME should split to
- Unique number: 312264 (0x4C3C8)
- Manufacturer:  114 (0x072) EControls, Inc 
- Device Instance Lower: 0
- Device Instance Upper: 0
- Device Function: 0
- Device Class: 0
- System Instance: 2
- Industry Group: 4 = Marine

Oscarwaterski, 2025-02-17:
Hi Timo,

I set a timing in the loop with an interval of 500ms, but the messages
are received with an interval that changes. close to 0.5s but still
0.57,0.58 how can I change this?

thank you

ttlappalainen, 2025-02-17:
That depends of your hw and sw. As I mentioned, you should not have any routine in your loop blocking loop for as single ms. So delay within loop or any function called inside loop is totally forbidden! If you e.g. read serial data from other port, you should not block reading. Instead you check is any bytes available and read only available bytes, save sate and exit your reader function. It is same with sending data to serial. You must aways check how much you can write to outbuffer. Check my NMEA0183 library. When other routines called in loop are properly designed variation is mostly <2 ms. That also depends of hw.

Oscarwaterski, 2025-02-17:
If I don’t P Put Any change in the loop, I’ve got 0.3 interval in my NMEA
2000 reader.
How can I set sending message every 0.5 seconds?
Thank you so much

ttlappalainen, 2025-02-17:
You do not seem to have syncronizing timer. You just send as fast as data is available. 

In your loop you have call
`while (Serial2.available() > 0) gps.encode(Serial2.read());`

That is forst problem. It most probably blocks loop until gps data is available and cause unexpected pause to loop.

`compass.read(); // Lecture brute`

That is second problem. No idea how it does reading and how long it spents time for it.

You should use N2kMessages functions for building up messages and not copy paste code from them. And also you should have timer for each message sender. Note that each message has different defined period, which you should use. Periods are documented in documentation or in NMEA2000.cpp file. E.g., **TemperatureMonitor example** shows how to use tN2kSyncScheduler timers. For data values like lat,lon you have to use globals and you need to rewrite gps reade and compass reader as non blocking functions.
```
// Define schedulers somewhere at beginning.
tN2kSyncScheduler PGN129025Scheduler(false,100,500);
tN2kSyncScheduler PGN129026Scheduler(false,250,510);
tN2kSyncScheduler PGN129539Scheduler(false,1000,520);
...
// Start them over in callback, which you have define in setup()
void OnN2kOpen() {
  // Start schedulers now.
  PGN129025Scheduler.UpdateNextTime();
  PGN129026Scheduler.UpdateNextTime();
  PGN129539Scheduler.UpdateNextTime();
...
}
...
void SendMessages() {
  if ( PGN129025Scheduler.IsTime() ) {
    tN2kMsg N2kMsg;
    SetN2kLatLonRapid(N2kMsg,Latitude,Longitude);
    NMEA2000.SendMsg(N2kMsg);
  }
  if ( PGN129026Scheduler.IsTime() ) {
  }
}

...
void loop() {
  ReadGPS();
  ReadCompass();
  SendMessages();
}
```

Oscarwaterski, 2025-02-23:
Hi Timo,

Thank you a lot for helping me?
I changed all my code to do it as you suggested. with periode at O I got à
0,2s periode out.
When i put 500 for the period, i got 0,48 ; 0,51 but not always 0,5. Why ?
Other question , I is possible to run the library on ESPMINI or ESP32 C3 ?

Thank You:)

ttlappalainen, 2025-02-23:
I expect you still have problem with loop timing. Modify your loop with code
```
void loop() {
  static unsigned long LongestLoop=0;
  static unsigned long ShortestLoop=10000000UL;
  static unsigned long NextTimingUpdate=millis()+2000;
  static unsigned long LoopCount=0;
  static unsigned long LoopAvgSum=0;
  unsigned long LoopStart=micros();

... your code here

  unsigned long LoopElapsed=micros()-LoopStart;
  LoopCount++;
  LoopAvgSum+=LoopElapsed;
  if ( LongestLoop<LoopElapsed ) {
    LongestLoop=LoopElapsed;
  }
  if ( ShortestLoop>LoopElapsed ) {
    ShortestLoop=LoopElapsed;
  }
  if ( NextTimingUpdate<millis() ) {
    NextTimingUpdate=millis()+2000;
    Serial.printf("Loop statistics:"
                  " Avg:%6.3f"
                  ", Count:%5lu"
                  ", Longest:%7.3f"
                  ", Shortest:%5.3f\n"
                  ,((double)LoopAvgSum)/LoopCount/1000
                  ,LoopCount
                  ,((double)LongestLoop/1000)
                  ,((double)ShortestLoop/1000));
    LoopCount=0;
    LoopAvgSum=0;
    LongestLoop=0;
    ShortestLoop=10000000UL;
  }
}
```

And check what you get as result. You should get Avg: <1 and Longest: <50

Oscarwaterski, 2025-02-23:
Thank you so much. I will try it.
About the esp32 C3 ?
It is possible to run the library with the smallest word than the esp32 ?
Thank you

ttlappalainen, 2025-02-23:
Why? How many € you will save?

There are some implementationf for S3 library, but I have not tested or done certification tests for them.

Oscarwaterski, 2025-02-23:
Just to do the system smaller ;)

Oscarwaterski, 2025-02-26:
do you think it will work with S3 ?

ttlappalainen, 2025-02-26:
As I mentioned there are some implementations for S3 library, but I have not tested or done certification tests for them.