ttlappalainen/NMEA2000 #25: Problem with pgn 127508 (battery)

sarfata, 2016-10-16:
I use the following code to send battery info to the network:
```
SetN2kPGN127508(msg, 0, 13.8, 0.8, N2kDoubleNA, sid++);
```

When converted in $PCDIN format, this is what it looks like:
```
$PCDIN,01F214,00000000,16,0064050800FFFF0C*5B
```

Which canalyzer translates as:
```
1969-12-31-16:00:00,0 0  22 255 127508 Battery Status:  Battery Instance = 0; Voltage = 13.80 V; Current = 0.8 A; SID = 12
```

However, the Raymarine i70 MFD refuses to display any battery information.

I now have access to iKommunicate and this is what it reports:
```
{
	"context": "vessels.urn:mrn:imo:mmsi:0",
	"updates": [{
		"source": {
			"type": "NMEA2000",
			"src": "1-22",
			"pgn": 127508,
			"label": "sources.N2000-01.022"
		},
		"timestamp": "2009-12-32T00:19:44Z",
		"values": [{
			"path": "electrical.batteries.8.voltage",
			"value": 13.799999
		}, {
			"path": "electrical.batteries.8.current",
			"value": 0.800000
		}]
	}]
}
```

Battery instance is always reported as 8. I have done tests with different values of `BatteryInstance` and always get the same result.

This could be a bug in iKommunicate but since the Raymarine also refused to display this battery voltage, I wonder if maybe there is a bug in the way `NMEA2000` encodes this message.

ttlappalainen, 2016-10-17:
I tested you message and it works right with NMEA Reader. I used library example MessageSender, which sends battery information for 2 banks (0,1) and NMEA Reader shows both of them right. I'll verify code with Garmin later, but now it is on boat. Have you tried to set some data  - like CToKelvin(27.15) - to the temp?

usauerbrey, 2016-10-17:
Hi
Just to add a comment concerning i70: 
My i70 (V3.05) displays the correct values of V, A, Temp, Charge. Battery Instance=0 is shown if I select Battery 1 and Battery Instance=1 shows the values on Battery 2 of i70.
Best Regards Uwe

sarfata, 2016-10-17:
> My i70 (V3.05) displays the correct values of V, A, Temp, Charge. Battery Instance=0 is shown if I select Battery 1 and Battery Instance=1 shows the values on Battery 2 of i70.

Interesting, I have v2.18. Maybe it is a i70 bug. I will try to find a way to upgrade it. This probably will not be easy.

> Have you tried to set some data - like CToKelvin(27.15) - to the temp? 

Yes. With and without current, with and without temperature. This does not make a difference on the i70 (does not work) but iNavX will only show the battery voltage if I provide a current different than `DoubleNA`.

closed 2016-10-25 by sarfata