ttlappalainen/NMEA2000 #27: Not responding to PNG 59904 Request for PNG 126996

Loafdude, 2016-10-23:
The library should automatically respond to  a request for product information (PNG 126996) but it does not appear to do so

I am using two Teensy 3.1 and a Lowrance HDS7 Gen1. I view through the actisense software to monitor the bus.

The library properly responds to 60928 with another 60928.
My MFD then requests product information with PNG 59904 but the library does not respond

What would the proper way to respond in my code be?

ttlappalainen, 2016-10-23:
It should and it does for my Garmin. 

Could you record situation with Actisense Reader and email .ebl file to me. Also your .ino may help.

Loafdude, 2016-10-23:
Here is the sketch, 

`// Demo: NMEA2000 library. Send main wind data to the bus.

#include <Arduino.h>
#include <NMEA2000_CAN.h>  // This will automatically choose right CAN library and create suitable NMEA2000 object
#include <N2kMessages.h>

void setup() {
  // Set Product information
  NMEA2000.SetProductInformation("000001111", // Manufacturer's Model serial code
                                 1000, // Manufacturer's product code
                                 "Simple wind monitor",  // Manufacturer's Model ID
                                 "1.0.0.11 (2015-11-10)",  // Manufacturer's Software version code
                                 "1.0.0.0 (2015-11-10)" // Manufacturer's Model version
                                 );
  // Det device information
  NMEA2000.SetDeviceInformation(1001, // Unique number. Use e.g. Serial number.
                                130, // Device function=Atmospheric. See codes on http://www.nmea.org/Assets/20120726%20nmea%202000%20class%20&%20function%20codes%20v%202.00.pdf
                                55, // Device class=External Environment. See codes on  http://www.nmea.org/Assets/20120726%20nmea%202000%20class%20&%20function%20codes%20v%202.00.pdf
                                2046 // Just choosen free from code list on http://www.nmea.org/Assets/20121020%20nmea%202000%20registration%20list.pdf                               
                               );
  // Uncomment 3 rows below to see, what device will send to bus                           
  Serial.begin(115200);
  // NMEA2000.SetForwardType(tNMEA2000::fwdt_Text); // Show in clear text. Leave uncommented for default Actisense format.
  // NMEA2000.SetForwardOwnMessages();

  // If you also want to see all traffic on the bus use N2km_ListenAndNode instead of N2km_NodeOnly below
  NMEA2000.SetMode(tNMEA2000::N2km_NodeOnly,75);
  // NMEA2000.SetDebugMode(tNMEA2000::dm_Actisense); // Uncomment this, so you can test code without CAN bus chips on Arduino Mega
  NMEA2000.EnableForward(false);
  NMEA2000.Open();
}

void loop() {
  SendN2kWind();
  NMEA2000.ParseMessages();
}

double ReadWindAngle() {
  return DegToRad(50); // Read here the measured wind angle e.g. from analog input
}

double ReadWindSpeed() {
  return 10.3; // Read here the wind speed e.g. from analog input
}

#define WindUpdatePeriod 1000

void SendN2kWind() {
  static unsigned long WindUpdated=millis();
  tN2kMsg N2kMsg;

  if ( WindUpdated+WindUpdatePeriod<millis() ) {
    SetN2kWindSpeed(N2kMsg, 1, ReadWindSpeed(), ReadWindAngle(),N2kWind_Apprent);
    WindUpdated=millis();
    NMEA2000.SendMsg(N2kMsg);
  }
}
`Two different captures are attached.
[Data Capture.zip](https://github.com/ttlappalainen/NMEA2000/files/546290/Data.Capture.zip)

I don't understand much of the codebase yet.
But if I delete lines 749 and 751 in NMEA2000.cpp 

`      case 126996L: /* Product information */
        // If query was for us, try to respond immediately
        if ( ( N2kMsg.Destination!=DeviceInformation[iDev].N2kSource ) || !SendProductInformation(iDev) ) {
          DeviceInformation[iDev].SetPendingProductInformation();
        }
        break;`
to

`      case 126996L: /* Product information */
        // If query was for us, try to respond immediately
           DeviceInformation[iDev].SetPendingProductInformation();
        break;`

The library then replies but the message is empty (I have attached a capture of that scenario.
The Lowrance HDS7 then shows blank information so I can confirm this is the message my MFD is looking for.

ttlappalainen, 2016-10-23:
Hi,

I tested your code and it works fine with Garmin GMI20. GMI20 shows "Simple win monitor" on device list. Are you sure you have proper bus termination resistors connected?  I had a some similar effects without resistors and received only short messages, but long messages like product information did not get through.

Loafdude, 2016-10-23:
Bus is definitely terminated, 60ohms measured

All other messages are appearing fine. For example MessageSender example is working fine. All data appears on the MFD.

The reply is not sent at all. Like I say, if i edit RespondISORequest I can get it to reply but with no data.

Strange it works for you. Are you using a teensy as well?

ttlappalainen, 2016-10-23:
Yes, I have Teensy 3.2. And I did not change the code. I tested it with both broadcasted request and addressed request, which Garmin uses.

Loafdude, 2016-10-23:
If/when you have time, can you paste the 59904 packet that requests pgn 126996 from your garmin. I'll try and replay it and see what I've got going on here.

Thanks for your time and trouble!

Loafdude, 2016-10-23:
I modified my sketch to N2km_ListenAndNode mode and it appears your code is sending the correct packet but both my MFD and 2nd teensy are unable to read it.
Somewhere it is lost/corrupted along the way

Thanks for the tips and help!
I'll start looking into my hardware :/

ttlappalainen, 2016-10-24:
What line driver you use on Teensy - MCP2562? Do you have terminators on both end of main line - not in middle? I understood you have two Teensy. Have you tried to switch them?

Please inform any solution you find. It may help others.

Loafdude, 2016-10-24:
Yes I have two teensy's and two line drivers (and one MFD)

I'm using MCP2551. It is a 5V part but is known to be working according to teensy forums (it will accept 3.3v logic and teensy is 5v tolerant so no problem)

I have swapped my line drivers and swapped my teensy's. I have tried with 1 teensy + MFD and just two teensy's. 

I scoped comparing teensy TX to CanH/CanL and they are in sync
I scoped CanH/CanL and they are perfectly inverted
I scoped CanH/CanL comparing to RX on MCP2551 and they are also in sync

Grrr!
I'm a bit at a loss at the moment.

ttlappalainen, 2016-10-24:
Do you run MCP2551 with 5V power? This is one difference - I have MCP2562 with 5V line power and 3.3V logic. It has own logic level selection but still drives bus with 5V power.

Loafdude, 2016-10-25:
Yes I feed it 5V Vcc and 3.3v logic.

But.... Solved!

I was using sarfata NMEA2000_teensy branch which is broken (You have already created a merge request, I just posted requesting he merge it)

For others you MUST use 
https://github.com/ttlappalainen/NMEA2000_teensy
when using a teensy. A typo in sarfata's code causes bad data to be sent!

ttlappalainen, 2016-10-25:
Good. This GitHub system is not 100%. Someone forked my NMEA0183 library and developed it. His branch does not even compile. Also original CAN bus shield has never updated their version. There are also some copies for NMEA2000 library, which are not up to date. So it is important that the most updated fork would be top on google.

sarfata, 2016-10-25:
Works for me too with the Raymarine i70 now that I have merged the fix.

closed 2016-10-25 by ttlappalainen

red8o, 2023-10-03:
I am having issues that are quite similar to the ones above:

My 1st setup is as a "NodeMCU 1.0 12E Module" with the ordinary MCP 2515 (possibly like this one https://aliexpress.com/item/32796488204.html). I swapped out the 8MHz crystal in favour of a 16 MHz crystral.
Whenever this setup receives this : "PGN:59904 Source:15 Dest:9 Len:3 Data:16,F0,1" (NodeMCU setup is node #9) it reliably crashes and does not send the PGN 126998. Crash loop continues 4 times, then resumes as expexcted.

My 2nd setup on the same bus it is an "SH-ESP32" by Matti Airas, that also gets a query from node #15 only seconds later. 
It uses an onboard CAN interface. This one receives: "PGN:59904 Source:15 Dest:20 Len:3 Data:16,F0,1" (SH-ESP32 setup is node #20). It does not crash and replies "PGN:126998 Source:20 Dest:255 Len:65 Data:2,1,2,1,3D,1,4E,4D,45,41,32,30,30,30,20,6C,69,62,72,61,72,79,2C,20,68,74,74,70,73,3A,2F,2F,67,69,74,68,75,62,2E,63,6F,6D,2F,74,74,6C,61,70,70,61,6C,61,69,6E,65,6E,2F,4E,4D,45,41,32,30,30,30".

My 3rd setup is a "ESP32-C3 Super Mini" (https://aliexpress.com/item/1005005757810089.html) which uses the ordinary CAN driver (https://aliexpress.com/item/4000496888935.html) and uses the esp32xx library (https://github.com/jiauka/NMEA2000_esp32xx). (This one is node #15 and for this test I issued PGN 59904 from node #20): It receives "PGN:59904 Source:20 Dest:15 Len:3 Data:16,F0,1" and replies "PGN:126998 Source:15 Dest:255 Len:65 Data:2,1,2,1,3D,1,4E,4D,45,41,32,30,30,30,20,6C,69,62,72,61,72,79,2C,20,68,74,74,70,73,3A,2F,2F,67,69,74,68,75,62,2E,63,6F,6D,2F,74,74,6C,61,70,70,61,6C,61,69,6E,65,6E,2F,4E,4D,45,41,32,30,30,30".

Conclusion: It looks like the ESP8266/MCP-Shield setup is somehow faulty.

I am not experienced, but if someone asks questions I am willing to answer as best as I can.

ttlappalainen, 2023-10-03:
With ESP8266/MCP-Shield it can be simply memory problem. Have you measured heap sizes/available memory during run? Have you tried to use smaller frame buffers?

I do not personally see any sense to use ESP8266/MCP-Shield. Most powerfull is to use Teensy 4. If you need WiFi, then ESP32. RPi for big projects.

red8o, 2023-10-03:
No, I am not into measuring that yet. I edited the question and found a workaround, as you can see.
The ESP8266 was just sitting there and I gave it a try. not using it for WiFi, just as a temperature sender.

ttlappalainen, 2023-10-03:
It actually can have something to do with PROGMEM. On Teensy and ESP32 PROGMEM does not have any effect. Reading from PROGMEM requires special code, which differs for platforms. One more reason not to use ESP8266 for NMEA2000.