ttlappalainen/NMEA2000 #32: Changing license

ttlappalainen, 2016-11-16:
Hi,

This is for all who has made changes to NMEA2000 library. I would like to change license from LGPL to MIT, which is more permissive than LGPL. This allows even companies to use library freely in their projects. I won't get millionaire anyway with this, so why keep stupid limitation. And anyway it would be fun to find after 10 years Garmin device running with this. And anyway any extra devloper may help to make library more general or robust.

So to make this 100% right I ask permission for change from all contributors.

Timo

thomasonw, 2016-11-17:
Works for me!

sarfata, 2016-11-17:
Works for me too. This is also the license I have chosen for KBox.

olesenrl, 2016-11-17:
Good for me. 

Regards,

Bob

denravonska, 2016-11-22:
Before I forget, I have a pending pull request if it gets accepted I suppose I too have to give permission. Just for the paper trail, the license switch works for me.

ttlappalainen, 2017-01-22:
No complains.

closed 2017-01-22 by ttlappalainen