ttlappalainen/NMEA2000 #36: PGN 65240 ISO Commanded address

ttlappalainen, 2016-12-20:
Hi,

I would like to add support for PGN 65240 to library. With this PGN some device can be commanded to change its source address to specified on command. Does anybody have system for testing it? I can write tester to my NMEA Simulator, but it should be proved with some "certified" system.

virtuvas, 2016-12-20:
Timo,

what exactly are the specs of the system needed for testing as I'm a bit confused on what you can achieve with that.
I have 3XGMI10 and a GPSMAP4008, as well as a VHF200 and an AIS600 (all Garmin N2K) is that good enough?

cheers

V.

ttlappalainen, 2016-12-20:
I would like to improve automated library functions. PGN 65240 can be used for forcing some device to have specific source address on N2k bus. I think normal N2k devices does not do that, since they simply rely address claiming, which is already supported on my library. Instead some N2k bus diagnostic tools or maybe some specific device may support this PGN.

Currently I only have Actisence device and NMEA Reader, but it treats that PGN as unknown. I can add support to library and handle it, but since all testing will be done with my own devices, which logic would be the same, then I can not prove that handling would be right.

ttlappalainen, 2017-01-22:
PGN 65240 has been implemented and tested with information I have. Cross testing with real system is missing.

closed 2017-01-22 by ttlappalainen