ttlappalainen/NMEA2000 #39: Support for Teensy 3.5/3.6

steveh57, 2017-01-25:
Timo, I'm just starting to test your excellent libraries on a Teensy 3.5.  I haven't set up with a real NMEA bus yet so haven't tested that part, but to get it to build I had to modify NMEA200_CAN.h as follows to include the new processors:

#elif defined(__MK20DX256__)||defined(__ATMEGA32U4__) || defined(__MK64FX512__) || defined (__MK66FX1M0__)
#define USE_N2K_CAN USE_N2K_TEENSY_CAN

I'm not familiar enough with GIT to use the proper change processes but thought I'd feed this back now.  I'll update you when I've done some more testing.
Steve.

ttlappalainen, 2017-01-25:
I can add it. Should it have __XXX__ around as others? Like __MK20DX256__ .

steveh57, 2017-01-25:
Took me a while to work that out - yes cut & paste seems to have dropped the underscores.

ttlappalainen, 2017-01-25:
Now it is updated. Should not require more, since the CAN interface is same.

closed 2024-01-21 by ttlappalainen