ttlappalainen/NMEA2000 #43: SetDeviceInformation for a multi device.

ieb, 2017-03-08:
Hi Timo,
Do I need to set multiple DeviceInformation blocks if I want to have a device that emits data for multiple things. eg Battery, Pressure, Temp, Acceleration, Magnetometer. I am using a Due, with a I2C 10Dof chip providing that information and also want to monitor battery voltages and currents using Hall Effect current sensors. I see the

#define Max_N2kDevices 1

but have seen others sending messages for multiple classes of device. May target network is a Raymarine ng network.

Thanks. (brilliant code btw, worked first time with SignalK)
Best Regards
Ian

ttlappalainen, 2017-03-08:
Hi,
You do not need to. It would be nice, but unfortunatelly multidevice support is still under my table. I do not know is it necessary at all. e.g. Alba Combi converts several types of analog data and still shows only one device.

ieb, 2017-03-09:
Hi,
Great. Thanks. I will give that a go.

closed 2017-03-09 by ieb