ttlappalainen/NMEA2000 #46: Overlap with Canboat

tkurki, 2017-03-26:
The following comment is based just on a quick look, I haven't used this library yet, so please disregard if I am mistaken here.

It seems that you are actually duplicating the information that is available in Canboat ([C](https://github.com/canboat/canboat/blob/master/analyzer/pgn.h), [XML](https://github.com/canboat/canboat/blob/master/analyzer/pgns.xml), [JSON](https://github.com/canboat/canboat/blob/master/analyzer/pgns.json)). 

This means that
- adding PGNs that are already in Canboat is manual work
- any updates and additions made here will not propagate to canboat and vice versa (see [example](https://github.com/canboat/canboat/pull/79))

[Adding PGNs to this lib](https://github.com/ttlappalainen/NMEA2000/pull/44/files) looks like something that could be done by code generation based on one of the linked files above.

Have you considered this approach?

tkurki, 2017-03-26:
As a reference: I wrote a small lib that uses Canboat's data to convert JavaScript objects with [proper fields](https://github.com/tkurki/to-n2k/blob/master/test/127250.js#L14-L18) to actisense_serial format:

https://github.com/tkurki/to-n2k

ttlappalainen, 2017-03-26:
When I first started with library, I looked also pgn.h. The problem with that was that it creates one huge object for all known PGN:s, which does not fit into systems with small memory. That is why I solved it with separate calls, which in compiler uses only that code, what you need.

The one thing what should be done is to fix pgn.h content. The pgn handling is based either OpenSkipper PGNdefns.N2kDfn.xml, which is originally based on Kees PGN data. So as reference I have compared both sources, but updated only one on OpenSkipper, when I have found errors.

tkurki, 2017-03-26:
How will N2kMessages.cpp be different, if you & others keep adding more PGNs to it? Is there some c++ magic here I am not familiar with?

If you generated the separate methods you could drive that with a configuration file that allowed you to generate code just for the PGNs you need.

ttlappalainen, 2017-03-26:
It is linker, which optimizes unused code away. So if you have Function1 and Function2in same source code  and you call somewhere Function1, but Function2 is not user anywhere, compiler will compiler both, but linker will optimize Function2 code away.

In some point configuration file could be used. But if you look in to the several messages definition, parameter order may differ from order in message and also there may be default values for data. Parameter order may differ to have most used parameters at first. All this is not so easy to have in configuration file. Also there may be different calls for same PGNs.

tkurki, 2017-03-26:
If it is the linker then it would not matter at all if N2kMessages had parse/set method for any number of PGNs - or have I understood something wrong?

After a quick look most of the default values are defaults for `Not Available`.

The convenience methods like `SetN2kTrueHeading` could just reference the generated method. Parameter order could be handled with convenience methods as well.

Would you consider this approach for future PGN coverage extension?

ttlappalainen, 2017-03-26:
In principle yes. Actually I would like to group functions somehow.

Maybe in future. I have no time now to write converter and test it with all messages. The current library messages I have verified with NMEA reader and my MFD as far it can handle them. Some has been tested by persons, who has added them.

Do you have offset for data? Check e.g. PGN 127513. The definition differs from what I have resolved and tested. There were several other messages, which definition I have fixed.

tkurki, 2017-03-26:
I see that 127513 is different in canboat and here, but can't really comment on it. Some 

Canboat data has bit offsets if that is what you're asking: https://github.com/canboat/canboat/blob/master/analyzer/pgns.json#L7818-L7908.

This is sort of exactly why I was thinking it would be wise to share and not divide efforts over a number of libraries..

This library doesn't have any tests with real world data either :-(

ttlappalainen, 2017-03-27:
Bit offset does not tell if value has offset. As with 127513 I found that peukert is one byte, but real values are from 1-1.504. So it should convert to real data with formula p=x*0.002+1, so having offset 1. The person who is using this has not confirmed it with real device. But it works with NMEA reader.

That I totally agree that one source for definition would be best. I have to apologise that I have used both pgn.h and OpenSkipper as sources, but I have only updated OpenSkipper. A bit lazy? It still could be possible to compare some old and new PGNdefns.N2kDfn.xml to see, what I have updated.

We should have
- script, which creates single set/get function for my library
- instructions how to use it
- instructions for testing and proofing it
- instructions for updating canboat or informing reponsives

On the other hand I am not sure is there sense to write a script. It is not big thing to do it manually, when you need it. At same time one can think about exeptions etc. But the rest would be more important.

tkurki, 2018-02-04:
Four Your Information: there is now https://github.com/sbender9/canboatjs, which is 100% based on the pgns structure database from canboat. In fact the module's code in itself knows almost nothing about the structure of the pgns, it just uses data in pgns.json to decode and encode N2K data.

It even has a bunch of tests with input and expected output.

sarfata, 2018-02-09:
If I can add my 2 cents, I agree that code generation from a single source seems possible and would probably be a better solution.

In my opinion there are two parts to this library:

 - The logic to be an entity on the network, claim an address, respond to request for name, etc. This part is a lot of carefully written and tested logic that is super helpful and cannot be generated. Ideally we would have more tests but that kind of stuff is super hard to test and it would be a big effort.
 - The "convenience" code to generate/read PGNs and that would be better generated.

closed 2024-01-21 by ttlappalainen