ttlappalainen/NMEA2000 #49: Parsing a String to a N2kMsg

Julius-Plehn, 2017-05-09:
Hello, 
thank you all for this great library. 
I would like to parse a message (e.g. 15FD0780264117753421B803) to an N2kMsg Object. However I am not using an Arduino, so I started to mess around with the code to build a smaller Arduino independent lib for my project. 
Sadly I didn't came very far so I am asking for help. I tried to reuse this code: [ActisenseReader.cpp#L60](https://github.com/ttlappalainen/NMEA2000/blob/025c39557cbb929d6a961c8f3fb0d118a4fad80d/ActisenseReader.cpp#L60) Is this the right place for me to start? 
What exactly should the MsgBuf variable contain? 
In `tActisenseReader::GetMessageFromStream(tN2kMsg &N2kMsg)` a NewByte is added to the MsgBuf. However MsgBuf is defined like this:
` unsigned char MsgBuf[MAX_STREAM_MSG_BUF_LEN];` So if I want to manually add a string in this format is MsgBuf expected to look like this? `unsigned char MsgBuf[] = {0x15, 0xFD, 0x07, 0x80, 0x26, 0x41, 0x17, 0x75, 0x34, 0x21, 0xB8, 0x03, 0}; `
I would really appreciate some kind of minimal code example on how to create a N2kMsg from a string like given above (created by Open Skipper) 
 
Thank you really much.

(this is not really an issue, maybe a feature idea)

ttlappalainen, 2017-05-09:
First note that library is not  anymore Arduino dependent. It can be used with MBED, RPi etc. For othes you need to write necessary wrappers like inherited N2kStream.

It is not possible to parse messages, if you do not know the data structure of it. For data structure you sould know the PGN of message. So How did you got that hexadesimal string? Note that ActisenseReader just parses messages in Actisense format.

And what you actually try to do?

Julius-Plehn, 2017-05-15:
Thank you for your help. My approach was just wrong, and I got it to work properly now. However I have an issue with decimal precision. Might be the same issue like described here: #11

closed 2017-05-19 by Julius-Plehn