ttlappalainen/NMEA2000 #51: PNG127245 Rudder no Parse function (only set)

PeterSte, 2017-06-04:
Great Libary. Many Thanks to the Autor(s).
But one thing is missing.  A Parse Function for the "Current Rudder Position".

I tryed to make the necessary changens in N2kMessages.h and N2kMessages.cpp Libary Files but I failed because looks like this Sentence is somehow special with this "Instance" instead of "SID" on another Position ...

Is there any Plan to create a Parse Function or is there allready another solution?
Would be great because for Performance sailing the Current Rudder Angle is a important parameter you don't wana miss 

Many Thanks for any Help
Peter

ttlappalainen, 2017-06-05:
Hi,

I added parser for rudder position.

PeterSte, 2017-06-05:
Hi

Wouwww, that was fast. Thank ´s a lot!

But seems a "cut and paste- Fault" made it into the code because the "Parse" Function is now double.
I removed the lines 336-341 and now it works :-)
336    inline bool ParseN2kRudder(const tN2kMsg &N2kMsg, double &RudderPosition) {
337      tN2kRudderDirectionOrder RudderDirectionOrder;
338      double AngleOrder;
339      unsigned char Instance;
340      return ParseN2kPGN127245(N2kMsg,RudderPosition,Instance,RudderDirectionOrder,AngleOrder);
341   }

Thanks again and best Wishes
Peter

ttlappalainen, 2017-06-05:
It was not double. The other was overloaded version so that you do not need to provide Instance,RudderDirectionOrder,AngleOrder, if you are not interested them.

PeterSte, 2017-06-05:
Ahhh, I think I got it know. "Instance" is truely the "instance" of Instance.RuderDirectionOrder and Instance.RudderAngleOrder. Sorry for understanding a bit slow :-)
Indeed if using your Libary for Performance Sailing only reading and parsing the RudderPosition is interresting. -> Just tested, works fine. 
If using your Libary for to controll a autopilot your "Set-example" describes exactly how to use the Libary.
So both Applications should be covered now.
Thanks a lot!

PeterSte, 2017-06-10:
Thanks again for the fast Implementaition.

closed 2017-06-10 by PeterSte