ttlappalainen/NMEA2000 #53: Teensy 3.2 not work with "universal" FlexCAN Libary

PeterSte, 2017-06-09:
Follow instructions to use Teensy 3.2 for Example: NMEA2000/Examples/ TeensyActisenseListenerSender you have to download and add these Libarys (in lets say C:\Program Files\libaries):
- Teensyduino
- FlexCAN_Libary-master
- NMEA2000_teensy-master
But if doing so the example will not work because Teensyduino has allready automatically installed a "universal" version of FlexCAN Libary  (FlexCAN.cpp Header says it is for Teensy 3.1-3.6), well hidden on C:\Program Files\Arduino\hardware\teensy\avr\libaries which is allways automatically taken by the Compiler.  Result is that everything works but the CAN will neither send or recive any NMEA2000 data.
-> Delete the Teensyduino FlexCAN Libary and so force the Compiler to use FlexCAN_Libary-master fixes the problem and everything works well now. 

Beside that:
I do not need to add the lines as described in the instructions:
#include <NMEA2000_teensy.h>
tNMEA2000_teensy NMEA2000;
Is this correct because the instruction say I should use it?

Thanks and best regards

ttlappalainen, 2017-06-09:
I have used other solution by replacing FlexCAN library under C:\Program Files\Arduino\hardware\teensy\avr\libaries as I mentioned on issue 50 https://github.com/ttlappalainen/NMEA2000/issues/50.

If you use automate selection header NMEA2000_CAN.h, you do not need to define board. So e.g. as in example TemperatureMonitor
#include <Arduino.h>
#include <NMEA2000_CAN.h>  // This will automatically choose right CAN library and create suitable NMEA2000 object
#include <N2kMessages.h>

With this you can easily change board on IDE. That is what I normally use since I have different board types and cross test them.

If you use only Teensy board, you can define
#include "NMEA2000.h"
#include "NMEA2000_teensy.h"
tNMEA2000_teensy NMEA2000;

PeterSte, 2017-06-10:
Replacing the FlexCAN Libary with a working Version is for sure the better solution. Sorry for no looking carefullly enough.  
How to use the Libaries is also clearified now.
Thanks a lot.

closed 2017-06-10 by PeterSte