ttlappalainen/NMEA2000 #57: read ProductInformation and DeviceInformation from N2k Bus

PeterSte, 2017-07-04:
Hello Timo
How can I READ ProductInformation and DeviceInformation (PGN:60928, PGN:126996 und PGN:126998) from N2k Bus?
In your Examples you describe very clear and detailed how to SEND this Information to the N2k Bus but I was not able to understand how to READ this very interresting information about the other devices on the N2k bus like Garmin...
All I could find was the function GetDeviceInformation (not even sure if this is the right one) but even this does not compile when called like: NMEA2000.GetDeviceInformation(); and creates fault: class tNMEA2000 has no member named "GetDeviceInformation" .
Please could you help me on this issue?
And I guess this would be interresting for others as well and should be added to one of the examples. 
Best Regards, Peter

ttlappalainen, 2017-07-04:
Hi,
I am making a separate class for collecting information about devices on the bus. This is not simple, since one has to handle all possible address changes also. So if you can wait for a while, you get ready code for this. I'll make sample too.

GetDeviceInformation returns information about your own device. It is necessary, since in NMEA2000 rules it is possible that some system changes your device instance values. For certified NMEA2000 devices, you should read those values, save them on e.g. EEPROM and restore them on startup. I have not yet published the sample.

PeterSte, 2017-07-04:
Hi Timo,
Seperate class and sample for "collecting information about devices" sounds great!
I guess also for other Libary users. I'am pretty sure it is not easy... 

Ahhh, thats the meaning of GetDeviceInformation. Got it.

To be hornest, without samples people on my skill level are pretty lost on that complicated bus.
Thanks again for your great job.

ttlappalainen, 2017-07-04:
You are also welcome to give comments for needed information like examples, comments there improving documentation etc.

ttlappalainen, 2017-07-12:
Hi,

Now I have first iteration of new class tN2kDeviceList. I have not published it yet, since I'll think a bit more logic and classes. I also would test it on weekend on my boat. At home I have only 2 devices for testing.

ttlappalainen, 2017-07-17:
Hi,

Are you still interested of this? Now I have full functioning tN2kDeviceList class. Some naming changes may be needed.

PeterSte, 2017-07-17:
Hi Timo

Sorry for my absence. I was not at home. 
Will be a pleasure for me to feedback comments and suggestions (as I allready did in the past). Thank you for the invitation.
Yes I am still very interrested. Looks like the test on your boat (real world :-) ) worked fine.
I was checking the NMEA2000 github but could not find the new tN2kDeviceList class to test it.  
Thanks and best Regards
Peter

ttlappalainen, 2017-07-17:
Well - yes and no. I was sailing on weekend, so I had time to test it and redesign data request logic. I think that now it is good and will notice all changes and devices.

I have not ye published it. I'll change some interfaces, before I am happy to it.

ttlappalainen, 2017-07-29:
Sorry for delay - I wanted to think this and required public interfaces carefully. Hopefully I succeed that so that I do not need to change any  public interfave.

So there is example DeviceAnalyzer, which lists you all devices on the bus.