ttlappalainen/NMEA2000 #64: PGN130306 Wind Data Length=6 or 8

ronzeiller, 2017-07-27:
Hi Timo,

first of all let me thank you and all the other contributors for your great work!

I had problems to show AWS and AWA in Expedition Software.
Data Input is coming from a teensy 3.6 with modified wind monitor sketch --> CAN Bus --> NGT-1 --> Expedition 

I found out on comparing with candump of merrimac-actisense-serial-2011.raw that PGN130306 is always length = 8
(Sorry I have no boat with real N2k bus)

Then I recognised, that you commented out in N2KMessages.cpp at line 1165 and 1166
    // N2kMsg.AddByte(0xff); // Reserved
    // N2kMsg.AddByte(0xff); // Reserved

Is there any reason why it is commented out?
Because after enabling N2kMsg.AddByte(0xff); Expedition shows AWS and AWA perfectly!
And NMEA Reader is ok too.
[image]

Best Regards
Ronnie

ttlappalainen, 2017-07-28:
Hi,

I have no idea, why it is commented out. I try to investigate this. Leave it uncommented on your code for a while.

Reserved field seem to mentioned on NMEA 2000 organization PGN database but not on hacked databases I use. PGN works with NMEA Reader and my Garmin GMI 20 without reserved field, so it has passed tests. Other systems like your Expedition Software may be more strict with defined data length and unnecessary reserved fields at end. NMEA Reader also shows reserved field, so I think it should be there. I'll test with my Garmin on weekend before pushing fix.

ttlappalainen, 2017-07-29:
Length has been changed now from 6 to 8 by adding reserved field, which has been defined on NMEA 2000 database. I also confirmed change with Garmin GMI 20.

closed 2017-07-29 by ttlappalainen

ronzeiller, 2017-07-30:
Could you please check **PGN 128259** Speed too?
I think there is the same issue and it should also be twice
N2kMsg.AddByte(0xff); // Reserved
with total length of 8

Speed is not shown in Expedition Number box (just in raw data) 
and in Kees canboat merrimac-actisense-serial-2011.raw example there is speed with length = 8 too.

Thanks,
best
Ronnie

ttlappalainen, 2017-07-31:
That is also fixed now. My Airmar DST 800 also sends 8 bytes.