ttlappalainen/NMEA2000 #67: Lowrance question

keenanstahl, 2017-08-26:
Please note this is not an issue with the code.  Really a question.  If there is another location questions such as this should be posted, please let me know.

I have a Lowrance HDS 5 (gen1).  I am able to communicate with the device fine.  I was curious if anyone has found a way to display messages on these devices?  I am trying to find a way to display text generated from my application on the device for testing and diagnostic purposes.

In looking through the PGN's the device supports it seems like I might be able to use an AIS message or DSC to accomplish this?  Any suggestions or tips would be appreciated.

I have tried sending PGN129802 (AIS Safety Broadcast) using the following code (my network has no AIS system or anything that would transmit or repeat this message).  I have a bus monitor and can confirm the message is sent, but the Lowrance doesn't show a message in it's "Vessels->Messages" tab. 

`void SendN2kAISBSafetyBroadcast()
{
	static unsigned long Updated=millis();
	
	if (Updated + 5000 < millis())
	{
		tN2kMsg N2kMsg;
		Updated += 5000;
		N2kMsg.SetPGN(129802L);
		N2kMsg.Priority = 2;
		//Msg1
		N2kMsg.AddByte(14);	 	//Bits 0-5 Identifier for Message 14; always 14
						//Bits 6-7 Repeat Indicator
		N2kMsg.Add4ByteUInt(201123456);	//Bits 0-29 Source Address (IMSI)
						//Bits 30-31 Reserved (0)
		N2kMsg.AddByte(0);		//Bits 0-4 AIS Transciever Info
						//Bits 5-7 Reserved (0)
		N2kMsg.AddStr("test message", 36);
		
		NMEA2000.SendMsg(N2kMsg);
	}
}`

ttlappalainen, 2017-08-26:
Have you tried to simulate that with NMEA Reader? If not I could do it tomorrow. One possibility is that N2kMsg.AddStr() is not right function to add the text message. E.g. PGN 126998 has different way for text data. It has first length of string +2, then 0x01 and then string data. See SetN2kPGN126998() on NMEA2000.cpp. I do not have confirmed data content for 129802 and also 126998 is declared wrong on Kees definitions.

keenanstahl, 2017-08-27:
I have not tried to simulate with NMEA reader.

ttlappalainen, 2017-08-27:
Try with code below. Works with NMEA Reader.

`void SendN2kAISBSafetyBroadcast(uint8_t MsgID, uint32_t SourceID, const char *msg)
{
  static unsigned long Updated=millis();
  int msgLen=(msg?strlen(msg):0);

  if (Updated + 5000 < millis())
  {
    tN2kMsg N2kMsg;
    Updated += 5000;
    N2kMsg.SetPGN(129802L);
    N2kMsg.Priority = 2;
    //Msg1
    N2kMsg.AddByte(MsgID);   //Bits 0-5 Identifier for Message 14; always 14
            //Bits 6-7 Repeat Indicator
    N2kMsg.Add4ByteUInt(SourceID); //Bits 0-29 Source Address (IMSI)
            //Bits 30-31 Reserved (0)
    N2kMsg.AddByte(0);    //Bits 0-4 AIS Transciever Info
            //Bits 5-7 Reserved (0)
    N2kMsg.AddByte(msgLen+2);
    N2kMsg.AddByte(0x01);
    for (int i=0; i<msgLen; i++) N2kMsg.AddByte(msg[i]);
    
    NMEA2000.SendMsg(N2kMsg);
  }

void loop() {
  SendN2kAISBSafetyBroadcast(14, 201123456, "test message");
  NMEA2000.ParseMessages();
}
`

keenanstahl, 2017-08-27:
- That worked!  Thank you.  The messages pop-up on the display and sound a tone (until acknowledged).  They also show up in the vessel->messages area with a time/date.
- I'd rather not use a safety related function to accomplish this so if anyone has a better suggestion, please advise.  Is there another message that can be used?
- This library is very mature and well thought out.  What is the recommended way to implement file logging instead of serial?  I have created a class to do this.
`class fileLogger : public N2kStream`
`{`
`public:`
`	virtual size_t write(const uint8_t* data, size_t size);`
`	virtual int read();`
`};`
In my application file (based on TemperatureMonitor.ino"  I have:
`fileLogger fl;`
`NMEA2000.SetForwardStream(&fl);`

This works.  But is this the intended approach?  Is there a built in way to filter the items that are logged? For example only logging data from a certain node.

I would also like to use file logging for the "DebugStream" defined in NMEA2000.cpp.  However I was trying to avoid making changes to the library.  How would you recommend to declare this and include it?
`#define DebugStream fl`

ttlappalainen, 2017-08-27:
Alert messages (PGN 126983-126988) would be probably right way to do it, but there is no description for messages and I do not know, are them supported on MFD:s.

Your file logger is one way. The name could be tN2kFileLogger (N2kStream should be tN2kStream to have same naming types). This does not need to be part of library or could be on own module.

There is no internal filtering. To log filtered messages you could use instead of fileLogger:
```
class tN2kMessageLogger : public  class tMsgHandler {
protected:
  N2kStream *logStream;
  unsigned char Source;
protected:
  virtual void HandleMsg(const tN2kMsg &N2kMsg) {
   if ( N2kMsg.Source==Source )  {
    N2kMsg.SendInActisenseFormat(logStream);
   }
  }
public:
  tN2kMessageLogger(tNMEA2000 *_pNMEA2000, unsigned char _Source, N2kStream *_logStream) : tMsgHandler(0,_pNMEA2000) { 
    Source=_Source;
    logStream=_logStream; 
  }
};

void setup() {
...
 new tN2kMessageLogger(&NMEA2000, 2, &fl);
 new tN2kMessageLogger(&NMEA2000, 4, &fl);
...
}
```
The problem on filtering by source is that it may change. To make it reliable, you should use tN2kDeviceList object and handle logging by unique ID:s to make it 100% sure. See example DeviceAnalyzer to see how tN2kDeviceList can be used.

To filter according to PGN:s you could redefine known messages by using SetSingleFrameMessages and SetFastPacketMessages. Then set SetHandleOnlyKnownMessages, SetForwardOnlyKnownMessages, and SetForwardSystemMessages as you wish.

DebugStream is meant for problem solving. To enable debug messages, you have to define e.g. NMEA2000_FRAME_IN_DEBUG. In Arduino IDE this is not possible without modifying code. With better IDE (e.g. Atom?) one can have different build settings with different compiler definitions. On the library definition could be changed to
```
#ifndef DebugStream
#define DebugStream Serial
#endif
```
Then you could use own compiler definition. But for Arduino IDE, you still need to modify code.

On the other hand you can not use #define DebugStream fl since library does not know fl. If this is important, I could think to add method:
 `void SetDebugStream(N2kStream* _stream) { DebugStream=_stream; }`

keenanstahl, 2017-08-29:
Thanks for the tips.  Very helpful.  It doesn't appear that the Alert message PGN's are supported by my device.

closed 2017-08-29 by keenanstahl