ttlappalainen/NMEA2000 #70: NMEA2000 update

janzan, 2017-09-05:
hi,
I have downloaded the newest NMEA2000 Library version at 01 Sept. After the update the compiler (Arduino IDE 1.8.4) spewed out a couple error messages like this:

/home/james/Arduino/libraries/NMEA2000-master/src/NMEA2000.cpp:241:6: error: prototype for 'void tNMEA2000::ClearCharBuf(int, char*)' does not match any in class 'tNMEA2000'
 void tNMEA2000::ClearCharBuf(int MaxLen, char *buf) {
      ^
In file included from /home/james/Arduino/libraries/NMEA2000-master/src/NMEA2000.cpp:24:0:
/home/james/Arduino/libraries/NMEA2000-master/src/NMEA2000.h:78:15: error: candidate is: static void tNMEA2000::ClearCharBuf(size_t, char*)
   static void ClearCharBuf(size_t MaxLen, char *buf); 
               ^
/home/james/Arduino/libraries/NMEA2000-master/src/NMEA2000.cpp:248:6: error: prototype for 'void tNMEA2000::SetCharBuf(const char*, int, char*)' does not match any in class 'tNMEA2000'
 void tNMEA2000::SetCharBuf(const char *str, int MaxLen, char *buf) {
      

 The same error on both station, MAC and Linux.  It seems to be a problem with "size_t"   definition, maybe..  i don't know...

rg
jan

ttlappalainen, 2017-09-05:
Sorry, for some reason NMEA2000.cpp was not updated. Download new version then it should work.

janzan, 2017-09-05:
Compiling succeeded! tnks!
i will test it coming soon on  GO7..  maybe on weekend