ttlappalainen/NMEA2000 #71: plain gcc makefile for examples

paulychris, 2017-09-13:
Hi
This is a great NMEA2000 library, and i would like to use it for a project based on a AT90CAN avr-board.
This board is unfortunately not supported in the arduino IDE, and anyway i would prefer to use plain avr-gcc compiler with makefile for the project.

However - my understanding of makefiles is not sufficient to modify one to include all the necessary nmea2000- and can-lib files to compile my own main.cpp...

As i understand, the NMEA2000 lib is already more or less independant of arduino - only the examples do need arduino IDE to compile.

For me, but maybe also for others, it would be extremely helpful, if one could include one simple example (like eg the battery monitor example) running with plain gcc. Having such an example, including a makefile, would then allow to easily adapt to own projects using this great lib.

Maybe anyone already has such an example, and is willing to post it or include it in the git ???

ttlappalainen, 2017-09-13:
I'll ask.

It is goal to keep NMEA2000 library as independent from environment. As mentioned on Readme.adoc (main page), it already have tested on avr, mbed, RPi out from Arduino

thomasonw, 2017-09-13:
Hello.  Just a couple of comments, which might help?

First, I have done an AVR_CAN driver, tested using the ATmega64M1 uC, but 'should' support the AT90CAN as well - if you find it helpful, and changes are needed please let me know.  (background, i based it off a Due port, and it is in some ways a mess - very large, with a few redundancies, but it is stable - and I wanted to make sure I matched the existing Due API)
[https://github.com/thomasonw/avr_can](https://github.com/thomasonw/avr_can)

As far as a non-Arduino implementation, I have a couple of non-Arduino ports working - though they are not posted online yet (using for larger project development at the moment).  One is a STM32F0 based in the MBED environment, the other an RPi port based on Linux.

[https://github.com/thomasonw/NMEA2000_socketCAN](https://github.com/thomasonw/NMEA2000_socketCAN)
[https://github.com/thomasonw/NMEA2000_mbed](https://github.com/thomasonw/NMEA2000_mbed)

For a straight forward include, I think you would need to re-create the 'bridges' as shown on either of these ports, and then in your main program replicate the includes and prototypes that normally lives in `NMEA2000_CAN.h`     Example, if you create a 'bridge' file called `NMEA2000_avr.cpp` and `NMEA2000_avr.h`, then in your CAN subsystem you could add:
```
#include "N2kMsg.h"
#include "NMEA2000.h"
#include <NMEA2000_avr.h>      
tNMEA2000_avrCAN NMEA2000;
tavrStream serStream;
```
making sure NOT to do the `#include <NMEA2000_CAN.h> `

Does this help?  If you wish, feel free to post here / contact directly - would be happy help you getting to a usable point.  Then perhaps your example could be shared as a prototype of others?

paulychris, 2017-09-13:
Yes, i already found your AVR_CAN driver and try to use it. The corresponding github references arementioned inside NMEA2000_CAN.h, same for NMEA2000_avr.h 

But why cant i include the NMEA2000_CAN.h directly. Should it not "bridge" to the right calls automatically, provided the corresponding defines ("__AVR_AT90CAN128__") are set ?

My first try waS, to simply copy NMEA2000_avr.cpp, NMEA2000_avr.h, avr_can.cpp, and avr_can.h
from your repositories into the NMEA2000-master/src directory.
And then try to tell my Makefile to include NMEA2000-master/src as include directory...

But i am still failing on the Makefile :-(

paulychris, 2017-09-13:
Now its trying to compile... But when compiling the AVR_CAN driver, i removed the Arduino.h include in avr_can.h...  Now i have problems with missing functions like "digitalWrite was not declared".

Does your AVR_CAN driver depend on arduino IDE ??? Do i still have to include Arduino.h ?

thomasonw, 2017-09-13:
Hello.  I just looked through the lib, and yes - there are some Arduino specific code in there!  Specifically around driving a port to control the Enable/standby pin on the CAN transceiver -- to be honest, I did not remember that was in there (Again, I ported directly from a Due example), and in all the code I do I control that pin directly, not through the lib.

I would be Ok with removing those Arduino specific calls. Perhaps that would allow more general use?  If you agree, let me know and I will push up a change pulling the Arduino specific code - or if you wish to do it, that is fine as well.

paulychris, 2017-09-13:
Making your AVR_CAN independant from Arduino.h would for sure help to compile one of the examples stand-alone -
But i am not sure if i will manage then. For the time being, i tried to simply include the Arduino.h and all other includes from the IDE. This will not work functionally, sure, but at the moment I would be happy if it would simply compile...
Apart from several warnings i now run into problems due to undefined reference to N2kStream::println(__FlashStringHelper const*)  in NMEA2000.cpp
???

thomasonw, 2017-09-14:
Give me a bit of time and I will de-Arduino the code..

ttlappalainen, 2017-09-14:
Thomas, have you checked my changes to due_can? It has been also merged to original collin git. I made own intenal buffer handling for reserved and common send buffers. It is the same logic as FlexCan and now on NMEA2000_mcp. The point is that you need bigger buffers fastpacket than for others, since there is only one channel for sending fastpacket but singles can be send from any box. And also singles has higher priority than fp. So if you sending a lot, it may buffer e.g. product information and configuration information to fast packet while sending singles.

paulychris, 2017-09-14:
I got it compiling with avr-gcc and plain makefile, though not linking yet due to the problem with N2kStream::println. And i still include the Arduino.h and other .h files from Arduino due to the dependencies of AVR_CAN on Arduino.

However, to get it compiled i had to do some minor modifications in the code, which i think should be corrected permanently:

in NMEA2000.h,  "memset" is not declared:
 -> had to add: #include <string.h>

in N2kDeviceList.cpp, "malloc" and "free" were not declared.
.> had to add #include <stdlib.h>

in N2kStream.cpp there is a duplicate definition of N2kStream::print(..) (lines  15, and again line 24.)
-> had to remove one of them: which one is the correct one ?

What could i do about the missing println:
"undefined reference to `N2kStream::println(__FlashStringHelper const*)'

thomasonw, 2017-09-14:
@paulychris   Hello.  Just now I pushed up a modified avr_can  lib with all the arduino code removed.  The side effect is the transceiver Enable/stby pin is no longer controlled inside the lib, and if you are using it you will need to manage it externally.

Concerning the println.  Without modifying Timo's lib, you will need to create a bridge of some type between the NMEA2000 lib and the environment you are working in.  I might suggest you look at the bridged I did for MBED and perhaps socketCAN for ideas:
[https://github.com/thomasonw/NMEA2000_mbed](https://github.com/thomasonw/NMEA2000_mbed)
[https://github.com/thomasonw/NMEA2000_socketCAN](https://github.com/thomasonw/NMEA2000_socketCAN)

Once those bridges are created you should be able to compile w/o linking errors..

thomasonw, 2017-09-14:
@ttlappalainen   That sounds like a nice change, tiered queues.   Will take a look.

paulychris, 2017-09-14:
Thanks a lot for the quick update !

By looking into the MBED example, i hope i will manage now to compile it.