ttlappalainen/NMEA2000 #72: NMEA2000 TimeStamp in PGNs

ronzeiller, 2017-09-21:
Hi Timo,

for a project I want to write, among other things, datas from IMU-Sensor (heel and pitch) to N2k CAN-Bus (using Teensy 3.6)
I was looking at the library to find out how I could set timestamps, but it seems that the library writes millis() at init of N2kMsg.
I found no possibility to write actual time to the PGN.

On the other hand I read the N2k Bus to get actual time e.g. from PGN 126992  SYSTEM DATE/TIME
I am sorry, but I have no boat for checking real CAN bus datas. 
Thus I am working with your (great!!) Kave NMEA Simulator writing to a Teensy N2k-Writer to a CAN-Bus simulation, reading the bus with another Teensy, running your library.
But when I read the PGNs from the simulator  I get just millis() as N2kMsg.msgTime

My questions:
1. is there actual real date/time in a N2k PGN-sentence on a boat CAN-bus?
2.if yes, would it be possible (and does it make sense) to write real time into the PGN with your library, or do I have to manipulate millis()?

Thank you!
Ronnie

ttlappalainen, 2017-09-22:
tN2kMsg.MsgTime is just for comparing, if you need to know message order. You should not try to use that for actual time. Unsigned long is not enough for saving actual time in ms.

1. PGN 126992 is the only time message for NMEA2000. I have not found PGN on CAN. It has 0.0001 s accuracy (100 ns). If you have GSP system on the bus, it probably will contain PGN 126992. If not, then you do not have actual time. Note also that even it has 100 ns accuracy, there may be delays, before you receive the message, If your system does 5 ms other tasks before reading frame buffers saved by interrupt, your MsgTime is 5 ms delayed. Lower level libraries does not unfortunately save frame time.

2. You can use PGN 126992 to send time to bus, if you have real time available. It makes sense, if some device listens and uses that PGN. Sending timestamp does not belong to main library task, since it is not mandatory. Note also that on the message source selection there are only real time sources, so you should not send that message with just some Teensy clock, which time you have set manually except in test purposes.

Simulator sends PGN 126992 to the bus, since it simulates GPS. So you can listen that message.

ronzeiller, 2017-09-22:
Thank you for clearing that up!

ronzeiller, 2017-09-23:
reopened the issue for one more question:
does this mean that on a real world CAN-Bus on a boat there is **one** counting timestamp for all units (as it is when I use your NMEA Simulator) or does every single N2k-unit on the bus have its own timestamp(system)?

ttlappalainen, 2017-09-23:
In priciple only one. Normally units does not need any timestamp. If you need e.g. to log messages, the logger should take timestamp from PGN 126992 and synchronize itself. Then messages will be logged according to logger clock.

But you can have several GPS units on the same bus, then you would get 126992 from multiple sources. Receiver must then decide, which source to use. Note that if you e.g. have normal single antenna GPS and more accurate dual antenna GPS compass on same bus, you can have also position messages showing different position due to single antenna GPS inaccuracy. This is the case I have on my boat. So in multi GPS system the device using any of those PGN:s should have possibility to choose source too. That should be possibel to simulate by runing two instances of simulator. You can use command line (e.g. -options %PROGDIR%\NMEA_Simulator1.ini) parameter to define different settings destination for them

Also note that source address can not be used for 100% sure as address to specific device due to address claiming. In steady system it is OK, but if you add new device, addresses may change. For permanent solution there is DeviceList class, which you can use for reading device ID:s. Unfortunately there is no example yet how to do that right.

ronzeiller, 2017-09-23:
Thank you for your fast and detailed answer.
First issue is clear now. I will take my internal timeStamp for all incoming messages, regardless when they "have been created" on the bus. (It should be a matter of just a few milliseconds delay, which means nothing in real sailing world)

The second matter is a very good hint!
Time in 126992 should be the same of all GPS devices, but you are right, COG/SOG and position we have to take care.
May be I am coming back in this DeviceList class thing after having a closer look at the library.....

closed 2017-09-23 by ronzeiller