ttlappalainen/NMEA2000 #73: ParseN2kWaterDepth Parsing -> depth not available

ronzeiller, 2017-09-26:
Parsing PGN 128267  Water depth with function from NMEA Library as written in display2.ino

PGN is coming from Kave simulator standard settings.
But PGN128267  Water depth is messaging: "Water depth is not available"

Should there not be a "**!**" in line 407?
if ( **!** N2kIsNA(DepthBelowTransducer) ) {
......
}
otherwise it will always jump to "not available"

ttlappalainen, 2017-09-26:
Yes, thanks. This is tested code:) I'll fix it to next release of library hopefully on this week.

BTW with curren code it will not allways jumpt to "not available". In case value would be NA, it would show you crazy depth. Value can be NA depending of transducer. Some transducers may send NA for depth to deep to measure.

ttlappalainen, 2017-09-26:
I made a separate fix.

ronzeiller, 2017-09-27:
Great, thank you 👍

ronzeiller, 2017-09-27:
A short follow up question just arose: 
if ( N2kIsNA(Offset) ) {
        WaterDepth = DepthBelowTransducer;
} else {
        if (Offset>0) {
          OutputStream->print("Water depth:");
        } else {
          OutputStream->print("Depth below keel:");
        }
}

If Offset == 0 should there not also be DepthBelowTransducer?
Like: if ( N2kIsNA(Offset || Offset == 0) ) {......

ttlappalainen, 2017-09-28:
Maybe you are right. 0 would be actually a bit crazy value. It means that you have transducer on the surface, where is won't work reliably. I changed code to:
      if ( N2kIsNA(Offset) || Offset == 0 ) {
        PrintLabelValWithConversionCheckUnDef("Depth below transducer",DepthBelowTransducer);
        if ( N2kIsNA(Offset) ) {
          OutputStream->println(", offset not available");
        } else {
          OutputStream->println(", offset=0");
        }
      } else {

ronzeiller, 2017-09-28:
Hmm, I think offset = 0 just means Water Depth below Transducer
It makes also sense, when you are looking to the "according" NMEA0183 sentence DBT (Depth below transducer)

See also image taken from Raymarine:
[image]

Some transducers have separate settings for measurement of transducer -> keel 
and transducer -> water surface.

May be the NMEA2000 PGN is then translated like:
If offset > 0 => automatic for Water depth sea
If offset < 0 => automatic for Depth below keel or more general Depth from bottom of boat
(as you had in debug print in your code.)

On chartered boats they sometimes make higher negative offset for to keeping people off shallow water :-)

ttlappalainen, 2017-09-28:
I am not quite sure. Offset should be defined NA for meaning "below transducer". There is no clear document for 0. Document says positive. So is 0 positive? With positive you can show water depth. So if 0 is positive, then it means water level and transducer difference is 0, so it also shows "below transducer", since it is same.

On the picture I do not see any sense for meaning 0 for depth from transducer. It just does not have any sense - who would like to show that specially on sailboat. And it is not first time I see error in Raymarine docs.

ronzeiller, 2017-09-28:
Which document?

For Raymarine I totally agree! :-)

ttlappalainen, 2017-09-28:
Google "20151026 nmea 2000 pgn_website_description_list". That says only about positive values, but I could not find doc, which defines NA behaviour. It may have been some general doc about NA, which I have "translated" to this.

BTW: I added Range, but did not yet publish it.

closed 2017-10-02 by ronzeiller