ttlappalainen/NMEA2000 #75: Logging N2k-Bus data

ronzeiller, 2017-10-02:
Hi Timo,
thinking of logging bus datas in Actisense format, because it is the smallest (data format).

This function should be integrated into my "normal" program as an option.
There is a handler and in another class I get hold on the N2kMsg (N2kMsg.PGN, N2kMsg.MsgTime etc. ) for some manipulations.

For logging I actually do not have a stream, but would need a String with the Actisense formatted datas to hand over to a "write to SD-Card" function.
As there is no timestamp in Actisense format (right?) I would add millis() to the string.

Quite desperate because I did not find a working solution, but may be I just do not see "the forest because of all the trees" as we say here :-)

Could you please help?

As logging without reading the datas later on would not make sense, could you give me a hint how this could be done too? (Here would be separate code for reading the sd-card over serial and the normal read stream handler should work, right?)

Thank You!

ttlappalainen, 2017-10-02:
Actually there is the message time in Actisense format. MsgTime will be saved there, so will have millis() included. If you can not write it by using stream, I could separate from N2kMsg::SendInActisenseFormat to one, which creates the buffer. Then you can do with char buffer, what you like. Then you just catch all messages with MessageHandler (as in DataDisplay2) and save them there.

ronzeiller, 2017-10-02:
included millis() is fine and char buffer sounds great!
Your help is very much appreciated!

ronzeiller, 2017-10-05:
May be there was a confusion about "Actisense Format" on my side:
Is it true, hat this format is a binary format, just for Expedition and a few other navigational programs?

In your library I found in ActisenseReader.cpp the remark for Actisense format:

`<10><02><93><length (1)><priority (1)><PGN (3)><destination (1)><source (1)><time (4)><len (1)><data (len)><CRC (1)><10><03>`
e.g.:
`Pri:6 PGN:130306 Source:16 Dest:255 Len:8 Data:0,56,4,9,89,0,FF,FF`

which is just a translation into reagable ASCII plus your comments PGN:, Source:,.....?

For Logging, now I did:
managed to do it with an own formatted string, right away from your N2kMsg which gives me already alI needed:

` @<Timestamp N2kMsg.MsgTime>,<PGN-Nr>,<source>,<DATA>*<Checksum>`

```
@935031,129025,0D,3AAAF923076D0C4A*27
@935031,129026,0D,00006F6AE201FFFF*2A
```

It is similar to Seasmart format, but for me nicer and easier to read.

Thanks for your help!

ttlappalainen, 2017-10-05:
Actisense format is binary format. With text format you will double size of data. 
Have you noticed that there is also Seasmart.cpp/.h? 

I would not prefer to invite a new format, it there is not very high improvement with it. Otherwise you need just an other converter for analyzing.

I started to make function for you function to get data in buffer as Actisense format. Then I noticed that e.g. Arduino Mega is dramatically slow on generating that. So I optimize code and dropped time to less than half from original (long message 608 us -> 260 us). But for some funny reason time on Teensy increased 65% (20 us -> 33 us). So I have to investigate a bit more befor publishing.

ronzeiller, 2017-10-05:
Yes, actually I took Seasmart conversion for my own adaption.
As it is internal logging, it is more of interest for us to make datas more readable for humans, than to have Hex numbers for timestamp and PGN in an format (Seasmart) which is here not needed at all.

Conversion later on should be easy with your library, as I parse all N2k-Sentences anyway.
(Easy to adapt  SeasmartToN2k)

**Feedback on Actisense Format:**
We have problems here to read from N2k-Bus (written with your library) converting to Actisense-Format with ActisenseListener (from examples) sending to Serial:

**NMEA Simulator -> USB / Teensy 3.2 --> CAN Bus --> Teensy 3.6 /ActisenseListener -> USB -> Expedition**

vs. 
**NMEA Simulator -> USB / Teensy 3.2 --> CAN Bus --> NGT-1**

The Teensy/ActisenseListener hangs after a while, but a the NGT-1 continuous to  work.
The weird thing is, that I could not see so much ASCII on the NGT-1 datas compared to Teensy/ActisenseListener (looking with PuTTY)

ttlappalainen, 2017-10-05:
I have noticed problem with NMEA Reader that ActisenseListener stops sending. I have added LED blink to my version and it shows that Teensy continues to run, but does not send anything. If you unplug it and replug, it continues run. One problem may be that NMEA Reader and Expedition are made for NGT-1, which expect some initialization commands at the beginning. So now Teensy/ActisenseListener is getting in some data, which may stuck the Serial. Could you try to add just dummy code to ActisenseListener loop:
```
...
  if ( Serial.available() ) { int ByteIn=Serial.read(); }
...
```
To empty serial in buffer. Try if this helps.
If I us putty, Teensy/ActisenseListener newer stuck.

Also note that you should enable all messages to NGT-1 otherwise it will not send data if you use putty.

ronzeiller, 2017-10-05:
First test results with:
1. Dummy Code
2. switched to 
```
#include <NMEA2000_teensy.h>
tNMEA2000_teensy NMEA2000;
```
Up to now, everything is working fine!
(Will keep you informed)

[image]

ttlappalainen, 2017-10-05:
What do you mean with 2.? You can just use #include "NMEA2000_CAN.h", since that automatically chooces library according to processor. So finally that make for Teensy exactly same as you did manually.

ronzeiller, 2017-10-05:
Ooh, sorry, I was not aware that you have this in the code.

Of course you have it!