ttlappalainen/NMEA2000 #76: PGN 126992 SystemTime

usauerbrey, 2017-10-04:
Hi Timo
The Nmea Doc says:
26992
The purpose of this PGN isto provide a regular transmission of UTC time and date; optionally synchronized to other parameter
groups from the same source.
Field # Field Description
1 Sequence ID
2 Source
3 NMEA Reserved
4 Date
5 Time

The same ist true for actisense NMEA reader:
[image]

but in the lib it is:
// System date/time
// Input:
//  - SID                   Sequence ID. If your device is e.g. boat speed and heading at same time, you can set same SID for different messages
//                          to indicate that they are measured at same time.
//  - SystemDate            Days since 1970-01-01
//  - SystemTime            seconds since midnight
//  - TimeSource            see tN2kTimeSource

Is this a bug?

best regards Uwe

ttlappalainen, 2017-10-04:
No. The fields are sent to bus in right order. In interface they are ordered in order of necessity to have time source at the end so that it can have default value and does not need to provide by caller. I do not know how you have 15 on time source, since known value range is 0-5. Actually I think that NMEA reader should show "Not Available".

And yes, there is small erro in code, which I'll fix on next version. I do provide to reserved 0 instead of 0xf. And on listening TimeSource may get crazy values, since it does not have filter xx & 0xf.

usauerbrey, 2017-10-04:
Besides the order, does the NMEAReader do a conversation? If not, it shows Date an Time as 04/10/2017 and not as 
SystemDate Days since 1970-01-01
SystemTime seconds since midnight
so I am unclear

ttlappalainen, 2017-10-04:
NMEA Reader does the conversion. And SystemTime is UTC.

You can find example NMEA0183ToN2k under https://github.com/ttlappalainen/NMEA0183/tree/master/Examples/NMEA0183ToN2k how I use it. RMC message provides same value, so I do not need any time.h. I have also code for converting DaysSince1970 to current date.

usauerbrey, 2017-10-04:
Hi Timo

Thanks very much, it is always a problem with monitoring, do they give raw data or already processed data.

Uwe

usauerbrey, 2017-10-04:
Add question, where can I find the code for converting DaysSince1970 to current date?

ttlappalainen, 2017-10-04:
Look NMEA0183Msg.cpp there is:
```
unsigned long tNMEA0183Msg::DaysToNMEA0183Date(unsigned long val) {
  if ( val!=NMEA0183UInt32NA  ) {
    tmElements_t tm;
    time_t t=val*SECS_PER_DAY; //daysToTime_t((val));
    breakTime(t, tm);
    val=tm.Day*10000+(tm.Month)*100+(tm.Year+1970-2000);
  }
  
  return val;
}
```
It converts DaysSince1970 to value ddmmyy used in NMEA0183. It uses timelib.h, which also have define for tmYearToY2k(tm.Year), which I could have use instead of (tm.Year+1970-2000). Write your own function, which returns date from tm in format you like to have.