ttlappalainen/NMEA2000 #77: PGN 128275 DistanceLog

usauerbrey, 2017-10-04:
Hi Timo

Is this definition correct or is it a copy&paste error??

//*****************************************************************************
// Distance log
// Input:
//  - DaysSince1970         Timestamp
//  - SecondsSinceMidnight  Timestamp
//  - Log                   Total meters travelled
//  - Trip Log              Meters travelled since last reset
// Output:
//  - N2kMsg                NMEA2000 message ready to be send.
void SetN2kPGN128275(tN2kMsg &N2kMsg, uint16_t DaysSince1970, double SecondsSinceMidnight, uint32_t Log, uint32_t TripLog);

inline void SetN2kDistanceLog(tN2kMsg &N2kMsg, uint16_t DaysSince1970, double SecondsSinceMidnight, uint32_t Log, uint32_t TripLog) {
  SetN2kPGN128275(N2kMsg,DaysSince1970,SecondsSinceMidnight,Log,TripLog);

ttlappalainen, 2017-10-04:
What do you mean? What part would have copy/paste error? It looks correct. And normally I test all new features with other devices like NMEA Reader.

usauerbrey, 2017-10-04:
Sorry forget it, better reading would help, sorry

best regard Uwe

closed 2017-10-04 by usauerbrey