ttlappalainen/NMEA2000 #78: Reference Values

PeterSte, 2017-10-09:
Hello Timo
My question is about the Reference Value of the NMEA2000 Sentences. I think that you use the NMEA2000 Manual from the NMEA Organisation, right? 
_Example Wind with 4 Reference Value from your code: 
https://github.com/ttlappalainen/NMEA2000/blob/master/src/N2kMessages.h (line 150-155)
150  enum tN2kWindReference {
151                                                    N2kWind_True_North=0,
152                                                    N2kWind_Magnetic=1
153                                                    N2kWind_Apprent=2,
154                                                    N2kWind_True_boat=3
155                                              };_

But it looks like some manufators may use (different?) or at least more Reference Values as you actually do.
Example Product with 5 Reference Values: https://www.rocktheboatmarinestereo.com/specs/MSNRX200I.pdf

So my question is if I can use this 5th value (as shown in your (very good) Reader examples) just by extending the code by another "if Reference=5" or if you limit the number of the "Reference Values" in your code?

Thanks and best Regards
Peter

ttlappalainen, 2017-10-10:
I do not have other NMEA 2000 documents than public available documents. If you buy NMEA Documents, you sign agreement, that you will not publish information. So main references are PGNDefns.N2kDfn.xml file on OpenSkipper project, which I update and which is originally based on Kees CANboat project pgn.h. New information comes from testing and different sources.

I updated wind reference to library. I do not know, why N2kWind_True_water=4 was missing. But as far as I know, there is no ref=5. Which example you mean?

Any certified manufacturer can not use their own definitions on any values. New definitions will be first added to NMEA database by organization. But since NMEA 2000 is closed system, that new information will not be available as public.

PeterSte, 2017-10-10:
Thanks for explaning the background your sources, and for updating N2kWind_True_water=4. 
So with other words the reference that I found was right and it was only a missing parameter in the library that has been updated.

You are absolutly right, there is no "ref=5". This was just a "calculation fault" :-) from my side because its 5 ref's (0-4) and I was not thinking and started counting at 1 instead at 0. Sorry for the confusion 👎 

But just for my curiosity. If a ref=5 for the "wind PGN" would arrive on NMEA2000 Bus and if I would have extended the code to a "if ref=5", would your library forward this formaly not correct  "ref=5" to the "if ref=5" or would it block this value?

ttlappalainen, 2017-10-10:
It cast readed byte to tN2kWindReference, so in principle you could test if (ref==5). But on the other hand that violates code and NMEA 2000 definitions. You can not do it to other directions: tN2kWindReference ref=5; You could do it with tN2kWindReference ref=(tN2kWindReference) 5; but then you again violating code rules.