ttlappalainen/NMEA2000 #83: Leeway PGN 128000

ieb, 2017-11-28:
I could not find the leeway PGN here so did this. I would do a patch, but I need to update version of your libraries first, and probably my display software.

    // Leeway
    // https://www.nmea.org/Assets/20170204%20nmea%202000%20leeway%20pgn%20final.pdf
    // 4x per second.
    void fillLeeway(tN2kMsg &N2kMsg) {
        N2kMsg.SetPGN(128000L);
        N2kMsg.Priority=4;
        N2kMsg.AddByte(sid);
        // int16  leeway/1E-4 in radians fro +pi to -pi +ve means leeway to starboard.
        N2kMsg.Add2ByteInt((int16_t)((double)leeway/(double)1.0E-4F));
        // 40 bits = 5 bytes, wonder what these will be used for ?
        N2kMsg.AddByte(0xff); // Reserved
        N2kMsg.AddByte(0xff); // Reserved
        N2kMsg.AddByte(0xff); // Reserved
        N2kMsg.AddByte(0xff); // Reserved
        N2kMsg.AddByte(0xff); // Reserved
    }

ttlappalainen, 2017-11-28:
I can try that. But on library to keep function similar  with others function definitions will be:
void SetN2kPGN128000(tN2kMsg &N2kMsg, unsigned char SID, double Leeway);
void SetN2kLeeway(tN2kMsg &N2kMsg, unsigned char SID, double Leeway);

Also right library way to add double is:
N2kMsg.Add2ByteDouble(Leeway,0.0001);

So the function will be
//*****************************************************************************
// Leeway
void SetN2kPGN128000(tN2kMsg &N2kMsg, unsigned char SID, double Leeway) {
    N2kMsg.SetPGN(128000L);
    N2kMsg.Priority=4;
    N2kMsg.AddByte(SID);
    N2kMsg.Add2ByteDouble(Leeway,0.0001);
    N2kMsg.AddByte(0xff); // Reserved
    N2kMsg.AddByte(0xff); // Reserved
    N2kMsg.AddByte(0xff); // Reserved
    N2kMsg.AddByte(0xff); // Reserved
    N2kMsg.AddByte(0xff); // Reserved
}

Reserved bits are for future use.

ttlappalainen, 2017-11-29:
I added functions to library. Unfortunately I do not have way to confirm this message, since it is so new that none of my test system recognize it. If you have device, which can receive or send that PGN, please test it.