ttlappalainen/NMEA2000 #84: PGN128259 Speed

ronzeiller, 2017-11-29:
Hi Timo,
reading values from my NGT-1 with Actisense NMEA Reader.
Data flow:
NMEA Simulator -> USB reader with your library -> test-N2k-bus -> NGT-1 -> Actisense NMEA Reader

In PGN128259 Speed (I do not know why Actisense is writing Speed, Water referred) I have two values:
Field 2: water referenced
Field 3: ground referenced

In your simulator one can set "current" 
If I turn off GPS in simulator and set some current -> Field 2 and 3 have different values. 
In simulator there seems to come one PGN including field 2 and 3 both filled with values.
Is this just for simulation or is this a "normal" behavior on a boat?

Is field 3: ground referenced always filled with a value?
I am just wondering, **why and from where there is coming a ground referenced speed** when tN2kSpeedWaterReferenceType is not  "Ultra_Sound" (which I think is ground referenced?)

ttlappalainen, 2017-11-29:
Water referenced has been measured by e.g. paddle wheel or ultrasound device (see tN2kSpeedWaterReferenceType). Ground referenced should come from GPS system. So if there is no current, speed should be same. In current conditions, which you can simulate with NMEA Simulator, values differs. So this is normal behaviour as far as I have understood.

If you have normal log, it fills only water referenced field and type (SWRT). In that case groud referenced will return NA value, which you can test on my library with N2kIsNA(val).

Ultrasonic is not ground referenced. E.g. Airmar has Ultrasonic speed sensors, which just measure speed of water particles by using ultrasound.

ronzeiller, 2017-11-29:
Thanks for the fast answer!
Aah, yes, now I remember this Ultrasonic Airmar...

Somehow strange. 
The measuring device e.g. paddle wheel sends a PGN.
The GPS is sending it´s own PGNs.
How there could come a mixed PGN with both values at one time? (on a real bus, who is doing the math?)

ttlappalainen, 2017-11-29:
Or maybe paddle wheel changes to ground referenced, when you sail to rocks:)

I think this is some kind of "could be". Some could make device with GPS and paddle wheel log. Then it could provide both speeds on this PGN. Or you can make kBox to send it!

ronzeiller, 2017-11-29:
:joy: 

Yes, this is why I was asking.
IMHO it makes no sense to read the "possibility" of ground referenced speed with the PGN128259  when we read out all the GPS PGNs anyway in KBox.

closed 2017-12-21 by ronzeiller