ttlappalainen/NMEA2000 #86: "not valid" or empty values in N2k

ronzeiller, 2017-12-21:
Hello Timo,
how should  **not valid** or **not existed** values be written in N2k sentences?

For example, if we want to write the PGN 127257 Attitude Yaw, Pitch, Roll and we do not have Yaw available what should be written?

Is there anything like in AIS where e.g. heading 361 degrees is synonym for "not valid" ?
In NMEA0183 it is just blank
In N2k?
When reading you return  `N2kDoubleNA` if a PGN-field doesn't contain numeric information, but could it also be written? Or does it depend on the PGN?

ttlappalainen, 2017-12-21:
Yes, it can be written. You have to provide right type N2kxxNA for function. See e.g. MessageSender example:
`SetN2kPGN127508(N2kMsg, 0, 13.8, 0.95, N2kDoubleNA, sid++);`
but for e.g.
`SetN2kPGN128275(N2kMsg,N2kUInt16NA,N2kDoubleNA,log,N2kUInt32NA)`

well probably it has been better to translate all real values to double and to conversion inside library functions.

ronzeiller, 2017-12-21:
Just got confused:

> Maretron SSC200 Solid State Rate/Gyro Compass Manual for example:
> 
> PGN 127257 – Attitude
> The SSC200 uses this PGN to indicate the vessel’s attitude (pitch and roll). The Yaw (field 2)
> is not used, therefore this field always contains 0x7FFF (data not available).

In NMEA2000library for a not valid yaw it is written `N2kDoubleNA (-1e9)`

ttlappalainen, 2017-12-21:
This is because I have device NaN value for -1e9, which is unprobable real value. Library functions will translate N2kDoubleNA for protocol to 0x7FFF. Also if you use ParseN2kPGN127257, it results N2kDoubleNA  (-1e9) for NaN value and you can test it with e.g. N2kIsNA(Yaw). On protocol it will transmit 0x7FFF, since the protocol only transmits integers and each field has its multiplier, which whould be used for changing it to real value. So there would be two options:
1. Have on parsers valid boolean like ParseN2kPGN127257(...double Yaw, bool YawValid,...).
2. Use it like I have now

If there would not be this special handling and it would just use multiplier, you would get for invalid yaw value 0x7fff*0.0001=3.2767. But for invalid EngineSpeed you would get 0x7fff*0.25=8191.75. So there would not be any standard way to test is data valid or not. 

With double values I mean that there is not much sense to have e.g. log as long integer. It just could be double and then handled inside library function to long integer. Now you may e.g. accidentaly convert log to km by log/1000, which cuts your value and you have define it (double)log/1000.

ronzeiller, 2017-12-21:
Thank you for this great explanation. Now I think I got it and it will help us a lot!

closed 2017-12-21 by ronzeiller