ttlappalainen/NMEA2000 #87: Getting TemperatureMonitor running on ArduinoDue

paulychris, 2017-12-30:
Hej
Your library is really great and comprehensive... However, i still dont get it running :-(
After unsuccessfully trying some time ago with a custom Atmel AT90CAN board (previous topic) i now obtained an original ArduinoDue board, hoping that it would run from scratch with this. But somehow i still dont have access.

I have an ArduinoDue connected to a Garmin GPSMap 720, and would like to display a tank level later. For the time being, i have installed the TempMonitor example, using NMEA2000, NMEA2000_due and  due_can libraries (latest versions) from this git repository. I got everything compiled without problems in the arduino ide, and uploaded the sketch.

But somehow the device does not appear in the "devicelist" on the Garmin gpsmap, and i think the address claim procedure is failing for some unknown reason.
I am pretty sure the NMEA2000 bus is cabled correctly. I checked with a scope, and i see quite some traffic from the GPSMAP 720, on both CANH and CANL line. I also see the traffic on the CAN_Receive pin on the arduino board, and i see a little traffic on the CAN_Transmit pin, which looks like the CAN Acknowledge bits, send for every received package. Its somehow correlated with the messages sent by the Garmin GPSMAP. But it looks like there is no outgoing messages from the arduinoDue.

The messages on the "serial monitor" say:
Start Initialize buffers
CAN device ready
Start address claim for device 0

and thats it.

if i put:
  NMEA2000.SetMode(tNMEA2000::N2km_ListenAndNode,23);
  NMEA2000.EnableForward(true);
Then i see a constant stream of messages with weird PGN (mostly 2048), but these even come in if i switch of the GARMIN GPSMAP, so i have no clue where they could come from.

[...]
5953 : Pri:0 PGN:2048 Source:211 Dest:127 Len:8 Data:C4,3F,8,0,0,50,0,21
15959 : Pri:0 PGN:2048 Source:197 Dest:63 Len:8 Data:0,0,0,B0,F,20,8,0
15965 : Pri:0 PGN:2048 Source:211 Dest:127 Len:8 Data:C4,3F,8,0,0,E0,0,21
15972 : Pri:0 PGN:2048 Source:38 Dest:46 Len:8 Data:0,0,0,B0,F,20,8,0
15979 : Pri:0 PGN:2048 Source:211 Dest:127 Len:8 Data:C4,3F,8,0,0,E0,0,21
[...]

To me it looks like there is still a problem with my installed version of the library, or maybe the interplay of the different libraries, but i am running out of ideas what and how to further test.
Was this recently tested on an ArduinoDue ?

And a last, probably unconnected observation:
If i try to change anything inside NMEA2000_CompilerDefns.h, uncommenting any of the #DEFINES there, then i can no longer compile.

Can anyone help, or have some advice ?
Can i somehow activate more debug messages on the serial monitor ?
Any other hints ???

Probably i am doing some stupid mistake. I am pretty sure its not on the hardware side...

Thanks for any help,
Christian

Chris00100, 2017-12-30:
Have you installed the CAN transciever in between? Due does not have the physical layer on-board.

Rgds

Chris00100, 2017-12-30:
BTW I am attempting the exact same thing also towards a Garmin, a 721xs. Not there quite yet as I have been concentrating on sending various temperatures up til now. The chip needed is MCP2551. An inexpensive 8 pin dil packaged IC.
/Christer

________________________________
From: Christer Hedlund
Sent: Saturday, December 30, 2017 11:40:52 PM
To: ttlappalainen/NMEA2000; ttlappalainen/NMEA2000
Cc: Subscribed
Subject: Re: [ttlappalainen/NMEA2000] Getting TemperatureMonitor running on ArduinoDue (#87)

Have you installed the CAN transciever in between? Due does not have the physical layer on-board.

Rgds

Chris00100, 2017-12-30:
Actually MCP2562 is a more recent version with better performance.

________________________________
From: Christer Hedlund
Sent: Saturday, December 30, 2017 11:47:35 PM
To: ttlappalainen/NMEA2000; ttlappalainen/NMEA2000
Cc: Subscribed
Subject: Re: [ttlappalainen/NMEA2000] Getting TemperatureMonitor running on ArduinoDue (#87)

BTW I am attempting the exact same thing also towards a Garmin, a 721xs. Not there quite yet as I have been concentrating on sending various temperatures up til now. The chip needed is MCP2551. An inexpensive 8 pin dil packaged IC.
/Christer

________________________________
From: Christer Hedlund
Sent: Saturday, December 30, 2017 11:40:52 PM
To: ttlappalainen/NMEA2000; ttlappalainen/NMEA2000
Cc: Subscribed
Subject: Re: [ttlappalainen/NMEA2000] Getting TemperatureMonitor running on ArduinoDue (#87)

Have you installed the CAN transciever in between? Due does not have the physical layer on-board.

Rgds

paulychris, 2017-12-30:
Brief update:

I tried once more, but downloading the (original ?) due_can lib (and can_common.h in addition) from collin80 git repository, instead of the (slightly modified ?) version from ttlappalainen git repository.

That works !!!!

So it seems there is some change in the version from this repo, which is not compatible at least with the Arduino-Due.

paulychris, 2017-12-30:
Yes, sure i am using a CAN transceiver chip inbetween, the MCP2562. 

Its really a software problem... the original can_due lib from collin80 works now !

Chris00100, 2017-12-30:
Ok great, good to know when I restart my efforts shortly!
/Christer

paulychris, 2017-12-31:
Good luck !
Here it works now. I put a potentiometer divider on one of the ADC inputs of the Arduino, to simulate my tanklevel sensor (analog voltage out, 0..5V). 
And i see the level live in the Garmin display. Cool !

So,
- what is causing the problem with the modified can_due lib from this git repo ?
- how can one deactivate eg the heartbeat in the NMEA2000_CompilerDefns,h without getting the following compiler errors:

-------------------------------------
...src/NMEA2000.cpp: In member function 'bool tNMEA2000::Open()':
---src/NMEA2000.cpp:767:35: error: expected type-specifier before 'tN2kGroupFunctionHandlerForPGN126993'
       AddGroupFunctionHandler(new tN2kGroupFunctionHandlerForPGN126993(this)); // Heartbeat handler
exit status 1
Fehler beim Kompilieren für das Board Arduino Due (Programming Port).
----------------------------------------

Those are the two issues i still have.
Apart from that: Really a great library, thanks to all contributors for this big effort to make NMEA2k so easily usable.

Best regards,
Chr

ttlappalainen, 2017-12-31:
Hi,

I am sorry for problems with Due. I just found that for some reason my update to NMEA2000_due was not updated to git. Now it is and it work with due_can under my repository. I have not yet checked colling80 due changes are all those compatible with NMEA2000. On NMEA2000 fastpackets are normally used and that requires that all packects will be send in order. CAN libraries often mixes them or send all packects through one slot, which causes performance problem.

paulychris, 2017-12-31:
Yeah, it works !
Even better than before: Before, sometimes the device description on the GARMIN display got corrupted during first seconds.
This seems to work perfectly now ! Thanks for the quick reply.

Only issue left is now, that changing any options in NMEA2000_CompilerDfns.h does not work for me. Am i doing still something wrong here ?
- But this is not so important. 

Thanks again.
Cheers,
Chr

ttlappalainen, 2017-12-31:
That is exactly what I meant with right frame order on fast packet. That was the fix I made for NMEA2000_can and due_can libraries.

With new release I also noticed that some defines on NMEA2000_CompilerDfns.h causes compilation errors. These defines are for minimizing RAM for e.g. Arduino Uno. But then you device does not full fill all NMEA 2000 requirements anymore, so I prefer not to remove any features, if you have enough ram.