ttlappalainen/NMEA2000 #88: NMEA2000_CompilerDefns.h -> uncomment fatal error at compil

OzOns, 2018-01-04:
Hi Timo,

After  happy New Year greetings ;-) and always many thanks for the N2K library, a little remark : 

when we uncomment 

#define N2K_NO_ISO_MULTI_PACKET_SUPPORT 1  // Uncomment this, if you do not need ISO Multi-packet support

in

NMEA2000_CompilerDefns.h

we got, at compil, a fatal error : 

In file included from C:\Users\xxxx\Documents\Arduino\libraries\NMEA2000-master\src/NMEA2000.h:50:0,

                 from C:\Users\xxxx\Documents\Arduino\libraries\NMEA2000-master\src/NMEA2000_CAN.h:46,

C:\Users\xxxx\Documents\Arduino\libraries\NMEA2000-master\src/N2kCANMsg.h: In constructor 'tN2kCANMsg::tN2kCANMsg()':

C:\Users\xxxx\Documents\Arduino\libraries\NMEA2000-master\src/N2kCANMsg.h:32:75: **error: class 'tN2kCANMsg' does not have any field named 'TPRequireCTS**'

   : Ready(false),FreeMsg(true),SystemMessage(false), KnownMessage(false), TPRequireCTS(false), TPMaxPackets(0) {

Thks again

Regards

Philippe

ttlappalainen, 2018-01-04:
I know. I'll fix it on next release. But why do you need that?

OzOns, 2018-01-04:
Sorry Timo,

I just saw on the previous thread that you are aware of this...

I am running on arduino Mega. but demanding lot of ram.. I have overflow on stack...

I want to spare the little tiny byte...

OzOns, 2018-01-04:
The library cost me +4100 bytes If I could spare 200 byte would be ok. hope so

ttlappalainen, 2018-01-04:
CHange the contructor temporaly to:
```
  tN2kCANMsg()
    : Ready(false),FreeMsg(true),SystemMessage(false), KnownMessage(false) 
#if !defined(N2K_NO_ISO_MULTI_PACKET_SUPPORT)
      ,TPRequireCTS(false), TPMaxPackets(0) 
#endif
    {
	  N2kMsg.Clear();
  }

```

OzOns, 2018-01-04:
It works 
Thks

class tN2kCANMsg
{
public:
  tN2kCANMsg()
  : Ready(false),FreeMsg(true),SystemMessage(false), KnownMessage(false) 
#if !defined(N2K_NO_ISO_MULTI_PACKET_SUPPORT)
      ,TPRequireCTS(false), TPMaxPackets(0) 
#endif

  {
	  N2kMsg.Clear();
  }
  tN2kMsg N2kMsg;
  bool Ready;  // Ready for handling
  bool FreeMsg; // Msg is free for fill up
  bool SystemMessage;
  bool KnownMessage;
#if !defined(N2K_NO_ISO_MULTI_PACKET_SUPPORT)
  unsigned char TPRequireCTS; // =0 no, n=after each n frames
  unsigned char TPMaxPackets; // =0 not TP message. >0 number of packets can be received.
#endif
  unsigned char LastFrame; // Last received frame sequence number on fast packets or multi packet
  unsigned char CopiedLen;
  
public:
  void FreeMessage() {FreeMsg=true; Ready=false; SystemMessage=false;

#if !defined(N2K_NO_ISO_MULTI_PACKET_SUPPORT)
  TPMaxPackets=0; TPRequireCTS=false;
#endif

N2kMsg.Clear(); N2kMsg.Source=0; }  
};

OzOns, 2018-01-04:
Also

#if !defined(N2K_NO_GROUP_FUNCTION_SUPPORT)	
    bool IsTxPGN=pNMEA2000->IsTxPGN(GetPGNForGroupFunction(N2kMsg),iDev);
    tN2kGroupFunctionPGNErrorCode PGNec=(IsTxPGN?N2kgfPGNec_RequestOrCommandNotSupported:N2kgfPGNec_PGNNotSupported);
#else
    tN2kGroupFunctionPGNErrorCode PGNec=N2kgfPGNec_PGNNotSupported;
#endif	

in N2kGroupFunction.cpp

for the 2 others comments

#define N2K_NO_GROUP_FUNCTION_SUPPORT 1 // Uncomment this, if you do not need Group Function support
#define N2K_NO_HEARTBEAT_SUPPORT 1 // Uncomment this, if you do not want device to send heartbeat

OzOns, 2018-01-04:
I spare 132 bytes in RAM only after compiling. I will see at execution the gain..

ttlappalainen, 2018-01-04:
Note that compiler does not show memory, which has been dynamically reserved. You can also queeze buffers. See "Memory requirements" on library main page.

I'll try to remember to check those. I prefer to forget those small memory boards for NMEA 2000. When you disable some features your device does not full fill NMEA 2000 requirements anymore. I prefer Teensy 3.2 for minimum. It is also smaller that Mega or Due.

OzOns, 2018-01-04:
I agree.
Yes for buffer shrinking... maybe later if I can't reach the goal.

I m looking for Raspberry ;-)

OzOns, 2018-01-04:
Compiler reserve static/gobal memory. -> realtime impact after compiling. 

Le croquis utilise 68450 octets (26%) de l'espace de stockage de programmes. Le maximum est de 253952 octets.
**Les variables globales utilisent 2468 octets (30%) de mémoire** dynamique, ce qui laisse 5724 octets pour les variables locales. Le maximum est de 8192 octets.

at execution dynamic alloc + stack

ttlappalainen, 2018-01-04:
NMEA 2000 library will reserve almost everything dynamically.

Rasperry is OK but:
- it draws much more current.
- I have not yet checked it's CAN driver. For NMEA 2000 fastpackets frames has to be sent in order. None of drivers (mcp_can, FlexCan, due_can) has done that right before my modifications.

OzOns, 2018-01-04:
Good. thks for reply :-)

Have a nice day Timo

ttlappalainen, 2018-01-07:
Now there is new release, which I compiled with all different cases. I run the tests only with all features included, since I do not have time to test all cases. Hopefully other cases works.

OzOns, 2018-01-08:
Thks a lot.

I can see the hard/deep work for that.

Question Timo :  I receive/read only (never send/write on canbus). Is there a way to spare dynamic memory ? (my obsession...)

ttlappalainen, 2018-01-08:
SetN2kCANMsgBufSize - If you do not have TP support, you can drop this to 3
SetN2kCANSendFrameBufSize - even you are not sending anything, library takes care on background of sending product information etc. Keep also configuration information short, so you can use 24 frame buffer size
SetN2kCANReceiveFrameBufSize - This depends how fast you are emptying frame buffer and how long messages you are expecting. Note that each call to ParseMessages moves frames from frame buffer to MsgBuffer. In max load there can be 25 frames/10 ms. If your bus load is about 20 %, you get 5 frames/10 ms. So if you expect 10 ms is maximum off time for ParseMessages, use at least 5 for 20% load. etc. I may have set 3 or 4 for minimum value because of bursts.

closed 2024-01-21 by ttlappalainen