ttlappalainen/NMEA2000 #89: Teensy3.6 sending but not receaving

astropigi, 2018-01-08:
Dear all,
I start to use your library with Teensy 3.6 with the Dual Can transceiver found on Tindie (https://www.tindie.com/products/Fusion/dual-can-bus-adapter-for-teensy-35-36/).  
The Flexcan CanTest (and the ObjectOriented test) works. Then I tried the WindMonitor coupled with a Raymarine screen and it works. 
This work with:
 tNMEA2000_teensy(uint16_t _DefTimeOut=4, uint8_t CANBusIndex=**1**); to use as default the can1
and 
pinMode(35, OUTPUT);
digitalWrite(35, LOW);
as first thing in the setup() to enable the can tranceiver resistors.

The problems start when I try to see the screen or the Airmar DST800(NMEA2000) sensor or another teensy with a WindMonitor with the DataDisplay2.ino or the DeviceAnalyzer.

With the Flexcan ObjOrientedextID I check that the bus work and there is traffic but for the DataDisplay2 or DeviceAnalyzer the Bus is always OFF (Not available or empty). Also with Actisense software (that it's work with the WindMonitor) as described in the nmea2000 library manual I can't see anything.

Seems that there are problems in receiving Nmea 2000 messages on teensy but is ok in transmission.

Have you got any hints?
Thanks a lot
Paolo

ttlappalainen, 2018-01-08:
I am using Teensy all the time and have not found any problem on receiving messages. I just tested both DataDisplay2 and DeviceAnalyzer with Teensy 3.5 on my test system and they both see TemperatureMonitor, DST800 and Garmin GMI data fine.

- Do you have Terminator resistors on the bus. You need at least one 120 ohm resistor. In full system you have to have 2 - one in both end of the bus.

- Which FlexCan library you use? Library under my repositories is tested. Collin80 has merged my changes. About others I can not say anything.

astropigi, 2018-01-08:
Terminator resistors are built in the dual can transceiver (and also in the Raymarine connectors). Anyway, when  I send the data to the screen it works.
I use the FlexCan library from your repos.

This is what I see with the ObjectOrientedCaNextended

1 ID: 19FD0816 Data: 1 1 4 7D 73 FF FF FF 
0 ID: 19FD0716 Data: 1 44 7D 73 FF 7F FF FF 
1 ID: 19FD0616 Data: 1 C1 70 FF FF FF FF FF 
0 ID: 19FD0816 Data: 1 1 4 7D 73 FF FF FF 
1 ID: 19FD0716 Data: 1 44 7D 73 FF 7F FF FF 
0 ID: 19FD0616 Data: 1 C1 70 FF FF FF FF FF 
1 ID: 19FD0816 Data: 1 1 4 7D 73 FF FF FF 
0 ID: 19FD0716 Data: 1 44 7D 73 FF 7F FF FF 
1 ID: 19FD0616 Data: 1 C1 70 FF FF FF FF FF 
0 ID: 19FD0816 Data: 1 1 4 7D 73 FF FF FF 
1 ID: 19FD0716 Data: 1 44 7D 73 FF 7F FF FF 
0 ID: 19FD0616 Data: 1 C1 70 FF FF FF FF FF 
Total Received Messages in 10 Sec:
12

When I try with the Datadisplay2
CANGetFrame(canId,len,buf) is always FALSE

ttlappalainen, 2018-01-08:
Have you tried with default settings by using Can0. I have never tried Can1

astropigi, 2018-01-08:
The problem is that the pcb of the dual can transceiver is connected to the alternative pins(29,30) and I don't understand how to change it (with can0.begin(250000, defaultmask,1,1)) in your library NMEA2000 (while is quite clear how to do that in the FLEXCAn examples).

astropigi, 2018-01-08:
I change the setup in order to use the CAN0 and it works but only for the first iteration.
This is what I see:
CAN device ready
Running...1600 : Pri:5 PGN:130311 Source:35 Dest:255 Len:8 Data:C1,C0,F6,73,FF,7F,FF,FF
In Main Handler: 130311
After this it stops to work.

Why? 

PS if I use the (for example) the wind monitor I see the sentence of both instuments (the virtual one and the real one).

astropigi, 2018-01-15:
Dear Timo,
I have done few steps more and I don't know how to go further. 
Actually, when I try to read from the bus the SOW it's work but only one time
CAN device ready
Running...o
1600 : Pri:2 PGN:128259 Source:35 Dest:255 Len:8 Data:50,0,0,FF,FF,0,FF,FF
In Main Handler: 128259
Boat speed: SOW:0.00, SOG:not available, Paddle wheel

Why?
The update rate is 1000ms.

On the bus, there is only the teensy 3.6 and the DST200 (Depth, Speed, Temperature) with right connections (it's the Raymarine bus with backbone resistors ecc, ecc).

I don't understand where  it's stop and why.

Really thank you for the help.
Paolo

ttlappalainen, 2018-01-16:
Difficult to say. I tested example DataDisplay2 as it is on Teensy 3.5 and there is no problem. It prints all messages - 128275, 128267, 130311. Except it does not parse 130311. 3.5 and 3.6 are equal exept on memory size.

My bus driver is connected to can0 (pin 3 and 4).

Are you sure you did not install FlexCan library with TeensyDuino? Check from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries that you do not have FlexCan there. If you have, remove folder.

astropigi, 2018-01-16:
I am using the CAN0 at alternative pin (it's work, with the correct frequency of 1Hz (from DST200 manual) when I use the example in your flexcan library ). 

Maybe I don't understand something of the library. I need a streaming of data and not only the first reading. Sorry for the trouble. Maybe I am using bad the library.

MohammedAlaloosi, 2018-10-09:
hello @ttlappalainen i am using teensy 3.6 and the dual CanBus transceiver i found on 
[https://www.tindie.com/products/Fusion/dual-can-bus-adapter-for-teensy-35-36/](url)  

i need a help on how to connect the hardware properly i have been trying to get this working for the past three weeks. all i want to do is to send a message from a CanBus and receive it on a different CanBus

ttlappalainen, 2018-10-09:
There are few things. If you want to use my library, it is in your case best to fix it to Teensy. So instead of having:
#include "NMEA2000_CAN.h"

use
#include <N2kMsg.h>
#include <NMEA2000.h>
#include <FlexCAN.h>
#include <NMEA2000_teensy.h>

Then you can define:
tNMEA2000_teensy NMEA2000_1();
tNMEA2000_teensy NMEA2000_2(4,1);

Then you should have two library objects, which you can initialize. The only problem is that I have never tested two objects at same system. But I know that it is possible to use second port.

The last thing is that your adapter uses alternate pins for CAN0. As default Teensy 3.5/3.6 has CAN0 on pins 3/4. So you have to change Teensy pins to get CAN0 working. I have not populated that functionality to NMEA2000_Teensy. You could try does it work, if you change pins on setup():
            CORE_PIN29_CONFIG = PORT_PCR_MUX(2);
            CORE_PIN30_CONFIG = PORT_PCR_MUX(2);
See FlexCAN.cpp void FlexCAN::setPins

And please use FlexCAN under my git. Delete all other FlexCAN installations possibly installed with Teensyduino.

Guruzillo, 2021-10-19:
I think I have a bit more simple question or don't follow the solution on the answer directly above.    I am using Teensy 3.5 and there is no problem using pin 3/4 combo.  CAN interface is working fine.   How do I change the Pins to 29/30 as I have built an entire circuit board around using  29/30 verse 3/4.  I don't get how then commands will work:

CORE_PIN29_CONFIG = PORT_PCR_MUX(2);
CORE_PIN30_CONFIG = PORT_PCR_MUX(2);

ttlappalainen, 2021-10-20:
FlexCAN will overwrite your setting on begin. Currently only way it to force pins on FlexCAN.cpp line 181:
    setPins(1,1);
Hopefully you use Visual Studio Code or some othet better environment than Arduino IDE so that you can have project specific libraries.
I'll add pin definition for NMEA2000_Teensy later.

SmartSeas, 2022-10-13:
Because of supply constraints we are moving to the teensy 4.1 board:(   We have installed the new library NMEA2000_Teensx and are attempting use it on the 3.6 Teensy, however, we cannot seem to get Pins 29 and 30 working.   We used your suggestion above and it worked perfectly for the NMEA2000_Teensy library on Teensy 3.6.   Do you have any suggestion how to use Pins 29 and 30 on Teensy 3.6 with NMEA2000_Teensyx as we get ready to transition and rebuild our HW for use of the 4.1 board Pins 30 and 31.    Any help would be greatly appreciated.

ttlappalainen, 2022-10-14:
[src.zip](https://github.com/ttlappalainen/NMEA2000/files/9782805/src.zip)

Try with above. I tested compiling, but not with hw.

If you use NMEA2000_CAN.h, you have to modify line 157
`tNMEA2000 &NMEA2000=*(new tNMEA2000_Teensyx(NMEA2000_TEENSYX_CAN_BUS,NMEA2000_TEENSYX_TX_PIN,NMEA2000_TEENSYX_RX_PIN));`

and add defines before including it:
```
#define NMEA2000_TEENSYX_TX_PIN tNMEA2000_Teensyx::pinAlternate
#define NMEA2000_TEENSYX_RX_PIN tNMEA2000_Teensyx::pinAlternate
```
Report if it works so I can update library.

SmartSeas, 2022-10-14:
Tried to run an example from the sample(Temperature monitoring) with latest library. by referring usage of the library that is mentioned in the README https://github.com/ttlappalainen/NMEA2000_Teensyx.
In the NMEA2000_Teensyx library the CAN related implementations are available in the NMEA2000_Teensyx.cpp. So NMEA2000_CAN.h is not needed right?
With that assumption I have altered the sample as follows        

#include <Arduino.h>
//#include <NMEA2000_CAN.h>  // This will automatically choose right CAN library and create suitable NMEA2000 object
#include <N2kMessages.h>
#include <NMEA2000_Teensyx.h>

Compilation got success but while running the code it gives error as sending failed.
if its a wrong approach, can you advice the right way? 

Note:
Board used : Teensy 3.6

ttlappalainen, 2022-10-14:
Did you then added constructor:
```
#include <Arduino.h>
#include <N2kMessages.h>
#include <NMEA2000_Teensyx.h>

tNMEA2000 &NMEA2000=*(new tNMEA2000_Teensyx(tNMEA2000_Teensyx::CAN0,tNMEA2000_Teensyx::pinAlternate,tNMEA2000_Teensyx::pinAlternate));

```

SmartSeas, 2022-10-16:
But the constructor didn't expect for 3 arguments.
 tNMEA2000_Teensyx(tCANDevice _bus=CAN1);
this is the constructor available in NMEA2000_Teensyx.h. and expecting only one arg.
So, I have called the constructor as mentioned below.
 tNMEA2000 &NMEA2000=*(new tNMEA2000_Teensyx(tNMEA2000_Teensyx::CAN0));
If you have any suggestion, please let me know. Thank you

ttlappalainen, 2022-10-17:
Did you read my comment 3 days ago? There is src.zip with updated code and instructions.

ttlappalainen, 2022-10-17:
I found my Teensy 4.0 and tested the chance. Unfortunately it does not work. There is something wrong with pin mux setting.

ttlappalainen, 2022-10-17:
Fix: It works as I described. My alternate Tx was not soldered. One problem is that alternate Rx pin is also led pin, so you led will lid all the time.

ttlappalainen, 2022-10-17:
Sorry I mixed. I tested with Teensy 4.0. Have to find free 3.6. But would expect that it works too.

SmartSeas, 2022-10-17:
Hello Timo, Thank you so much for helping us out on this.   We are up and running again.  I am including some pictures of our project here.   It is a digital switch with the following functions:  Switch Banks, LED RGBW ligthing control, and Gauges.   The next todo on our list is getting the 127501 and 502 PGN to allow our switch bank to be controlled on an MDF.    Would love to chat not sure how to contact you directly. 

Again Thanks.

Mike 

[image]

[image]

[image]

[image]

ttlappalainen, 2022-10-18:
Email works best.

Hopefully you have isolated tranceiver on your board.

One problem with digital switchin is that it is not well supported on MDF:s. Some MDF:s supports CZone, but I do not know how that communicates - I mean that I know it uses proprietary PGN:s, but no idea of their contents.

SmartSeas, 2022-11-03:
Hello Timo,

We have been able to bring up the newer NMEA switching imbedded in the garmin and change status of switches and the change voltage to a pin so in effect we should have a fully operation switch integrated to garmin.  I have an engineer that will clean up our modifications to your lib and share back.   I am new to use of  free libraries so any guidance is welcome.

Your question, The transceiver pulls voltage from switch and not NMEA bus.  But that I can change easily on the next spin of the board to isolate it.

Just want to say we appreciate you code and responses.

Thanks.

Mike Chouinard

ttlappalainen, 2022-11-03:
If those changes are Garmin proprietary, please make own module for it and do not use general messages.