ttlappalainen/NMEA2000 #90: Compile error on NMEA2000_teensy

steveh57, 2018-01-10:
Hi Timo, sorry this is in the wrong place but there doesn't seem to be an option to open an issue on NMEA2000_teensy.  I'm getting a compile error on NMEA2000_teensy.cpp line 74 when compiling for a teensy 3.5.  

'CAN_message_t' has no member named 'rtr'	

I'm guessing line 74 is supposed to be out.timeout=0, but I'm not really sure what this code is doing so perhaps you could check.

Thanks, Steve.

ttlappalainen, 2018-01-10:
I think you have wrong FlexCan library. Remove FlexCan library, which comes with TeensyDuino and install the one under my repository.

steveh57, 2018-01-10:
Apologies, I thought I had.  I downloaded your library but hadn't updated the reference.  Now all fixed.
Thanks for quick response.

closed 2018-01-10 by steveh57