ttlappalainen/NMEA2000 #92: Merging data from two physical N2K devices on PGN127489  [Engine Parameters] ?

virtuvas, 2018-01-18:
hello Timo et al,

more of a Question but I'm stuck at the moment and want to prepare code and test it over the weekend on the boat.
I have a No Land Engineering RS11 engine monitor setup (before finding you and your work, otherwise I wouldn't buy it!) which gives to the bus

- Oil Pressure
- Coolant Temp
- Engine Boost
- RMP

for two engines.

Works fine and Garmin GMI10s and GPSMAP4008 see and present the data.

Now, working on a teensy which will take: 

- exhaust gas temperature
- gearbox oil pressure
- gearbox oil temperature

and pass them onto the bus.
So two physical devices, two different addresses, both sending PGN127489 data to the bus.
Q::
**A.** in my teensy, should I first read the bus for this message (effectively picking the RS11 info), merge my 3 data (x2 for the two engines...) and resent it to the bus?
or
**B.** just formulate the sentence and sent my 3 data (x2) and hopefully they will merge with the others?

Note, I wont be sending the SAME info, just adding new info to the SAME PGN. Haven't done that before, so bit confused.

Afraid that on option two I'll have blinking data for all 6 parameters, or a massive messup!

thanks!

Vassilis

ttlappalainen, 2018-01-18:
B: is allways the right solution. If you provide same PG with additional data, you set NA to all, what you do not provide. But also if there will be same PGN with same data, normally on MFD you choose device, which you would like to show. That is why it is important to set unique number different on each own device as it is documented on samples.

I am bit lost with your description. For "exhaust gas temperature" you should use PGN 130316 (SetN2kTemperatureExt), since others are obsolete and should not be used. If you Garmin does not support that use 130312 (or both). For gearbox pressure and temp you use PGN 127493. In both cases you have to set instance parameter same as in PGN 127489 (defaults are 0=Port,1=Starboard). So there would not be any way to merge them with RS11 info.

virtuvas, 2018-01-18:
thanks Timo!

OK, I also prefer solution B! 
Just to clarify, how do I sent NA (not available) to a field? leave blank? All the things I've done todate are on sentences that you had to fill everything else it wouldn't work.
 
For example, say I want to add to the enginedynamicparams engine oil temp, alternator voltage and fuel rate on top of the RS11 coolant temp, rpm, oil pressure and boost. 
Having defined and given the right values to: 
engineinst
Oiltempval
VoltagefromAlt
instfuelcons

do I fill all the others with blanks and the following works?

SetN2kEngineDynamicParam(N2kMsg, engineinst, , Oiltempval, , VoltagefromAlt, instfuelcons, , , , , , , , , , , , , , , ...how many they are (MANY!!!) )
or do I put zeros? 
SetN2kEngineDynamicParam(N2kMsg, engineinst, 0, Oiltempval, 0, VoltagefromAlt, instfuelcons, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, ...how many they are (MANY!!!) )
or FALSE for the two dozen last parameters which are booleans?

I'd have thought 0 will conflict with the values sent by the RS11, and not sure on FALSE hence the original question.

You are right the other fields I want to add go to other sentences, I just forgot to add all the fields I wanted and it seems I picked up the wrong ones...

thanks
V.

Chris00100, 2018-01-18:
Hi I have the exakt same question for this PGN!

ttlappalainen, 2018-01-18:
You can not leave parameter empty on c++. You allways give some value. If you look definition for SetN2kEngineDynamicParam, it has default parameter starting from EngineCoolantPress. So the rest you do not need to define.

So you write call:
`SetN2kEngineDynamicParam(N2kMsg, engineinst,N2kDoubleNA, Oiltempval,N2kDoubleNA, VoltagefromAlt, instfuelcons,N2kDoubleNA);`

Note that you have to choose right NA definition according to type. So for double you use N2kDoubleNA, but e.g. uint16_t you use N2kUInt16NA and int16_t N2kInt16NA.

virtuvas, 2018-01-18:
brilliant!

did my prototype at home with a GMI10 and it works.
Now I'll test it during the weekend at the boat with the engine data running as well.

thank you very much!

V

closed 2018-01-18 by virtuvas