ttlappalainen/NMEA2000 #96: MessageSender/DataDisplay2 does not work as intended

CapitanoPedro, 2018-02-03:
I just started playing with the NMEA 2000 libraries. Great work especially from Timo ... thanks!
At first I connected two Arduino Megas (with MCP2515) as desicribed. I.e. One is running MessageSender and the other is running DataDisplay2. Did not get any packet on the receiver side.
After that, I connected an NMEA-Multiplexer (ShipModule) => all is fine, I could see the packets. With that, the transmitter (MessageSender) seems to work.
With some debugging in DataDisplay2 I discovered, that with setting "DEBUG_RXANY (1)" in MCP_CAN.cpp all is fine.
In the default way MCP is programmed to use filtering but no filters are set. And consequently, no message appears in the buffers.
There is a good chance, that I have not got all details (I am new with that technology), but simply putting together MessageSender//DataDisplay2 does not work instantly.
Any idea?

ttlappalainen, 2018-02-03:
I run my Mega board with mcp_can ( under my repository) default settings so DEBUG_RXANY undefined and it works fine. First thing comes again in to my mind that do you have terminal resistors connected? You need at least one 120 ohm resistor on small test system (short wires), but you can have two as in real CAN bus implementation.

ttlappalainen, 2018-02-03:
The filters are not in use. You need them, if you want to catch just specific frames. Here we catch all. And please do not be so sure. This code has been used by several users and I also have Mega on my test bed and I have never had any problems. Did you really download mcp_can from my repository and deleted all others - if installed. Compiler should warn you, if you have multiple of same libraries installed.

Have you checked that your interrupt pin is same as set on my samples? If not, you have to set that with #define N2k_CAN_INT_PIN xx.

MessageSender seem to be as default non interrupt mode = no definition for N2k_CAN_INT_PIN

ttlappalainen, 2018-02-06:
So does that mean that I have been just lucky on not having any problems with mcp_can?

Does the fix effect enything for the use mcp_can with or without extended? If not, you could make PR for mcp_can.

CapitanoPedro, 2018-02-06:
The error is odd, because it depends on the (uninitialized) content of the six filter registers.
Some people are happy, and others get mad (see also the thread in my comment, above).

You may do following checks ..
a) print the contents of the mask registers directly after mcp2515_init. You might observe, that these registers report a random are random content.
b) fill the 6 registers with 0x00000000L. I think at this point you might see, that the lib does not work anymore. None of the EXIDE bits is set, and with that the filtering will block the messages.
c) Now, load the filters with 0x00080000L and you might see, everthing is working well.

For whatever reason, the filters are not part of the chip internal RESET sequence, and some customers might have problems and others not.

I am convinced (ok ok ok ... NOT shure :-), that mcp_can should have an improvement similar to my solution. I also think that filling registers after reset with defined values is a good way. Note, that the datasheet of MCP2515 does NOT denote a reset state for these registers (only for the control registers).
Also, filling the registers with my solution is at least better, than leaving uninitialized. Best would be, if the developers of mcp_can have a closer look to this.

Could you pls. give me the correct link to the mcp_can guys. For me it is not clear where this lib originates, just took it from you repository. I would get in touch with them foir further discussion.

Thanks and greetings

ttlappalainen, 2018-02-06:
If I read everything right, Seed_studio (https://github.com/Seeed-Studio/CAN_BUS_Shield) same version as I have, except someone has enabled debug definition. So you could try if they answers.

By the way you should see the link where is has been originally forket under repository name.

nyinyinyanlin, 2018-04-24:
Thank you so much CapitanoPedro!!!

I had the same issue. Arduino Mega2560 <-> MCP2515 TJA1050 <-> Garmin Intelliducer N2K Depth Sounder

My oscilloscope showed me that the transducer has been transmitting two message frames at regular intervals and the differential signal is clean. I have also checked TJA1050 CAN Transceiver RX and the differential signal is properly being converted to logic level signal.

I couldn't figure out what was happening with the CAN Controller as everything ahead of it is running properly in receiving side. And I tried sender example and the device is transmitting properly on the CAN Bus.

I have found your issue and gave it a try, and I'm now receiving depth data from my Intelliducer properly. 

And Timo, thank you so much for your amazing work. I wouldn't have the capability to integrate NMEA2000 devices in my project without your hard work.

Thanks and greetings to all.