ttlappalainen/NMEA2000 #97: ESP32 support?

kephaII, 2018-02-07:
Hi
I tried this a ESP32 DEVKIT C board, with Aruduino IDE, but lots of compiler complains.

It would be very nice to use this nice library with ESP32 DEVKIT C.
Is there plans for that?

Best regards

Hans

ttlappalainen, 2018-02-08:
No plans currently. I found rather little information about CAN on ESP32. By googling found that CAN module should be SJA1000 compatible and Tx is on GPIO32 and Rx on GPIO33 (see https://github.com/eerimoq/simba/commit/46778af636a4fe58773753b5ed9da83916cfaa06). The driver code does not look that bad, but it would be best to write completely to own NMEA2000_ESP32 module without using any external library.

By using external MCP2515 based SPI module you could use library by forcing it to use mcp_can.

kephaII, 2018-02-08:
Thanks for fast answer.

I tried with this library 
https://github.com/ThomasBarth/ESP32-CAN-Driver  
http://www.iotsharing.com/2017/09/how-to-use-arduino-esp32-can-interface.html
The examples worked.
It also communicates with the Mega with a TJA1050, 16Mhz modules.

Could that canbus library be utilized for the controller function?

Seems like the compiler errors are due to nameing conflicts.

My first challenge is to send the heading from the ESP32 to the Navico 4g radar nmea2000 interfase.
The above mentioned library is seemingly capable of the sending extended frames, 
but given a 29 bit header and not a 32bit one, i feel it is a little tricky to use that library... without your excellent library as front end for the NMEA2000 headers.

Coming from Arduino, the ESP32 was a heaven sent. 120 times the speed, plenty memory, wifi and BLE, which works out of the box. wow!

Best regards

Hans

jabadabado, 2018-03-31:
Hi,
I am also looking for something to get started with the ESP32 and NMEA 2000 via the CAN bus. Thank's for both leads!

Best regards

ttlappalainen, 2018-03-31:
I am waiting ESP32 module, which I'll try.

mrbubble62, 2018-03-31:
+1 for ESP32 CAN 
This would be great.

kephaII, 2018-04-01:
Hi
Once i had made a formatting function for my telegrams,
 then the communication worked like a charm. 

I use a simple tranciever only, since the ESP32 has built-in controller. 

Full library would be preffered... 

Best regards 
Hans

ttlappalainen, 2018-04-06:
There is now first version for ESP32 support. E.g. example NMEA2000ToNMEA0183 works fine with it.

olesenrl, 2018-04-06:
> Reply: ttlappalainen/NMEA2000 <reply@reply.github.com>

Cool! Thanks!

jabadabado, 2018-04-07:
Fantastic!

Taglio81, 2018-06-26:
hello, congratulations for the work done, I'm passionate about this library, but unfortunately the lack of programming knowledge block me ..
I would like to make a display (probably with 3.2 "nextion) to display engine data in my boat with evinrude etec.
I'm trying to study DataDisplay and DataDisplay2, using esp32. Would any expert kindly help me with the sketch? I should also use a second serial of the ESP32 (I imagine with myserial) to send the values to the display

Thank you so far

ttlappalainen, 2018-06-27:
DataDisplay2 gives a basic example how to read data from NMEA2000 bus. So instead of just printing it to serial (like I used function PrintLabelValWithConversionCheckUnDef), you need to write functions to write data to the screen. When you have DataDisplay2 working and showing NMEA2000 data to serial on your desk, save as new sketch EngineMonitor under your own scketshes and play with that and learn how to show information you need and remove unnecessary code. Then take some Nextion "Hello world" example and study how you can e.g. show some incrementing value on specific position on screen e.g. by writing function:
void ShowOnScreen(int x, int y, double value) {
// Commands to show double value on Nexition display on position x,y
}
Then you go back to your EngineMonitor sketch and add necessary functions from your Nexition sketch and start to play with that.

It sounds easy or messy depending your programming skills.

Taglio81, 2018-07-04:
hello Timo, I'm doing some tests on the bench, unfortunately the connectors for my existing n2k network and the controller SN65HVD230, MCP2551 and TJA1050, I have yet to get.
I managed to implement the serial1 with HardwareSerial.h on ESP32, you would have the opportunity to help me with DataDisplay2.ino, I would like to use the native USB serial with your NMEA_Simulator and read the outputs on the second serial I have activated, or vice versa. So i can develop the nextion code from home, the boat is 100km away.
many thanks in advance

Taglio81, 2018-08-20:
hello everyone, the 3.5 "nextion display is continuing, with ESP32, a second hardware serial (so I keep the usb on board for debugging) and a shift level, I send the data to the 5v serial display.
The data that I can read from the engine are varied and for now I am very satisfied.
For the case of the display for now I'm using an IP68 spotlight, to which I removed LEDs and drivers, it is not nice, but it should resist!
A question, connecting the esp32 to the N2K T bus, before starting the engine, works, but disables the original Evinrude instruments, while if I connect it with the engine already on, they both work. Do you have any ideas? Reading the references of the library, if I have done well, the default mode is ListenOnly, eventually where can I check?
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ttlappalainen, 2018-08-20:
So did I understood it right that you are not sending any data to the N2k bus? 

There is SetMode method for changing more from default N2km_ListenOnly. My ESP32 example seem to set it to N2km_ListenAndNode.

If you do not send anything to the bus, it should not effect to the Evinrude instruments. How about terminating resistors? With too many or too less terminating resistors one generates funny effects.

You should try with simple Teensy/Arduino board with example ActisenseListener.ino does it effect anything. If it does not, then you gould spy traffic while turning system on ESP32 connected or not.

Taglio81, 2018-08-20:
It is correct, the project provides only the reading of data.

The example I used as a start is DataDisplay2.ino, but the SetMode is not included in the setup, so I hypothesized that the default is in ListenOnly. Possibly what would be the correct command to force it?
NMEA2000.SetMode (tNMEA2000 :: N2km_ListenOnly);
right?

The termination resistors are present, those of series, I procured a standard T and a connector.

Eventually when I go back to the port with the PC I do further tests, to understand what happens on the BUS.

As a solution, what do you think of a delay of 6 / 7sec before starting the esp or possibly a manual switch

ttlappalainen, 2018-08-20:
It should use default ListenOnly. NMEA2000.SetMode(tNMEA2000::N2km_ListenOnly); is correct.

If you do not have extra Teensy/Arduino, you should be able to run ActisenseListener on ESP32. Then you see that does Evinrude instruments work, when you have that connected.

Manual switch is a bit stupid. First should find, what causes Evinrude instruments locking. Delay is OK, ut since device is listen only, it should not effect. Since you are not sending, you could also try disable transmitter MCP2562 by pulling pin 8 to Vcc.

tmcadam, 2018-09-17:
I am hoping to try the ESP32 soon. I can see some people have tried different transceivers in this thread. It would be useful to know which is the recommended CAN transceiver to use with ESP32? Thanks

ttlappalainen, 2018-09-17:
I have been using MCP2562 on all my systems - also with ESP32 - and newer had problems. So I can not say how other tranceivers works.

Taglio81, 2018-09-17:
in my project, I started with MCP2551 the first transmitters arrived, with logic level converter 5v-3,3 fors esp32, and the system worked immediately. In the meantime, they arrived also SN65HVD230 and TJA1050, but not yet tested.

ttlappalainen, 2018-09-17:
MCP2562 has build in logic level converter.

MENIER, 2019-06-11:
I have tested https://github.com/MENIER/nmea2k/tree/master/DeviceAnalyzerESP32
on my RO6800 VHF nothing....
I send 'u' char for analyse and answer is '*************************************'
(I have tested the pinout it is ok with :https://github.com/MENIER/nmea2k/tree/master/test_RXD_TXD_pinoutEsp32 )
Don't know what to do more?

Taglio81, 2019-06-12:
Hi Menier, I don't know if it can help you, but in my notes I found that using ESP32 Dev kit the pins used are:
- GPIO16 -> CAN TX
- GPIO4 -> CAN RX

eventually waiting for official information from Timo

MENIER, 2019-06-16:
Hi taglio81, I don't understand the ESP32 CAN TX RX has fix pinout ?

    GPIO16 -> CAN TX
    GPIO4 -> CAN RX

ttlappalainen, 2019-06-19:
Sorry for late answer. ESP32 RX/TX pinout can be selected with defines before include as below.
```
#define ESP32_CAN_TX_PIN GPIO_NUM_32
#define ESP32_CAN_RX_PIN GPIO_NUM_34
#include "NMEA2000_CAN.h"
```

Some pins can not be used for TX - e.g. GPIO36-GPIO39. These has been described on datasheets.

AK-Homberger, 2019-07-22:
Hi Timo,
Many thanks for the ESP32 support. It works great.
https://github.com/AK-Homberger/NMEA2000WifiGateway-with-ESP32 

BTW: The AZDelivery ESP32 NODE MCU is using:

GPIO2 -> CAN TX
GPIO4 -> CAN RX

Regards,
Andreas

jimarcher, 2019-08-03:
Hello Timo...

I saw you are using that ESP32 board from Paul in Czechia. I'm trying to do exactly what you did, that is, log all the N2K data to a memory card. Would you be willing to add that code to the examples?

Also, with this board, I see he has large vias for CAN H & L, and also 12V and GND. Is the 12V via intended to be connected to the 12V on the CAN bus to power this board? I just wanted to be sure before I destroyed my board, it took months to get. 

Thank you!

ttlappalainen, 2019-08-03:
Hi,

I have not made any data logger. I have used only ESP32 to convert N2k to wifi. That board did not had any 12V input either. So I do not know what board you mean.

Then NMEA 2000 bus has 12 V voltage, which can be user to power devices. If your board has 12-16 V power input, you can  use N2k power. Note that normally N2k is powered with boat batteries, so it can rise up to 15 V depending of your charger system.

jimarcher, 2019-08-03:
My apologies, I must have mistaken you for someone else. The board I referred to is the one linked below, and someone said they were able to use it to get an N2K data logger running in an afternoon. I expect using your library I can do that as well. Thank you for your quick reply and your generous effort with this library!

[https://www.tindie.com/products/Fusion/can32-an-esp32-dev-board-with-can-bus-v2/](url)

ttlappalainen, 2019-08-04:
According to docs you can power it directly from N2k power lines. Anyway I did not found that how powering has been organized. So can it have N2k power and USB at same time. E.g. powering Arduino Due at same time from USB and 12V requires diode on 12V side. See https://github.com/ttlappalainen/NMEA2000/blob/master/Documents/ArduinoDUE_CAN_with_MCP2562.pdf

Also CAN H and CAN L can be connected directly to the N2k bus CAN H and CAN L.

I did not understood from docs on which pins are CAN Tx/Rx. But note that it has also sleep mode for tranceiver, which should not be enabled, otherwise tranceiver will not transmit anything.

jimarcher, 2019-08-04:
Thanks again, especially for the warning about the 12VDC, that had not occurred to me.

sarfata, 2019-08-04:
Hi @jimarcher - I believe that was me ;) What I did was create a simple project based on the espidf framework. I just tried compiling it again but it does not build with recent versions of platformio/espidf.

What I did afterwards is port the source code of my kbox project to a new esp32 board that I designed. The source code for it is in the [esp32 branch of kbox-firmware](https://github.com/sarfata/kbox-firmware/blob/esp32/src/esp32/main.cpp). It is very much "prototype quality", not well tested and with a huge main.cpp that includes lots of tests for my new hardware board but it does logging, signalk and tcp streaming. It should be relatively easy to make it work on the can32 board.

sarfata, 2019-08-04:
Timo is right that the documentation is very lacking for this board.

The pins are:
```
#define ESP32_CAN_TX_PIN GPIO_NUM_5
#define ESP32_CAN_RX_PIN GPIO_NUM_4
#define GPIO_CAN_DISABLE ((gpio_num_t)16)
```

You need to pull `GPIO_CAN_DISABLE` low to enable the module.

Also there is not a lot of information on how the power supply is built on this board. There is indeed a risk of 5v leaking to the 12v. You should use one or the other but not both. 

I think the biggest risk would actually be to have different grounds on your boat and if you plug USB and NMEA2000 power at the same time, you will force them both to the same level _via_ the CAN32. In practice I have done that many times without issues but be careful, especially if you have a laptop plugged onto shore power or an inverter while you do this.

jimarcher, 2019-08-06:
Hello @sarfata thanks VERY MUCH! I have a project I need to get going and I thought that board would be an easy way to jump in, and thanks to you and Timo it just may be after all. I'll check out your source carefully, and thank you for the pin info and the voltage warning as well!

jimarcher, 2019-09-14:
Hello @sarfata I'm having trouble getting the Arduino SD library to work with this board. I have tried the ESP32 examples but I'm not sure if the pins are set correctly in the library and I don't know how to change them. Can you offer some advice? I'm just trying to do what you did, use this N2K library to log data to an SD card. Thanks!

sarfata, 2019-09-14:
@jimarcher I think I have only used this board with the espressif SDK (not Arduino). In my code I was initializing the SD card as a SPI card and the pins were:

```
#define PIN_NUM_MISO ((gpio_num_t) 19)
#define PIN_NUM_MOSI ((gpio_num_t) 23)
#define PIN_NUM_CLK  ((gpio_num_t) 18)
#define PIN_NUM_CS   ((gpio_num_t) 17)
```
If you include `SD.h` and then call `SD.begin(SDCARD_CS_PIN /* 17 */)`, i think things should work immediately but you can debug this by printing `SD.cardType()` to try to figure out why it's not working.

jimarcher, 2019-09-14:
Thanks @sarfata ! Actually I had to do SD.begin(2) to get it working. The datasheet shows the CS pin is on GPIO2 which is pin 22. However, your SD.begin(parameter) tip is what got me going so thank you!

ronzeiller, 2019-11-20:
(Moved your answer from https://github.com/ttlappalainen/NMEA2000/issues/154#issuecomment-553715371, because this thread already provides nice informations on ESP32 in general)

> One thing with ESP32 is that you need good power to use WiFi. We have 0.5 A PSU feeding ESP32 and just on power pin of ESP32 470 uA capacitor. Otherwise it rebooted itself by WiFi transmitting. In isolated systems we use ISO1050 and on those, which does not require isolation (=not connected any other devices or connection is isolated) MCP2562. Have not seen problems with communcation. But to get everything work, I passed default WiFi library.

Do you mean 5V PSU and the Capacitor on 3,3V PIN?
Or is it good enough to have the capacitor on the 5V side?

ttlappalainen, 2019-11-20:
Capacitor on 3.3V pin - as near as possible. I have it on 2 mm away.

Also found that ADC give different values WiFi On or WiFi off. I got it working better by setting analog clock div to 60. passing Arduino analogRead and making "task" (like NMEA2000.ParseMessages()), which starts analog conversion and polls it until it is ready. The total conversion time is about 200 us, but naturally task spends only few us in maximum. With all corrections and filtering I reached with ESP32 ADC +-0.05V accuracy and +-0.03 noise over 32V. In Teensy we have +-0.007V accuracy. I am not sure about the noice, since my controlled PSU start to limit that.

tokonur218, 2020-10-13:
Hi,

I have a board that works fine with your Library. Which is using ESP Wroom 32. I am changing the Chip with ESP Wroom 32E-v3  and it is not working. 

https://www.esp32.com/viewtopic.php?t=15581 
i found that there is some changes on new chip.
How can I change bus speed ?

ttlappalainen, 2020-10-16:
Currently there is no option for that. I have to get one new version and improve code.

JohnMacrae, 2020-10-16:
I have one here - I will program it and test on my bench network in the next few days.  Worth noting that (last time I looked) Mouser had V2 chips in stock too (if you need a quick work-around).

tokonur218, 2020-10-16:
Thanks for your library and your reply;
I changed this Code  

    speed(CAN_SPEED_500KBPS), TxPin(_TxPin), RxPin(_RxPin),
// CAN_SPEED_250KBPS  for ESP Wroom-32,  CAN_SPEED_500KBPS for  ESP Wroom-32E

in  NMEA2000_esp32.cpp and it is working fine.

ttlappalainen, 2020-10-16:
Thanks. I would like to add this as parameter so that it is possible define on constructor. Do you know does it have some special processor definition to differ it from ESP32? For normal ESP32 I use code
#elif defined(ARDUINO_ARCH_ESP32) || defined(ESP32)
E.g. every Teensy version can be tested:
#elif defined(__MK20DX256__) || defined(__MK64FX512__) || defined (__MK66FX1M0__)

ttlappalainen, 2020-11-15:
Please test with updated NMEA2000_ESP32 library.

g7wbd, 2023-10-16:
esp32 dev module selected as chip

 sketch\DataDisplay2.ino.cpp.o:(.literal.startup._GLOBAL__sub_I_NMEA2000+0x4): undefined reference to `tNMEA2000_esp32::tNMEA2000_esp32(gpio_num_t, gpio_num_t)'
Error compiling for board ESP32 Dev Module.

any help solving this complier error please

ttlappalainen, 2023-10-16:
Did you installed also NMEA2000_esp32 library?

g7wbd, 2023-10-16:
"Did you installed also NMEA2000_esp32 library?"
yes

ttlappalainen, 2023-10-17:
"undefined reference to..." is linker error and means that it can not find referenced function/object - in your case tNMEA2000_esp32::tNMEA2000_esp32. That is defined on NMEA2000_esp32.cpp file. If you are using Arduino IDE, check that you have under your libraries folder NMEA2000_esp32 and under it NMEA2000_esp32.h and NMEA2000_esp32.cpp. Also check that you do not have NMEA2000_esp32.h on any other place.

closed 2024-01-21 by ttlappalainen