ttlappalainen/NMEA2000 #99: CAN BUS SHIELD V2, init problem

kephaII, 2018-02-08:
When i try the NMEA2000 examples on a MEGA with the latest CAN BUS SHIELD v2.0 from Seeed:
Then it will not initialize the CAN bus.
Either it says nothing or  CAN DEVICE FAILED TO OPEN,   (DataDisplay2)

I had started with trying the SEEED studio examples, i got the same problem, those demos also failed.
SEEED studio then  sent me an old version (attached). 
And that works works with its demos. 

May be your library have the same challenge with this new CAN BUS Shield? (see below)

Best regards

Hans

[CAN_BUS_Shield_(1) (1).zip](https://github.com/ttlappalainen/NMEA2000/files/1706983/CAN_BUS_Shield_.1.1.zip)

Re: Can bus Shield V2 init failure
seeed_support <tickets@seeed.uservoice.com>
fr 26.01, 04:07
Du;techsupport@seeed.cc
Hi Hans,

We've checked the code, it seeems that there is something wrong with the library.Then engineer is working on it. At present, you can download the old library i attched.It works well.

When you add this library, you can find the demo code by  File->Examples->mpu_can .In case you can not find it here, you can open the send.ino file directly by xxxx\CAN_BUS_Shield\examples\send.

ttlappalainen, 2018-02-08:
Have you tested with CAN_BUS_shield under my repsitory? I comapred the files on the zip and the files under my repository and they has no differencies.

kephaII, 2018-02-08:
Hi again

I removed the library i got from SEEED and replaced it with yours.

**receive_check.ino** works fine, the int and RX flashes as the ESP32 sends data every second. 

**receive_interups.ino**    says CAN BUS Shield init ok!, the RX led is on constant, and the int LED is off all the time. But no data is displayed on the monitor.

I wonder if it has something with the interupt pin.
I do not know what that is for this shield, I tried with 21 and 2, but no help.
in **receive_interups.ino**   the MCP2515_ISR  =342      
     attachInterrupt(0, MCP2515_ISR, FALLING); //start interrupt

best regards

Hans

kephaII, 2018-02-08:
Hi

I have tried many of the samples under NMEA2000, and they fail during initializing. with the message
CAN device failed to open.

Hans

kephaII, 2018-02-08:
Hi

I tried to comment out the INT pin, but same result.
I tried with attachInterrupt(digitalPinToInterrupt(2), MCP2515_ISR, FALLING); then it stopped.

But i get it to display data on monitor now,  I must have altered the code in the first run. 

So now **receive_interups.ino** and **receive_check.ino**  works.
only NMEA2000 fails to initialize

Hans

ttlappalainen, 2018-02-08:
Have you tried to set:
#define N2k_SPI_CS_PIN 9
before 
#define N2k_CAN_INT_PIN 2

kephaII, 2018-02-08:
Bingo!!!!!!!!!!!!

That did the trick!
In Main Handler: 10752
29480 : Pri:0 PGN:10752 Source:250 Dest:127 Len:8 Data:0,A6,6,A7,6,65,78,FD
In Main Handler: 10752

I try to send the heading data from  the ESP32 with header as
{0, 0x2A, 0x7F, 0xFA};
compass addres 42    PGN  127250

Seems that i got it wrong. when i feed  as address  into ESP32-CAN-Driver 

best regards

Hans,
 from snowing Oslo

ttlappalainen, 2018-02-11:
If you have working CAN driver, you can write similar inherited tNMEA2000_ESP32 as e.g. tNMEA2000_avr (see. https://github.com/ttlappalainen/NMEA2000_avr). That is the simpler one and does not use interrupts etc. The better ones line tNMEA2000_Teensy uses several buffers and interrupts. But as far as I unrderstood, ESP32 has only one output MailBox.

kephaII, 2018-02-11:
Thanks for helping me!

It now works very well. I got the bits in the header sorted out....
I had to slow down the ESP32 a lot, for the Mega could not handle that throughput.

But i am afraid that i am to busy for making an ESP32 version, for several months to come.
Hopefully some else does it in the meanwhile, your library is very nice.
SInce the ESP32 has built in Wifi and BLE, i hope to transmit messages as NMEA that way too. (but only once per second perhaps).
(hope somebody pick up that idea and make a version for that also)....
The more flexible, the better.

I also send out 127257 message with yaw pitch roll. 
The yaw had strange value in DataDisplay2. 
What  sort of angle convention is that telegram?
pitch and roll came out as expected though.

Best regards
Hans

ttlappalainen, 2018-02-12:
Check out from N2kMessages.cpp how to fill data fields right. That code is tested against other systems.

Each message has default send period. 127257 has 1000 ms and e.g 127250 and 127251 100 ms. With those rates Mega has no problems on receiving.