ttlappalainen/NMEA2000 #101: NMEA 2000 Resource

ben22ee, 2018-02-13:
Hello Timo,  
Thank you for your sharing your work.  I am trying to get a Raspberry Pi 2 working with the Pican2 Duo Isolated board.  I am able to use cansend and candump to transmit data between the two channels, but have not been able to get the sample code on the NMEA2000_socketCAN site to display network traffic yet.  Here is the link to the issue I created on the NMEA_socketCAN site:  https://github.com/thomasonw/NMEA2000_socketCAN/issues/1

I was able to use Wireshark to select the can port and view/record the data that I saw when using candump so I know the hardware is working.  If you have any suggestions on how I might be able to get it working, please let me know.  I have a Pi 3 that I might try to see if it makes a difference.

As I have been trying to figure things out, I found these links online and thought the information might be helpful if you do not already have it.
http://www.nmea.org/Assets/july%202010%20nmea2000_v1-301_app_b_pgn_field_list.pdf
http://continuouswave.com/whaler/reference/PGN.html

Thanks again for sharing your work.

ttlappalainen, 2018-02-13:
First, do you have teminator resistors connected? The behaviour is very unexpectable, if you do not have.

I added also response to the https://github.com/thomasonw/NMEA2000_socketCAN/issues/1. Read there some instructions for forcing SocketCAN.

Which example you are running? If you run e.g. TemperatureMonitor, it should send temperature periodically in every 2.5 sec. You could increase that to 1 sec by changing `#define TempUpdatePeriod 2500` to `#define TempUpdatePeriod 1000`. Then on setup(), uncomment debugging lines:
 ```
Serial.begin(115200);
 NMEA2000.SetForwardStream(&Serial);
  // If you want to use simple ascii monitor like Arduino Serial Monitor, uncomment next line
 NMEA2000.SetForwardType(tNMEA2000::fwdt_Text); // Show in clear text. Leave uncommented for default Actisense format.
```
Then you should see some debug messages in clear text like could it open CAN: "CAN device ready" or "CAN device failed to open", if it fails. After a while it will write message: "PGN 130316  send failed", if transmitting does not work at all.

ben22ee, 2018-02-14:
Thanks for the reply to both sites.

Yes, I do have termination resistors installed.  I tested with the Pican onboard resistors by soldering on screw terminals so I can easily add a jumper wire to enable/disable the resistors.  I used Electronics-Salon PCB Universal Screw Terminal Blocks Assortment Kit - https://www.amazon.com/gp/product/B017ATZY7W/ref=oh_aui_detailpage_o05_s00?ie=UTF8&psc=1 and they work well on the Pican.

I also tested with NMEA cables/T's/120 ohm terminations on both ends (with pican resistors taken out of the circuit).  I used a Lowrance starter kit for this setup that I bought a local store.  I get the same response regardless of which termination resistor setup I use.

To test the hardware, I use candump to display can 0 data and execute a Python 3.4.2 program to generate can 1 data.  I see the data on can0 in candump and on wireshark but not in the NMEA2000_socketCAN sample program.  When connected to the NMEA network, I see device traffic on candump and on wireshark but not the NMEA2000_socketCAN sample program.  Here is my python program that I use when not connected to other devices (only in loopback config) from modified sample code I found online (I am not an expert programmer).  It adds 1 to each byte as it sends continuous messages (until you press ctrl+c).  Its a walking 1's program and it works.

import can
bus = can.interface.Bus(channel='can1', bustype='socketcan_native')

a=0
b=1
c=2
d=3
e=4
f=5
g=6
h=7

while True:
    if a == 255:
        a = 0
    elif a <=255:
        a = a + 1
    if b == 255:
        b = 0
    elif b <=255:
        b = b + 1
    if c == 255:
        c = 0
    elif c <=255:
        c = c + 1
    if d == 255:
        d = 0
    elif d <=255:
        d = d + 1
    if e == 255:
        e = 0
    elif e <=255:
        e = e + 1
    if f == 255:
        f = 0
    elif f <=255:
        f = f + 1
    if g == 255:
        g = 0
    elif g <=255:
        g = g + 1
    if h == 255:
        h = 0
    elif h <=255:
        h = h + 1

    msg = can.Message(arbitration_id=0x1, data=[a, b, c, d, e, f, g, h],
    extended_id=False)
    bus.send(msg)
    print ("sending message on Can1", msg)
    
There was an issue I had in following the socketCAN setup instructions for automatically bringing up the can ports when the Pi boots.  The instructions say to do the following:
sudo nano /etc/rc.local
# Initiate the CAN ports
sudo /sbin/ip link set can0 up type can bitrate 250000
sudo /sbin/ip link set can1 up type can bitrate 250000

This did not work for me.  I found someone had the same problem and posted this way to automatically bring up the can ports on bootup.  This worked for me.

sudo nano /etc/network/interfaces

/*  then add the following lines at the bottom: */
auto can0
iface can0 inet manual
        pre-up /sbin/ip link set can0 type can bitrate 250000 triple-sampling on
        up /sbin/ifconfig can0 up
        down /sbin/ifconfig can0 down

auto can1
iface can1 inet manual
        pre-up /sbin/ip link set can1 type can bitrate 250000 triple-sampling on
        up /sbin/ifconfig can1 up
        down /sbin/ifconfig can1 down

I'm not sure yet, but I think there might be a problem with how the interrupts are mapped.  The Pican website says RX interrupts are mapped to GPIO 24 and 25 but in the schematic I do not see those being wired up or used.  There are also differences between how the GPIOs are mapped between a Pi 2 and Pi 3.  Copperhills Technologies has been very helpful in answering questions so I just emailed them to help clarify how the interrupts work and where Interrupt 24/25 come from.  I will report back on what I hear.

ben22ee, 2018-02-14:
I heard back from Copperhills Technologies.  It appears the website is correct with interrupts being on 24 and 25.  The kernel maps the ports with can0 labeled as CanB and can1 labeled as CanA on the board.  This is good to know to verify connections are made properly when only using one can port or using both can ports on different networks.

ben22ee, 2018-02-15:
This is my hardware setup:

Raspberry Pi 2 Model B (also have Pi3 to test with)
     *  Raspbian GNU/Linux 8 (jessie)
     *  Linux 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux
     *  Netbeans 8.2
     *  Real VNC 6.2.1

SanDisk Ultra 128GB microSDXC Card
https://www.amazon.com/gp/product/B010Q57S62/ref=oh_aui_detailpage_o04_s00?ie=UTF8&psc=1

Pican2 Duo Isolated canbus board
	http://copperhilltech.com/pican2-duo-isolated-can-bus-board-for-raspberry-pi-2-3/
	screw terminals for termination resistors: https://www.amazon.com/gp/product/B017ATZY7W/ref=oh_aui_detailpage_o05_s00?ie=UTF8&psc=1

Sunfounder 10.1” HDMI Monitor
https://www.amazon.com/gp/product/B01J52TWD4/ref=od_aui_detailpages02?ie=UTF8&psc=1

WiFi Adapter
	https://www.amazon.com/gp/product/B003MTTJOY/ref=oh_aui_detailpage_o06_s00?ie=UTF8&psc=1

Standoffs (I think I should have bought M2.5 size for better fit but I bought M2 size below and they work)
	https://www.amazon.com/gp/product/B01HBV79JU/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1

Keyboard/Mouse
For portable use:  https://www.amazon.com/Rii-Wireless-Keyboard-Rechargable-Battery-Black/dp/B00Z81U3YY/ref=sr_1_1?s=electronics&ie=UTF8&qid=1518651747&sr=1-1&keywords=rii+keyboard&dpID=51cLu1CX-ZL&preST=_SY300_QL70_&dpSrc=srch
At home (if not connected via VNC Server):  https://www.frys.com/product/9301630?site=sr:SEARCH:MAIN_RSLT_PG

ttlappalainen, 2018-02-15:
Did you test, what I suggested?

Note that NMEA2000 system may refuse to handle any non real NMEA2000 data. So if just sent some CAN bus data, NMEA2000 library may not show anything.

I forgot to mention that the best Open Source PGN referencies are:
https://github.com/canboat/canboat/tree/master/analyzer
https://github.com/OpenSkipper/OpenSkipper/blob/master/OpenSkipperApplication/Resources/PGNDefns.N2kDfn.xml
and my library N2kMessages.cpp.

Others has been inherited from canboat, but all fixes and information on others  has not been implemented to the canboat. N2kMessages.cpp has been created PGN by PGN for real use, so as far as I know, all of them has been also tested.

ben22ee, 2018-02-16:
Thank you and thomasonw for your help.  

I just downloaded the new NMEA2000_SocketCAN files.  It compiled but I noticed some yellow light bulbs on some lines which said "Unable to resolve identifier...".  I found an article on stackoverflow.com (https://stackoverflow.com/questions/14913243/netbeans-7-2-shows-unable-to-resolve-identifier-although-build-is-successful) that suggested to reparse the project.  I did that and the yellow bulbs went away and I was able to compile it again.

I will test this weekend with my Pi2.  If it doesn't work, I will switch over to my Pi3 to see if that makes a difference since that is what thomasonw has been testing on.

ttlappalainen, 2018-02-19:
Good to know. I have RPi 3 on the shelf waiting for NMEA2000. It would be nice to have simple instructions how to get it running with NMEA2000. I have been lazy to dig all the links what to load, all the commands and settings to run etc.

The example ActisenseListener reads all messages from bus as default, but does not show itself on the bus. It also reads all proprietary messages. The problem with fastpacket messages is that library has to know, which messages are fast packet format. I have not found any other system than list the messages on table. Library has own default table for fastpacket messages, which you can extent with library command:
```
...
const unsigned long ExtraFpMessages[]={123456,123457,123458,0}; // Terminate list with 0.
...
NMEA2000.ExtendFastPacketMessages(ExtraFpMessages);
...
```

ben22ee, 2018-02-19:
I have a detailed log of steps I did to get it working.  I need to clean it up and will post them.

I actually have a pi, GPS hat (I wanted to timestamp data but not really necessary if GPS is already on the network and being logged), then the pican hat on top of GPS hat.

thomasonw, 2018-02-19:
@ben22ee :  Good to know things are working.   Hey, for RPi guide, if you could look here and perhaps make changes that would be great:   

[https://github.com/thomasonw/NMEA2000_socketCAN/blob/master/Raspberry-Pi-CAN.md](https://github.com/thomasonw/NMEA2000_socketCAN/blob/master/Raspberry-Pi-CAN.md)

Can keep the info in one place!

ben22ee, 2018-02-20:
@thomasonw :  Your instructions were very helpful in getting things working.  I made some suggestions and a pull request.  I'm learning Github, so if I submitted suggestions using wrong process, please let me know.

thomasonw, 2018-02-20:
Looks like it worked well!   Thank you for the input.

 

-al-

joshoreefe, 2023-02-15:
Hi all, some advice is needed to get the updated system up and running. I have a working setup with older sources. Now testing with the updated sources the can socket won't open. I have tested the suggested runtime setting of can port, but no joy.
Platform is RPi 4, 2G RAM,  Raspbian GNU/Linux 10 (Buster) with a twin port can hat.

Here's an excerpt from my code:

```
char socketPort[50];
#define SOCKET_CAN_PORT socketPort
…..

    strcpy(socketPort, "can0");

    if (!NMEA2000.Open()) {
      //       cout << "Failed to open CAN port " SOCKET_CAN_PORT << endl;
       cout << "Failed to open CAN port "  << endl;
       return 1;
   }

```
This fails with "Failed to open CAN port"
I also tried the other can port, can1 with equal result.

As said above, I have another setup with previous sources from a couple of years back, and it is working okay.
Any ideas why this fails?

BTW, currently there's nothing connected to the physical N2k output socket.