ttlappalainen/NMEA2000 #103: 1785 byte message length support?

lstrz, 2018-02-22:
If I were to change MaxDataLen from 223 to a higher number, and try to send a message longer than 223 bytes, would things work properly?

ttlappalainen, 2018-02-22:
I do not think so. Most likely you will blow yout your memory, since there are reservation for message reception, which you will then multiply with 5. And second currently there is no command to send message in TP format. It has been under my plans, since TP support already exists, but no other has asked it. Where do you need that long messages and are you sure the other end will support it?

lstrz, 2018-02-22:
I need it for PGN 129540 -- tracked GNSS satellites.

The receiving device says that it can receive PGN 129540, so if it's not lying and if it implemented the standard correctly, everything should work. My concern is correct implementation on my side, not potential incorrect implementation on the receiving :)

With GPS, GLONASS and a quality GNSS receiver, the total number of tracked satellites easily reaches 19 or more satellites. 19 satellite * 12 bytes per satellite = 228 bytes > 223 bytes.

In any case, thank you! The library is fantastic and saved me a huge amount of work. Having TP in would be icing on the cake.

ttlappalainen, 2018-02-23:
When do you need that support? Can you live for a while with 18 satellites?

Have you found my NMEA simulator? I have there some code for testing TP as broadcast, so I could probably add functionality for testing PGN 129540.

lstrz, 2018-02-26:
I can live with 18 for a while, sure. Not shipping N2K support on the device for another few months. Even then, it will probably be a beta at first.

I will take a look at the simulator.

ttlappalainen, 2021-02-06:
This is now finally supported on library. I also added it to examples MessageSender and DataDisplay2. Also NMEA Simulator simulates it with real satellite data.

lstrz, 2021-02-06:
Thanks a bunch! I will close this issue now since this is done.

closed 2021-02-06 by lstrz