ttlappalainen/NMEA2000 #105: NMEA2000 Library in a NMEA Certified Product

atestani, 2018-03-02:
Hi Timo

I just want to let you know that your library is used in a product that recently passed NMEA Certification Testing.  Thanks very much for making the library available to everyone.  

The library version in the certified product is dated 15.01.2018.  I recommend that you create a branch to the library to archive the 15.01.2018 version and do all development and modification on a separate code base.  

Thanks again for the library!

Al

ttlappalainen, 2018-03-02:
Thanks, that is great information! So I have managed to make something usefull.

I hope that others also find the library so that we would get more innovative devices for NMEA 2000.

I'll make some organization with brances. After 15.01 there is no effective changes on core so I can use current version.

mrbubble62, 2018-03-07:
That is awesome news, congratulations Timo.

atestani, 2018-03-26:
> I made a branch Production. There are some new features between 15.01 and
> Production branch, but they does not effect old.

Sounds good.... thanks.

melucas, 2018-04-22:
Congratulations Timo.

hcalin, 2021-01-25:
Hello guys,

@atestani Can you tell me please which board did you use for certification? And also for which level did you certify the product? I'm having trouble with NMEA certification tool for some tests right now for Arduino Due/Arduino Mega. I tested with production branch and also with development branch and some tests still fail. 

For e.g. for battery monitor example:
If I disable the transmission of battery specific messages test C.3.3.1 will also pass, if not it will fail to transmit PGN 65240 on request and the testcase will fail.
But of course I will need the transmission of those cyclic PGNs for other testcases.

I'm noting that I try to certify a Level A product. 

Did you have a session of bug solving yourself to get the product certified or was it plug&play for the board you used? 

Thank you in advance.

atestani, 2021-01-25:
We used a custom board in a custom product.  There are multiple issues with
the test tool that you will encounter and have to be documented as part of
you submission.

hcalin, 2021-01-25:
@atestani 

Thank you for your fast answer. I am aware that some of the tests failed due to the tool, but I don't think those cover the tests cases where the SW faild to answer to a request.
So I would appreciate very much if you could answer these questions:

1. Did you do any modification to the software stack before the certification? Put otherwise, did you find any bugs in the software stack? I'm referring only to the NMEA2000 stack, not the CAN related libraries. Although if you ported the CAN libraries, this would be valuable information also.
2. What level has the product which you certificated? since the test cases are different in number for a level A or level B product.

Thank you again for your time.

atestani, 2021-01-25:
The level of the library, at the time of certification, was archived. It
captured any changes made to pass certification.  This was in 2017 so I do
not know if any changes made to the library since then are problematic.
Certification was to level A

ttlappalainen, 2021-01-26:
As I hold the master version, I have also tested all changes. The current version has just been passed certification tests for RPi and ESP32. For low level drivers I know passed devices done for Teensy 3.1, 3.5 (FlexCan), RPi (Socket can) and ESP32.

Certification is commercial process for which I can help for 95 €/h.

orca-hydromancer, 2021-01-26:
@hcalin I suggest taking up Timo on his offer. Last month I went through testing latest master with the (outdated) testing tool. All the tests that fail have good explanations why they fail and Timo was able to point those out.

davidhoy, 2021-04-02:
Does anyone have a list of which tests fail, and the explanations?  

I'm seeing issues with test C.4.8.x, all around the address claim on startup.  The tool says that it received messages before the address claim is complete, but monitoring the bus with a separate tool I see no such messages.  But the entire C.4.8.x set of tests fails because of this one issue.  

Also seeing issues with test C.4.11.1, which I'm pretty sure is the test tool.  It monitors the bus for about one minute then complains that it cannot determine the frequency of the heartbeat message, which is transmitted every 60s.  How can it ever determine the interval if it only monitors the bus for one minute?

I'm running the stack on a custom board using a Microchip PIC32 microcontroller.

-David

ttlappalainen, 2021-04-02:
I have not published that list, since certification will be normally done for commercial products. I do that as commercial consulting and normally costs are 100-300 € depending how much time I need to spent. I also need some cheese over bread. If you are interested, please contact me directy.

hcalin, 2021-04-02:
> Does anyone have a list of which tests fail, and the explanations?
>
> I'm seeing issues with test C.4.8.x, all around the address claim on startup. The tool says that it received messages before the [...]

Hello David, please note that for address claim tests you must unplug the power before starting the test in the certification tool, and then power it up. That may explain why the board sends messages before address claim.

For the C.4.11.1 that is indeed the test tool; I'm pretty sure it will pass from NMEA side.