ttlappalainen/NMEA2000 #107: SN65HVD230 CAN Bus Transceivers

melucas, 2018-04-22:
Hello,

Found these CAN bus transceivers on eBay ([SN65HVD230](http://www.ti.com/lit/ds/symlink/sn65hvd230.pdf)).  Will they work with this NMEA200 library?

Thanks,
Michael

ttlappalainen, 2018-04-22:
Library does not see tranceivers. Only CAN controller matters. As you can read from documents currently supported contollers are Arduino DUE, ESP32, Teensy 3.2-2.6, MCP2515. Tranceiver makes a physical connetion from controller to bus. With quick look I did no saw reason why that would not work with CAN bus - that what it is for. Just take care that your controllers is 3.3 V.

melucas, 2018-04-26:
Will a Mega board work?  I have a sparkfun CAN-BUS shield (DEV-13262) which has a MCP2515 chip to connect to the Mega.

melucas, 2018-04-29:
Started testing with a DUE.  I'm trying to get the Getting Started section, I'm trying to get the TemperatureMonitor to run.

I'm seeing the following error:

PGN 130310 send failed
37725 : Pri:5 PGN:130310 Source:22 Dest:255 Len:8 Data:1,C1,70,FF,FF,FF,FF,FF

How do I debug the sending to find out what the send issue is?

Thanks,
Michael

melucas, 2018-04-29:
Is it possible to tell the NMEA 200 library which SDA and SCL pins to use?
The Due and CAN-BUS shield are connected to SDA1 and SCL1.

melucas, 2018-04-30:
Dumped the CAN-BUS shield and a connection to my Raymarine SeaTalkNG network working with the CJMCU-230 Can Bus Transceiver:

- CTX pin to DUO CanTx pin
- CRX pin to DUO CanRx pin
- CANH pin to White Raymarine spur cable
- CANL pin to Blue Raymarine spur cable

closed 2018-04-30 by melucas