ttlappalainen/NMEA2000 #113: Building an executable from NMEA2000ToNMEA0183 example code

davidsanner, 2018-05-18:
I have a RPi3 w/ a can0 interface (using canable.io) and would like to output 0183 to kplex (FIFO or tcp)
I've cloned: NMEA0183  NMEA2000  NMEA2000_socketCAN with some success in compiling things but not much when linking.

I was able to build libnmea2000.a using cmake/make but running into a wall getting the NMEA2000ToNMEA0183 example to build/link.     

Am I missing something obvious to build NMEA2000ToNMEA0183?  What does it need to link to? Thanks.

ttlappalainen, 2018-05-18:
Those should be all necessary libraries. But did you also build NMEA0183 and NMEA2000_socketCAN?

davidsanner, 2018-05-18:
I was able to manually (no make file) compile all of the .cpp files into .o and create using cmake/make the libnmea2000.a library. I'm just not having any luck linking them into an executable for the main.cpp example code in NMEA2000ToNMEA0183

example of compile:  
 cwd: ... NMEA0183/
 g++ -o NMEA0183LinuxStream.o -c NMEA0183LinuxStream.cpp
and cwd:  ... NMEA2000/Examples/NMEA2000ToNMEA0183
g++ -I ../../src  -I ../../../NMEA2000_socketCAN/ -I ../../../NMEA0183/ -c -o N2kDataToNMEA0183.o N2kDataToNMEA0183.cpp
g++ -I ../../src  -I ../../../NMEA2000_socketCAN/ -I ../../../NMEA0183/ -o main.o -c main.cpp

I'm just sure what/how to link them into an executable.  I've been trying many combinations of something like this:   g++ N2kDataToNMEA0183.o main.o -o main   ../../src____.o  ../../src/libnmea2000.a    
Am I missing something obvious? A build tool, common library?  Thanks much.

ttlappalainen, 2018-05-19:
Sorry I am not familiar with g++ make files. That is why I used Code Block environment, where it is easy to add necessary files to project. I would have like to use Atom and Platform IO, but could not find a way to get it running on RPi. The Code Block is easy. The one trick to get it stable was to disable Code completition: Plugins-Manage plugins... disable Code completition

davidsanner, 2018-05-19:
codeblock!   I was able to compile after a few links back to the projects from lib dir.  Thanks.
NMEA0183 -> ../../../../NMEA0183
NMEA2000 -> ../../../../NMEA2000
NMEA2000_socketCAN -> ../../../../NMEA2000_socketCAN/
( cwd:  NMEA0183/Examples/NMEA2000ToNMEA0183/lib )
And making sure it could find all the .h files.

I was going to see if I could get the RMC output  w/ 100th of a second resolution (guess work off of millis since last GNSS time update).  I was noticing that with my current 10Hz rapid update GPS & Heading sensor all I'm seeing in the output is HDG/VTG/RMC ... no ROT or GLL even though there's some code to support those sentences and it's reading PGN's like 127251  (GSV would be nice as well w/ 129540 & Attitude - pitch/roll with 127257).  ROT just not enabled? 

(btw, I had a great time in Turku & out on a tiny island there years back... Finland!)