ttlappalainen/NMEA2000 #115: NMEA2000 master or slaver ？

craigtao, 2018-06-08:
this lib is for master or slaver？

thank you！

craig

craigtao, 2018-06-08:
@ttlappalainen 
thank  you so much！

Now, I have a project based on the can bus, the application layer protocol is NMEA2000

Can this library be used in real projects?
Can it meet the requirements?

hardware， I choice arduino due

thank you！

craig

craigtao, 2018-06-08:
This is a ship monitoring system，
All sensors support NMEA2000 protocol

craigtao, 2018-06-08:
(*￣︶￣)

craigtao, 2018-06-08:
Ship navigation system

ttlappalainen, 2018-06-08:
Yes you can. As I mentioned the library is certification ready. See issue https://github.com/ttlappalainen/NMEA2000/issues/105. For full certification you need to save some parameters from library to somewhere e.g. EEPROM to be restored on system start. Also you need to join NMEA community (800 $/year) to get manufacturer number, buy product number and certification costs. Totally some thousand dollars.

craigtao, 2018-06-08:
About OpenSkipper
Welcome to the OpenSkipper project, which provides Open Source C# code for Windows for integrating and displaying NMEA 0183, NMEA 2000 and AIS data from nautical instruments, GPS units and internet data sources. OpenSkipper can be run on a laptop aboard your boat to show electronic instruments displaying speed, heading, etc. Open Skipper can also receive and transmit data over multiple connections, including a serial port (for NMEA 0183), an ActiSense NGT-1-USB NMEA-2000-to-USB converter to read NMEA 2000 (N2K) data, and wired and wireless network connections (including TCP and UDP). It also contains a built-in webserver, so you can run OpenSkipper on a laptop and use this to display data on an iPad or Android phone or tablet.

product  NGT-1-USB NMEA-2000-to-USB

Where can I buy？

ttlappalainen, 2018-06-09:
You do not need to buy NGT-1. You can use arduino and some of my exmples. For listen only ActisenseListener does the same thing. Just for listening bus traffic load Actisense NMEA Reader for use with Arduino and ActisenseListener.

I also prefer to go Teensy family instead of Arduino Due. Teensy 3.2 for simple things. Teensy 3.5 can do very complex. ESP32, if you want to have WiFi.

craigtao, 2018-06-09:
teensy3.2 not support can-bus，ESP32 Development environment is a bit tedious，
arduino due is easy to use，
(*￣︶￣)

thank  you so much

craig

sarfata, 2018-06-09:
@craigtao If you want to see an exemple of a teensy 3.2 with CAN Bus, look at my project KBox: https://github.com/sarfata/kbox-hardware

You can buy pre-built units from my tindie page: https://www.tindie.com/products/sarfata/kbox-open-source-boat-gateway/

ps: I hope Timo will forgive me for this shameless plug ;)

craigtao, 2018-06-14:
@ttlappalainen  Hi,

i want use arduino mega , can run NMEA2000 lib ?

Due version is a bit expensive。teensy expensive too (*￣︶￣)

thank you,

ttlappalainen, 2018-06-14:
Yes you can, but its memory is very limiter and you can not do very complicate systems with Mega. I prefer Teensy.

craigtao, 2018-06-14:
@ttlappalainen  teensy 3.2 is enough？ or 3.6 ？
Teensy's IO pins are few，

Is its memory bigger than Mega?

craigtao, 2018-06-14:
i want to buy this
http://skpang.co.uk/catalog/teensy-canbus-breakout-board-include-teensy-32-p-1507.html

But I don't know how to operate

i‘m in China’

ttlappalainen, 2018-06-14:
KBox is based on Teensy 3.2, so you can do rather big project for it. It has also extra IO on pads under board. 3.6 has more memory and SD card reader build in.

That breakoutboard should work fine, since it simply has DC-DC converter and MCP2562, which I have used with all different systems.

craigtao, 2018-06-14:
which I have used with all different systems
is breakout board  or teensy3.2 ？

ttlappalainen, 2018-06-15:
MCP2562 tranceiver chip. It can be used on Teensy, DUE, ESP32, RPi etc. I have also used Teensy 3.2 for many systems.

craigtao, 2018-06-17:
How about stm32f10x?
Can I run this NMEA2000 library?
My concern is that MDK does not support the features of C++.

ttlappalainen, 2018-06-17:
There are NMEA2000 driver classes for Teensy, Due, MCP_CAN, Socket_CAN, AVR, MBED, ESP32. If you want to use stm32f10x, you have to write driver class by yourself or send me one board, so I can write the driver.

I still wonder why to make things complex? Teensy is very powerfull and I just measured that my Teensy 3.2 board connected to NMEA2000 bus takes 14 mA power 24 mHz or 24 mA 94 MHz, while Due board on similar work takes 121 mA power. My Teensy solution reads several 1-wire sensors and sends data to the bus and works fine with 24 MHz. I also have system with Teensy 3.5 with 7" touch display taking only 150 mA display on and 90 mA display off. All those has been measured on 12.5 V.

Do you have some special system or why you should use something different than what is currently supported and tested? And if you do not need WiFi, I think that Teensy is the best option.

craigtao, 2018-06-17:
Is the ESP32 development environment Arduino IDE?

ttlappalainen, 2018-06-17:
Yes it is possible. Also example is for Arduino IDE. I prefer for any complex projects e.g. Platform IO.