ttlappalainen/NMEA2000 #116: CAN bus just receive 8 Byte, this lib how to NMEA2000 message？

craigtao, 2018-06-09:
I have a question，

CAN bus just receive 8 Byte, this lib how to NMEA2000 message？

this class ActisenseReader ？

craig

ttlappalainen, 2018-06-09:
See the example codes like ActisenseListener. Example DataDisplay2 just catches several messages and translates them to clear text to serial monitor. Take also look into the document https://github.com/ttlappalainen/NMEA2000/blob/master/Documents/NMEA2000_library_reference.pdf and follow the sample to make TemperatureMonitor. It may help you undestand NMEA2000 and library better.

craigtao, 2018-06-09:
ok，thank you，

Today I read the source code，

A little understand the principle of processing
N2K_NO_ISO_MULTI_PACKET_SUPPORT

ttlappalainen, 2018-06-09:
This is if you desperately try to run library on low RAM systems like Uno and need to save RAM. Use systems enough RAM like Teensy 3.2 or higher, ESP32 or Due and library with default definitions. Disabling any feature causes that device is not certification ready anymore.

closed 2018-06-10 by ttlappalainen

craigtao, 2018-06-10:
Teensy 3.2 and MCP2515 or MCP2551 can run this NMEA2000 lib？

ttlappalainen, 2018-06-11:
It is possible, but do not see any reason why. Teensy internal CAN is better than MCP2515. Note also that MCP2515 and MCP2551 are not comparable, since one is controller and other is tranceiver.