ttlappalainen/NMEA2000 #120: Noobie question: Arduino comile error, and Fuel calculation idea

bsamaras, 2018-08-03:
First, can't tell you how thrilled I am to find this NMEA2000 library and the excellent documentation.
Me, an EE by trade. Do a lot of HW design and play quite a bit with Arduino. I get by with C++.

Have a boat with an older Raymarine C120 MFD. It can display NMEA2000 limited engine instruments. I purchased an engine monitor device (RS11) and got the RPM, temp, oil and fuel flow to work on the MFD  with an additional NMEA2000 fuel flow sensor.  The MFD can display "total fuel used" too, but there is no real engine controller to calculate this and the MFD is too dumb to perform this calculation.
What I'd like to do is build an Arduino DUE with NMEA2000, read the fuel flow sensor(s), perform a fuel integration calculation and write of "total fuel used" out through the proper PGN for display on the MFD.
What I'd really like to calculate is miles per gallon, but not sure I can get distance data from the MFD without a 183 bridge.
Does this sound do-able?

Also, I loaded all of the DUE libraries into the Arduino IDE. Got the correct HW selected for DUE and attempted to compile the temperaturesensor demo.
I get this compiler error:
'dsdsdd' was not declared in this scope

Any guidance is appreciated.  I'm sure I made some noobie error.
Thanks!
...bill

sarfata, 2018-08-03:
Bill, What you are describing sounds completely doable. I will try to help if I can with the compile error. Could you send a full screenshot or the complete output? It would make it easier to figure out what is going on.

Getting distance through NMEA would be pretty easy too. Timo also has a library for NMEA0183: https://github.com/ttlappalainen/NMEA0183 - Do you know what sentences are sent by your equipment?

bsamaras, 2018-08-04:
I discovered from Raymarine that my old MFD could display "fuel remaining" as a reported by PGN 127497, Engine trip parameters.
Also, here's a screen shot of the compiler error.  Compiled for Arduino DUE with DUE libraries, and the tempsensor demo with appropriate line commented/uncommented per directions.
[image]

ttlappalainen, 2018-08-04:
You have probably accidentaly typed "dsdsdd" to the beginning of the line. Just remove that. It does not exist on original example: https://github.com/ttlappalainen/NMEA2000/blob/master/Examples/TemperatureMonitor/TemperatureMonitor.ino

Have you monitored what data you have on the NMEA2000 bus. You can e.g. use sample ActisenseListener and Actisense NMEA Reader to see all data. Then you see, if you have distance data on bus. If you have it only on NMEA0183 side, you can use other port on due to read that by using NMEA0183 library.

bsamaras, 2018-08-04:
Yes indeed, dsdsdd was a fumble fingered insertion.   Duh. The code complies OK now.
I have a real Actisense NGT-1 so I can take a look at what I'm getting.  Off the top, don't recall any nav data on NMEA2000.  My DUE HW is still in flight, but I'm intrigued I can perform the reader function too.  Very nice.
Do I need transceivers or buffers to talk/listen to 183?  I haven't hunted the 183 space yet.

Thank you.

ttlappalainen, 2018-08-04:
Yes, you need. DUE inputs are TTL level. Just find ISL8488 or MAX3488 or any RS422 level tranceiver.

bsamaras, 2018-08-04:
While I'm waiting for my DUE to arrive, I have a few other curiosity questions. Hopefully, not too far off topic. 
1. Can an Open CPN display engine or other (NMEA2000 NMEA0183) gauges? I did not see any Open CPN plug-in for this.
2. Making the CPN WiFi wireless is what I'd like to do, especially if able to see engine data on a tablet.  How difficult is this?

Still thrilled I can make a NMEA2000/183 converter and just about any NMEA2000 device I can dream up.  Thanks for all of this. Hopefully, I'll be able to contribute.

ttlappalainen, 2018-08-05:
OpenCPN has very limited parameters, which it can show on dashboard. It is meant for navigation not for all data. If you use Windows PC, there is OpenSkipper, which can show allmost everything and you can configure displays for your purposes. I use on Win PC OpenSkipper and I am changing to OpenCPN from Nobeltec Admiral. On tablet I use remote connection (Tight VNC server on Win) to PC.

For Android there are some apps for showing bus data. In that case it would be best to use ESP32 on bus, so it could send all data to the WiFi. There is also example https://github.com/ttlappalainen/NMEA2000/tree/master/Examples/ESP32/NMEA2000ToWiFiAsSeaSmart

ttlappalainen, 2018-08-05:
Please move your further questions to NMEA 2000 shield Arduino forum so all other interested will see the responses. Here thery are bit hidden after issue has been closed.

1. Yes it can. I have DUE as NMEA0183->Nk2 converter and N2k->PC converter. It also handles some digital inputs. 

2. NMEA0183 is not network. It is point-to-point connection. Only several listener can listen one talker. Normally there is no terminators on NMEA0183 connections, since baud rates are so small.

3. On NMEA2000 topology there is one terminator resistor on both end of main bus. Also drop cables should not exceed 6 m. This means that if you have wind wane on top of sail boat mast, that should be the main bus and you should have terminator at that end. In some drawings I have seen main bus from front to back of boat and mast has been as drop. See e.g. https://aaclive-attainableadvent.netdna-ssl.com/wp-content/uploads/2013/05/NMEA-2000.jpg On drop you never use terminator! So there will be 2 120 ohm terminators so total impedance for bus is 60 ohm. That you can also measure between H and L on N2k bus. If you measore different resistance, you have either too many or too few terminators.

Note also that making hub for N2k is not rocket science and you do not need to byu expensive hubs. I can send picture to forum.

3a.  It does no matter, which tranceiver you use. There just happened to be ready picture for DUE dual tranceiver. With that one could build bridge beween two large N2k bus. Normal boater will not need it, so just forget it. The terminators has jumpers, so you can connect them or not. If your device is at the end of bus, it can have terminator. E.g. B&G Wind wane has build in terminator, which you can not disconnect.

And as last I want to remind that I prefer Teensy 3.2 and up for N2k devices. It draws less current than DUE. I have e.g. temperature monitor compiled to 24 MHz and it draws only 13 mA from 12 V. I think DUE took over 100 mA.