ttlappalainen/NMEA2000 #121: Connecting NMEA 2000 to USB

ryall, 2018-08-21:
I noticed in the docs you mention creating an adapter for N2K <-> USB. I was wondering if you could provide an example and the Arduino extensions I would need to achieve this?

ttlappalainen, 2018-08-21:
There is example ActisenseListener for listen only and ActisenseListenerSender for both direction. I prefer to use Teensy boards  e.g. http://skpang.co.uk/catalog/teensy-canbus-breakout-board-include-teensy-32-p-1507.html or just Teensy and MCP2562 tranceiver as in e.g. https://github.com/ttlappalainen/NMEA2000/blob/master/Examples/TeensyActisenseListenerSender/Documents/Teensy_Actisense_listener_sender_schematics.pdf

ryall, 2018-08-21:
Thanks, I didn't notice that before. I see you have a WiFi broadcaster example too - I'm wondering if it's possible to output the Actisense signal over WiFi, maybe using a dongle, or would that not work?

ttlappalainen, 2018-08-21:
"Actisense signal" is just protocol, where N2k packets has been formatted. There no applications, which could use that format. Instead there are applications, which can use SeaSmart protocol as in example.

closed 2018-08-21 by ryall

ronzeiller, 2019-10-28:
> "Actisense signal" is just protocol, where N2k packets has been formatted. There no applications, which could use that format.
Just a small note to mention: 
Expedition program reads Actisense format perfectly direct via USB with commented  `// NMEA2000.SetForwardType(tNMEA2000::fwdt_Text);`

ttlappalainen, 2019-10-29:
That is correct. Also Nobeltec Admiral reads that. I do not know about Max Sea or Trident products. Trident was also from Nobeltec. 

The question was more about using WiFi. Is there applications, which reads Actisense format from network?

ronzeiller, 2019-11-02:
Yes, Expedition reads Actisense format from network too. (TCP/IP or UDP)